openwrt/mt76 #4: Call trace in MT7602EN

RedDwarf69, 2015-05-15:
Using the driver with the latest OpenWRT (mt76 2015-03-11) in an TWSZ AP699GE8C2 device (MT7602EN WiFi chip).
In station mode mode, doing a file transfer, I can easily reproduce this message.

Not sure how to debug it. Not sure what triggers the call trace, everything seems to continue working fine.

[   76.080000] WARNING: CPU: 0 PID: 1093 at kernel/softirq.c:146 __local_bh_enable_ip+0x54/0xc8()
[   76.080000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xn
[   76.080000] CPU: 0 PID: 1093 Comm: scp Not tainted 3.18.11 #3
[   76.080000] Stack : 00000000 00000004 00000006 800591a8 00000000 00000000 00000000 00000000
          803756e2 00000031 00000000 00000000 00000000 82de2d30 802d33c8 803365a3
          00000445 00000000 8037445c 82de2d30 00000640 832e5848 82fc4000 80059cf4
          8033aa20 80028d14 00000003 802d8dc0 802d7000 82c519c4 82c519c4 802d33c8
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[   76.080000] Call Trace:
[   76.080000] [<80016084>] show_stack+0x48/0x70
[   76.080000] [<8015bdc4>] dump_stack+0x64/0x84
[   76.080000] [<80028e1c>] warn_slowpath_common+0x84/0xb4
[   76.080000] [<80028ed4>] warn_slowpath_null+0x18/0x24
[   76.080000] [<8002bb2c>] __local_bh_enable_ip+0x54/0xc8
[   76.080000] [<82d01ab8>] ieee80211_tx_status_noskb+0x1c8/0x530 [mac80211]
[   76.080000] [<82d642c4>] mt76_set_channel+0x4f0/0x5d0 [mt76pci]
[   76.080000] 
[   76.080000] ---[ end trace f888427e6d7913e2 ]---

mehlis, 2015-06-20:
can reproduce with openwrt 15.05-rc1 on MT7621 Firefly FireWRT:

```
[  486.370000] ------------[ cut here ]------------
[  486.370000] WARNING: CPU: 0 PID: 0 at kernel/softirq.c:146 __local_bh_enable_ip+0x54/0xc8()
[  486.370000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mmc_block ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc sdhci_pltfm sdhci nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack mtk_sd mmc_core iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt i2c_mt7621 i2c_dev i2c_core mt76pci mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher leds_gpio xhci_plat_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  486.370000] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 3.18.11 #1
[  486.370000] Stack : 00000000 00000004 00000006 80059404 00000000 00000000 00000000 00000000
          803a56e2 00000034 00000000 00000000 00000000 80364b20 802fe8d8 80364783
          00000000 00000000 803a445c 80364b20 00000640 8ed6d848 8e64c000 80059f50
          80368c60 80028d94 00000003 803042d0 80302510 80353b4c 80353b4c 802fe8d8
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[  486.370000] Call Trace:
[  486.370000] [<80016104>] show_stack+0x48/0x70
[  486.370000] [<80169190>] dump_stack+0x64/0x84
[  486.370000] [<80028e9c>] warn_slowpath_common+0x84/0xb4
[  486.370000] [<80028f54>] warn_slowpath_null+0x18/0x24
[  486.370000] [<8002bbac>] __local_bh_enable_ip+0x54/0xc8
[  486.370000] [<8ed81ab8>] ieee80211_tx_status_noskb+0x1c8/0x530 [mac80211]
[  486.370000] [<8ece42c4>] mt76_set_channel+0x4f0/0x5d0 [mt76pci]
[  486.370000] 
[  486.370000] ---[ end trace aaf337e2a89279ba ]---
```

nbd168, 2015-07-06:
fixed in the current version

closed 2015-07-06 by nbd168