openwrt/mt76 #5: Support for ASUS USB N53 B1 dual-band wireless adapter?

davidcsterratt, 2015-06-20:
I've [filed an issue](https://github.com/kuba-moo/mt7601u/issues/24#issuecomment-113781329) with @kuba-moo about getting support with this device, which I think has an MT7662 chipset. I didn't know about your work until @kuba-moo told me about it.

nbd168, 2015-08-27:
I have no plans for working on USB device support at this point. Maybe that'll change after I add support for some newer chips, e.g. MT7603E

closed 2015-08-27 by nbd168

davidcsterratt, 2016-05-18:
In the School of Informatics at the University of Edinburgh students have the option of working on a project during a "summer of code", I was wondering if writing an MT7662 driver would be a feasible project. If so would it be best to start from this codebase or  @kuba-moo's [mt7601u](https://github.com/kuba-moo/mt7601u) code?

geki-yaba, 2016-12-23:
If anyone is interested in MediaTek USB dongle support (mt7610u and mt7612u for german AVM Fritz!WLAN AC 430 and AC 860 Sticks AFAICT), I would recommend to check upstreamed mt7601u driver[0] for USB layer and OpenWRT's mt76 driver for (at least) mt7612 chipset access and merge them. I guess the *e and *u chipsets do not differ badly, other than the access layer(USB vs PCI). Both are build on mac80211 interface. There is one more repository working on these chipsets based on the old MediaTek codebase[1]. There, one may cherry-pick details, if at all.

[0] https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt7601u
[1] https://github.com/ulli-kroll/mt7610u