openwrt/mt76 #7: Device hang after load firmware.

changeway, 2015-07-13:
I'm porting openwrt to a MT7620A+MT7612
E board. But my board hang after load firmware, and no respond.

[   14.300000] ASIC revision: 76120033
[   14.330000] ROM patch already applied
[   14.350000] Firmware Version: 0.0.00
[   14.350000] Build: 1
[   14.360000] Build Time: 201410061140____
"^啖EqY籯煌\?褓澓.褓q)–歉壂^z@SM煨@鄤w?

Do this driver/firmware support on specific ASIC revision?
words print on my chip on board : 
MEDIATEK
MT7612EN
1350-BMAL
CTPVN536

omerk, 2015-08-20:
We have a similar problem:

```
[   14.400000] ASIC revision: 76120044
[   14.430000] ROM patch already applied
[   14.440000] Firmware Version: 0.0.00
[   14.440000] Build: 1
[   14.450000] Build Time: 201406241830____
[   14.480000] Firmware running!
[   14.480000] pci device driver attached
[   14.490000] ASIC revision: 76020044
[   14.520000] ROM patch already applied
[   14.520000] Firmware Version: 0.0.00 
[   14.530000] Build: 1
(boot hangs)
```

After sprinkling some debug printk()s I have determined that this is the offending line: https://github.com/openwrt/mt76/blob/master/init.c#L383

Driver executes everything until `mt76_set(dev, 0x530, 0xf)`  and fails straight at that point. Commenting that out causes the driver to fail at a later point. Using Mediatek's `mt76x2e` driver causes the same problem, though, so I don't think `mt76` is solely responsible for this. 

A chat with Mediatek suggests that this might be a power issue but on the boards that fail we haven't identified any problems on the power rails. As for the chip revision, all the boards that fail are Revision D.

@nbd168 Have you seen this behaviour before?

nbd168, 2015-08-27:
I've seen similar issues in cases where the PCI device wasn't properly reset over a reboot. Either way, I'm pretty sure this is not an mt76 bug - it's probably either a hardware issue or a kernel PCI initialization issue.

closed 2015-08-27 by nbd168