openwrt/mt76 #8: mt76 sometimes panics the kernel

sodz, 2015-07-17:
mt76 sometimes panics the kernel, and when it happens, it ususally causes consecutive reboots because the same issue occurs few hundreds of seconds after the first few reboots. the following is the crashlog.

```
<0>[90064.190000] skbuff: skb_over_panic: text:872d0998 len:1832 put:1832 head:872e5800 data:872e5820 tail:0x872e5f48 end:0x872e5f40 dev:<NULL>
<4>[90064.210000] Kernel bug detected[#1]:
<4>[90064.210000] CPU: 0 PID: 0 Comm: swapper Not tainted 3.18.17 #1
<4>[90064.210000] task: 8032ec90 ti: 80328000 task.ti: 80328000
<4>[90064.210000] $ 0   : 00000000 00000001 0000007d 00000000
<4>[90064.210000] $ 4   : 8032db74 8032db74 8032de30 00000000
<4>[90064.210000] $ 8   : 3e4c4c55 303a646e 32373878 34663565
<4>[90064.210000] $12   : 00000000 00000000 00000000 770dc320
<4>[90064.210000] $16   : 878650c0 87259894 87259720 00000000
<4>[90064.210000] $20   : 000002f0 872e5800 00000040 87259850
<4>[90064.210000] $24   : 00000000 80191b9c
<4>[90064.210000] $28   : 80328000 80329cb0 00000728 801e388c
<4>[90064.210000] Hi    : 00000000
<4>[90064.210000] Lo    : ec4e4000
<4>[90064.210000] epc   : 801e388c skb_panic+0x58/0x5c
<4>[90064.210000]     Not tainted
<4>[90064.210000] ra    : 801e388c skb_panic+0x58/0x5c
<4>[90064.210000] Status: 1100f403      KERNEL EXL IE
<4>[90064.210000] Cause : 50800024
<4>[90064.210000] PrId  : 00019650 (MIPS 24KEc)
<4>[90064.210000] Modules linked in: pppoe ppp_async iptable_nat rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nfnetlink nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress mt76pci mac80211 cfg80211 compat ledtrig_usbdev xt_LED ledtrig_heartbeat ledtrig_gpio ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables bonding tun dm_crypt dm_mirror dm_region_hash dm_log dm_mod ipv6 eeprom_93cx6 chainiv eseqiv xts crypto_wq algif_skcipher algif_hash af_alg krng rng gf128mul cbc arc4 crypto_blkcipher usb_storage sd_mod scsi_mod ext4 jbd2 mbcache crc16 crypto_hash leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
<4>[90064.210000] Process swapper (pid: 0, threadinfo=80328000, task=8032ec90, tls=00000000)
<4>[90064.210000] Stack : 00000001 80306630 872d0998 00000728 00000728 872e5800 872e5820 872e5f48
<4>[90064.210000]         872e5f40 802fa8a8 ffffffff 801e38d8 00000004 00000000 ff000100 87259de8
<4>[90064.210000]         87259894 872d0998 00000101 00000000 00000000 00000000 87259864 87259858
<4>[90064.210000]         87259de8 8032e860 87259720 00000040 00000040 8032e868 80330000 80330000
<4>[90064.210000]         80301508 872d0ae0 00000000 00000000 87259720 87259894 87259720 87259de8
<4>[90064.210000]         ...
<4>[90064.210000] Call Trace:
<4>[90064.210000] [<801e388c>] skb_panic+0x58/0x5c
<4>[90064.210000] [<801e38d8>] skb_put+0x48/0x50
<4>[90064.210000] [<872d0998>] mt76_kick_queue+0x2a8/0x434 [mt76pci]
<4>[90064.210000]
<4>[90064.210000]
<4>[90064.210000] Code: 24841178  0c011e4e  00603821 <000c000d> 8c830060  8c8200a0  0003182b  00030336  8c87005c
<4>[90064.480000] ---[ end trace 2332bd0dddbb8cc6 ]---

===================================
```

nbd168, 2015-07-27:
fixed in the current version

closed 2015-07-27 by nbd168

sodz, 2015-07-28:
Yes, skb length validation has fixed this issue. Thanks.