openwrt/mt76 #9: [Q] Support for LED(s)

pepe2k, 2015-07-19:
Hello,

I have a device based on MT7621AT with MT7602EN and MT7612EN. I have almost finished support for it in OpenWrt, but I don't know how to control LEDs for both radios.

They are connected directly to pin 50 on both MTK radio chips (I have found that it's GPIO/LEDA I/O, at least in MT7612E, I wasn't able to find information about second chip) and are "active in low" (connected to VCC over current limiting resistor).

How does it look with that kind of setup, is control of LED/GPIO I/Os supported (or planned) in mt76 driver (I wasn't able to find anything specific in sources)?

Thanks for any advice!

new-joiner, 2015-12-07:
it seems inside MTK SDK, led was enabled directly inside the driver itself. see attached patch file

chrismonks, 2016-06-07:
Any update on this, the patch file is an invalid link.

nbd168, 2017-10-27:
mt76 now has proper mac80211 LED support for 7612 and 7603

closed 2017-10-27 by nbd168