openwrt/mt76 #15: Channel 149 and 161 Connection Timeouts

dakman, 2015-12-03:
When in client mode (sta) and when connecting to ASUS RT-AC56U Router, and when broadcasting a 5G 802.11ac access point on 161 or 149 (VHT80) MT7612E/ MT76 fails to connect. 
The only way to use 149 or 161 is to force the AC56U to transmit on VHT20/VHT40 . All other channels work great:

36 (VHT 20/40/80 OK)
40 (VHT 20/40/80 OK)
44 (VHT 20/40/80 OK)
48 (VHT 20/40/80 OK)
**149 (VHT 20/40 ONLY)**
153 (VHT 20/40/80 OK)
157 (VHT 20/40/80 OK)
161  (VHT 20/40 ONLY)
**165 (VHT 20/40/80 OK)**

[  649.030000] wlan0: associate with 14:dd:a9:99:c0:ac (try 1/3)
[  649.240000] wlan0: associate with 14:dd:a9:99:c0:ac (try 2/3)
[  649.450000] wlan0: associate with 14:dd:a9:99:c0:ac (try 3/3)
[  649.660000] wlan0: association with 14:dd:a9:99:c0:ac timed out
[  652.050000] wlan0: authenticate with 14:dd:a9:99:c0:ac
[  652.440000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 1/3)
[  652.650000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 2/3)
[  652.860000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 3/3)
[  653.070000] wlan0: authentication with 14:dd:a9:99:c0:ac timed out
[  655.860000] wlan0: authenticate with 14:dd:a9:99:c0:ac
[  656.250000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 1/3)
[  656.460000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 2/3)
[  656.670000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 3/3)
[  656.880000] wlan0: authentication with 14:dd:a9:99:c0:ac timed out
[  660.170000] wlan0: authenticate with 14:dd:a9:99:c0:ac
[  660.560000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 1/3)
[  660.770000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 2/3)
[  660.980000] wlan0: send auth to 14:dd:a9:99:c0:ac (try 3/3)
[  661.190000] wlan0: authentication with 14:dd:a9:99:c0:ac timed out

barhom, 2018-02-10:
is this still a problem?
if not then close the ticket.

barhom, 2018-07-08:
Close this ticket.

barhom, 2018-07-08:
@nbd168 @LorenzoBianconi 
I would like to help you clean up this issue tracker.
If issues are this old and no one have written anything in years then it should be closed.

closed 2019-01-31 by nbd168