openwrt/mt76 #16: PA disabled on Xiaomi mini

psyborg55, 2015-12-20:
all test so far indicate that power amplifier on this device doesn't function.

with stock fw power level is between 80 and 90% on ubuntu notebook but with openwrt about 50-60% if the power isn't raised in chip directly with https://github.com/openwrt/mt76/commit/1413f553f866e9ae53871013294c25ff85588cf3

the information available in MT7612E datasheet suggest to enable GPIO 13 and 14 in IO_MODE 7 for PA to work.

can someone implement this in driver code?

nbd168, 2015-12-20:
What do you mean by 'if the power isn't raised in chip directly with 1413f55' - if you change that part of the code, tx power increases with mt76?
Please post the output of cat /sys/kernel/debug/ieee80211/phy*/mt76/txpower

psyborg55, 2015-12-20:
yes with that commit txpower increases but only a half compared to stock fw.

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy*/mt76/txpower
cat: can't open '/sys/kernel/debug/ieee80211/phy*/mt76/txpower': No such file or directory

ignisf, 2015-12-20:
@psyborg55 it's `/sys/kernel/debug/ieee80211/phy<insert number of phy here>/mt76/txpower`

psyborg55, 2015-12-20:
i know it's just site formatting removed the star for some reason, there's no debug dir at all.

nbd168, 2015-12-22:
What OpenWrt version are you using?

nbd168, 2015-12-22:
And did you disable CONFIG_PACKAGE_MAC80211_DEBUGFS for some reason?

psyborg55, 2016-01-01:
i'll try recompiling with debugfs enabled, but not now because i want to test device stability. currently its uptime is more than 8 days.

psyborg55, 2016-01-09:
ok, here you go:

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy*/mt76/txpower
Target power: 34
     Delta: -2 -2
       CCK:  3  3  3  3
      OFDM:  2  2  2  2  2  2  0  0
        HT:  2  2  2  2  2  2 -1 -1  2  2  2  2  2  2 -1 -1
       VHT:  2  2  2  2 -2 -2 -2 -2  0  0

psyborg55, 2016-01-09:
comment from padavan on site: http://4pda.ru/forum/index.php?showtopic=596689&st=4700

W) PA / LNA 
Each chip has a WiFi radio transmitter and receiver. To send a signal it must be reinforced through PowerAmplifier (PA) to the level of 20dBm (100mW), the levels vary with the laws of each region. In some WiFi PA chip is embedded in the device may be diluted with a built-like (iPA), and external (ePA). External PA are always well received signal amplifier Low Noise Amplifier (LNA), so if there is ePA, the circuit ePA / eLNA. If you are using a built-iPA, it can be applied iLNA or external chip eLNA. 
On a subject: 
- 2.4 radio wired as iPA / eLNA + T / R switch 
- Radio 5 wired as iPA / iLNA (T / R switch and built-in) 

When placed external chip LNA, the input from the WiFi chip should be muted (normal gain LNA 12dB, depending on the characteristics of the particular eLNA), otherwise the LNA will supply high level, which will be distorted. For this purpose the EEPROM registers parameter "External LNA Gain", which tells how much dB you need to reduce the sensitivity of the input to which a signal from the LNA. 
When using the internal LNA, it is common practice "External LNA Gain" is not filled in, ie It is the nominal sensitivity. On sabzh factory in EEPROM 5GHz hammered 2dB, it decreases the value of "hearing" at radio, ie the worse the hearing, the worse the AP hearing the client at his disposal and greatly reduces the speed of the link.

psyborg55, 2016-05-19:
iw list outputs:

RX STBC 1-stream for 5GHz

could it be that the driver doesn't set txpower properly because of incorrect radio capabilites?

i know output power should be higher when 2 streams are used, maybe everything else is fine after all but power is not at 100% because the driver assumes it works with single chain only?

nbd168, 2016-05-23:
The capabilities have nothing to do with this. 'RX STBC 1-stream' does not imply that the device is limited to 1 stream only.

psyborg55, 2016-09-04:
i've managed to increase signal strength for about 5-7dB but the tx rate drops to 90Mbps and affects throughput. is there some temperature compensation code needed to change in order to maintain tx rate?

this is debug output with https://github.com/openwrt/mt76/commit/1413f553f866e9ae53871013294c25ff85588cf3 revert:

> root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy*/mt76/txpower
Target power: 34
     Delta: 37 30
       CCK: -41 -41 -41 -41
      OFDM: -41 -41 -41 -41 -41 -41 -41 -41
        HT: -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41 -41
       VHT: -41 -41 -41 -41 -41 -41 -41 -41 -41 -41

psyborg55, 2017-08-13:
very good signal strength measurements were done by gazambuja: https://forum.openwrt.org/viewtopic.php?pid=319519#p319519

Stock firmware from Xiaomi
2.4 --> -35 / -38
5   --> -39 / -42
Stable version - CC 15.05.01
2.4 --> -57 / -61
5   --> -55 / -58
Trunk - some commit with kernel 4.3 - http://git.openwrt.org/openwrt.git?p=op … ce6ea6df9c
2.4 --> -39 / -56
5   --> -47 / -54
Trunk r49161 (from few days ago... unstable on 2.4G)
2.4 --> -36 / -41
5   --> -48 / -51
Fork of changeway - https://github.com/changeway/openwrt_CC_for_MiWifi_Mini
2.4 --> -34 / -36
5   --> -45 / -47

shawnchain, 2018-01-04:
Same issue here. @nbd168 
I have 2 MT7620 board, on with external PA/LNA while another is not. 
I would like to know if the latest MT76 driver could handle those 2 different board design ?  Or what should I tweak on the driver configurations ?

Thanks,
Shawn

nbd168, 2019-01-26:
Please try the latest version