openwrt/mt76 #17: wrong signal-avg values with 'iw ... station get'

bittorf, 2015-12-30:
signal avg looks wrong. tested with OpenWrt r48014

```
root@xiaomi-mini:~ iw dev wlan0 station get 10:6f:3f:0e:31:8f
Station 10:6f:3f:0e:31:8f (on wlan0)
        inactive time:  120 ms
        rx bytes:       37767883
        rx packets:     161734
        tx bytes:       258015651
        tx packets:     187789
        tx retries:     4589
        tx failed:      0
        signal:         75 [72, 75] dBm
        signal avg:     0 [-3, 0] dBm
        tx bitrate:     54.0 MBit/s
        rx bitrate:     54.0 MBit/s
        expected throughput:    3.808Mbps
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        connected time: 6191 seconds
```

psyborg55, 2016-01-01:
expected throughput too low also

	inactive time:	10 ms
	rx bytes:	7392131
	rx packets:	59432
	tx bytes:	121123183
	tx packets:	96594
	tx retries:	2732
	tx failed:	1
	signal:  	59 [59, 56] dBm
	signal avg:	-8 [-8, 11] dBm
	tx bitrate:	300.0 MBit/s MCS 15 40MHz short GI
	rx bitrate:	216.0 MBit/s MCS 13 40MHz
	expected throughput:	5.859Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	1714 seconds

bittorf, 2016-01-03:
@psyborg55 the "expected throughput:" can be correct. you dont know the circumstances and i did not provide e.g. rate-control data, so let's ignore this for now.

psyborg55, 2016-01-03:
ok, but you can check that easily:

root@OpenWrt:~# iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.5.1 port 5001 connected with 192.168.5.2 port 53408
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.0 sec  88.5 MBytes  74.1 Mbits/sec
[  4] local 192.168.5.1 port 5001 connected with 192.168.5.2 port 53411
[  4]  0.0-10.0 sec  89.4 MBytes  74.8 Mbits/sec

bittorf, 2016-01-04:
@psyborg55 i know how to check that. what i said was: the value seems ok and is caused be my local circumstances (e.g. rate-control)

psyborg55, 2016-01-08:
it is too low even in your case. at 54Mbps you should expect at least 20Mbps of throughput and i'm sure you could measure a lot more than 3.8Mbps

bittorf, 2016-01-08:
@psyborg55 this is really another issue. i will explain: every interface which has not enough traffic, will show a low estimate througput. thats the nature of the used algo. it has nothing to do with the issue and should not be discussed here.

psyborg55, 2016-01-08:
you got something wrong, it's clearly incorrect with this driver, read the comment: https://github.com/openwrt/luci/issues/540#issuecomment-170049249

while it may not have anything to do with the original issue, let's not exclude the possibility both bugs might be related to the same cause.

bittorf, 2016-01-09:
ok, i described the wrong throughput-values at #18

psyborg55, 2016-01-14:
the patch at https://dev.openwrt.org/ticket/20343#comment:17 fixes this bug

nbd168, 2016-01-14:
fix committed and pushed

closed 2016-01-14 by nbd168