openwrt/mt76 #19: missing Efuse defines

psyborg55, 2016-01-18:
there is no defined EXT_EE_SEL pin in eeprom.h so no way to choose between Efuse/EEPROM

nbd168, 2016-01-18:
There is no support for an external EEPROM in the driver, so I think such a pin check would be rather pointless. Do you know of any device that even uses an externally connected EEPROM on a mt76x2 chip?

psyborg55, 2016-01-18:
I thought EEPROM refers to data read from factory partition. According to datasheet: EXT_EE_SEL is used to identify if the external EEPROM or the internal Efuse is used.

Since the OP in this topic https://forum.openwrt.org/viewtopic.php?id=61250 hasn't mentioned anything about 5GHz wifi I made a conclusion original miwifi firmware uses Efuse for both 2.4 and 5 interfaces rather than EEPROM from factory partition, which would explain how they got 5GHz signal almost strong as 2.4GHz.

Also, a blown Efuse would probably make the device read calibration data from EEPROM which would explain why I can't no longer get strong signal on 5GHz with miwifi firmware, as described here: https://forum.openwrt.org/viewtopic.php?pid=307997#p307997

nbd168, 2016-01-18:
Choosing between calibration data from flash and using efuse does not require any changes to the EEPROM code. If the calibration data pointer is set up in the device tree file, the driver will use the data from flash. If not, it will use efuse. There is also a mixed 'calibration free' mode, which uses the data from flash as template and pulls in overrides from efuse. All of these modes are supported, and the efuse override mode is used by default, if the data contains suitable values.
If you want to try using only efuse and not the factory partition, you can simply remove it from the device tree file and rebuild.

closed 2016-01-18 by nbd168

psyborg55, 2016-10-21:
with efuse only wifi comes up but very low power, works within a range of meter or two.

since the calfree mode is used by default that must mean both eeprom and efuse are enabled. how could one disable efuse completely?

i've tried dumping this data with dd if=/sys/kernel/debug/ieee80211/phy0/mt76/otp of=/tmp/otp but can't understand it all. 50-9F seems to be holding power values but i have no idea about other bytes there.