openwrt/mt76 #20: wrong 'signal' / 'signal avg:' values on r48616

bittorf, 2016-02-03:
this must be wrong:
the other side is a Linksys WRT54G

{{{
# r48616 on Xiaomi Miwifi mini
root@xiaomi-mini2:~ iw dev wlan1 station get 00:13:10:14:c4:fe
Station 00:13:10:14:c4:fe (on wlan1)
        inactive time:  110 ms
        rx bytes:       122899664
        rx packets:     778197
        tx bytes:       344
        tx packets:     4
        tx retries:     1
        tx failed:      0
        signal:         -101 dBm
        signal avg:     -102 dBm
        tx bitrate:     1.0 MBit/s
        rx bitrate:     6.0 MBit/s
        expected throughput:    0.82Mbps
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        connected time: 39194 seconds
}}}

OLSR-values indicate, that the connection is good. the linksys says this:

{{{
root@linksys:/# iw dev wlan0 station get 64:09:80:7f:48:85
Station 64:09:80:7f:48:85 (on wlan0)
        inactive time:  220 ms
        rx bytes:       194398378
        rx packets:     149588
        tx bytes:       2258
        tx packets:     27
        tx retries:     4
        tx failed:      0
        signal:         -65 dBm
        signal avg:     -64 dBm
        tx bitrate:     6.0 MBit/s
        rx bitrate:     1.0 MBit/s
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
}}}

bittorf, 2016-02-03:
this seems to happen with g-stations, here a TP-LINK 1043ND:

```
root@xiaomi-mini2:~ cat /sys/kernel/debug/ieee80211/phy1/netdev:wlan1/stations/f4:ec:38:9d:7a:1e/rc_stats

best   __________rate_________    ________statistics________    ________last_______    ______sum-of________
rate  [name idx airtime max_tp]  [avg(tp) avg(prob) sd(prob)]  [prob.|retry|suc|att]  [#success | #attempts]
        1     0    9738    0.9       0.9     100.0      0.0     100.0   1     0 0            73   73       
        2     1    4922    1.6       1.6     100.0      0.0     100.0   1     0 0             1   1        
        5.5   2    1858    4.7       4.7     100.0      0.0     100.0   2     0 0             1   1        
 B     11     3     982    9.1       9.1     100.0      0.0     100.0   2     0 0             1   1        
        6     4    1648    5.3       1.8      31.1      1.5       0.0   2     0 0             9   29       
        9     5    1112    8.0       0.9      11.0      1.1       0.0   2     0 0             8   56       
       12     6     844   10.5       4.1      35.7      1.2       0.0   2     0 0           122   172      
A   P  18     7     576   15.5      15.5      96.8      1.5     100.0   2     1 1          1538   1701     
   D   24     8     440   20.4       7.1      31.6      2.0       0.0   2     0 0            74   102      
  C    36     9     308   29.1       8.2      25.5      1.6       0.0   2     0 0             6   22       
       48    10     240   37.3       6.8      16.5      1.8       0.0   2     0 0             2   16       
       54    11     216    0.0       0.0       0.0      0.0       0.0   2     0 0             0   8        

Total packet count::    ideal 1802      lookaround 95

root@xiaomi-mini2:~ iw dev wlan1 station get f4:ec:38:9d:7a:1e
Station f4:ec:38:9d:7a:1e (on wlan1)
        inactive time:  60 ms
        rx bytes:       132103362
        rx packets:     410138
        tx bytes:       203192
        tx packets:     1897
        tx retries:     285
        tx failed:      0
        signal:         -99 dBm
        signal avg:     -99 dBm
        tx bitrate:     18.0 MBit/s
        rx bitrate:     24.0 MBit/s
        expected throughput:    1.418Mbps
        authorized:     yes
        authenticated:  yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        connected time: 41554 seconds

```

psyborg55, 2016-02-03:
is that 2.4G wifi?

bittorf, 2016-02-04:
@psyborg55 yes - 2.4GHz

nbd168, 2016-02-04:
The 2.4 GHz wifi on Xiaomi Miwifi mini does not use mt76

closed 2016-02-04 by nbd168

bittorf, 2016-02-04:
@nbd168 sorry, you are right.