openwrt/mt76 #21: MT76 warning

wintonliuwen, 2016-03-02:
Openwrt Version: CC
mt76: 2015-10-12
Kernel: 3.18.23
Hardware: MT7602 & MT7612

WiFi Seems work fine, the station could connect the AP. But the console shows following warning.

[ 2281.060000] WARNING: CPU: 0 PID: 0 at /home/winton/workspace/fnic/mhwtsd2000.git/build_dir/target-mipsel_1004kc+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2015-10-12/mac.c:67 mt76_mac_process_rx+0x19c/0x248 [mt76pci]()
[ 2281.080000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc7 crc_itu_t crc_ccitt mt76pci mac80211 cfg80211 compat ledtrig_usbdev xt_NFQUEUE nfnetlink_queue nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ntfs autofs4 nls_utf8 ipv6 eeprom_93cx6 arc4 crypto_blkcipher mmc_block sdhci_pltfm sdhci mtk_sd mmc_core fuse leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod gpio_button_hotplug ext4 jbd2 mbcache nls_iso8859_1 crc16 crypto_hash
[ 2281.180000] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 3.18.23 #2
[ 2281.190000] Stack : 00000000 00000004 00000006 00000002 00000000 00000000 00000000 00000000
          803e56e2 00000034 00000000 00000000 00000001 80396de0 80322d9c 80396a43
          00000000 00000000 803e4468 80396de0 8df3e000 00000040 8ecc58fc 8005a2fc
          8039af20 80029184 00000003 80328798 803269d8 80385c1c 80385c1c 80322d9c
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[ 2281.220000] Call Trace:
[ 2281.220000] [<800160a0>] show_stack+0x50/0x84
[ 2281.230000] [<80169694>] dump_stack+0x64/0x84
[ 2281.230000] [<8002928c>] warn_slowpath_common+0x84/0xb4
[ 2281.240000] [<80029344>] warn_slowpath_null+0x18/0x24
[ 2281.240000] [<8edf51f0>] mt76_mac_process_rx+0x19c/0x248 [mt76pci]
[ 2281.250000] [<8edf0ea0>] mt76_kick_queue+0x468/0x580 [mt76pci]
[ 2281.250000] 
[ 2281.260000] ---[ end trace 62f640e14bfa8dd4 ]---

noblepepper, 2016-03-11:
Also seeing this on openwrt trunk with mt7628. Steps to repeat:

1.Flash new image either through uboot or sysupgrade -n
2. Enable wifi in /etc/config/wireless, execute 'wifi'
3. Connect to OpenWrt AP
4. Use Luci to connect to PSK2 encrypted network as sta, dhcp client

Seems to repeat itself every 67 seconds, no ill effect other than console warnings. Rebooting eliminates warnings.

[  346.590454] ------------[ cut here ]------------
[  346.595339] WARNING: CPU: 0 PID: 0 at compat-wireless-2016-01-10/net/mac80211/tx.c:3724 ieee80211_csa_is_complete+0x4d8/0x66c [mac80211]()
[  346.607965] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt mt7603e mt76x2e mt76 mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  346.676919] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.4 #4
[  346.683885] Stack : 8033dbd4 00000000 00000001 80380000 80381078 80380d03 8031f5f4 00000000
	  803c342c 8037bc88 00000000 83372f30 00000001 80049f90 80324acc 80380000
	  00000003 8037bc88 80322ed0 8037bb6c 00000001 80047f48 0003aaaa 00000000
	  8037fe30 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  ...
[  346.720048] Call Trace:
[  346.722537] [<800144ec>] show_stack+0x50/0x84
[  346.726974] [<80024d54>] warn_slowpath_common+0xa0/0xd0
[  346.732279] [<80024e08>] warn_slowpath_null+0x18/0x24
[  346.737511] [<831a7074>] ieee80211_csa_is_complete+0x4d8/0x66c [mac80211]
[  346.744462] 
[  346.745990] ---[ end trace d946883541bdabe0 ]---

nbd168, 2017-12-03:
Fixed in the current version

closed 2017-12-03 by nbd168