openwrt/mt76 #23: SD card issue in MT7620

ObaidAshraf, 2016-03-18:
I have a ZBT-WE826 router with built-in SD Card slot. I am trying to mount my SD card in my Router. I've openWRT CC with kernel 3.18. I've download mmc_over_gpio and related packages.. But the problem arises, when I run "/etc/init.d/mmc_over_gpio restart", i get this error "no support for SD card's voltages" ..
There is also a ticket open in openWRT for this on, "https://dev.openwrt.org/ticket/21834" ..

How can i solve this error?? i tried multiple SD cards, but no use..

blogic, 2016-03-18:
This repo is for a wifi driver called MT76. Your bug report has nothing to do with wifi.

closed 2016-03-18 by blogic

ObaidAshraf, 2016-03-18:
Oops, Sorry .. My bad !! ..

blogic, 2016-03-18:
https://dev.openwrt.org/ would be the place to post your bug