openwrt/mt76 #26: Issue with iPhone 6S and MT7603

kristrev, 2016-03-27:
Hi,

I have a router with mt7621 chipset and an mt7603/mt7612 wifi combo. The image I am using is built from OpenWRT trunk and has kernel 4.4 + all the latest mt76 and mac80211 fixes/changes. The mt7612 (used for the 5GHz) is working perfect, but I am having some issues with the mt7603 and an iPhone 6S. There might be other devices experiencing the same problems, but I dont have any available.

When I initially connect the iPhone to the 2.4 GHz wifi (i.e., the mt7603) everything works fine. However, when the phone has been idle for a while and tries to reconnect, something goes wrong. When looking at the settings on the phone, I see that it has acquired an IP and this seems to match the output in syslog. However, no traffic goes through. It seems that the traffic sent from the phone is dropped somewhere. I also see the following warning in dmesg (not sure if it is connected or not):

```
[71401.890000] WARNING: CPU: 0 PID: 12201 at /home/kristrev/src/openwrt/trunk-git/openwrt-2/build_dir/target-mipsel_1004kc+dsp_musl-1.1.14/linux-ramips_mt7621/mt76-2016-03-04/mt7603_mac.c:1202 mt7603_mac_work+0x204/0x280 [mt7603e]()
[71401.910000] Modules linked in: rtl8192cu qcserial ppp_async option iptable_nat usb_wwan rtl_usb rtl8192c_common rt2800usb rt2800pci rt2800mmio rt2800lib rndis_host qmi_wwan ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cdc_ncm cdc_ether ax88179_178a asix xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nfacct xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet ts_kmp ts_fsm ts_bm slhc rtlwifi rt2x00usb rt2x00pci rt2x00mmio rt2x00lib nfnetlink_queue nfnetlink_acct nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast iptable_rawpost iptable_raw iptable_mangle iptable_filter ipt_ECN ipheth ip_tables ip6table_rawpost crc_itu_t crc_ccitt compat_xtables cdc_wdm cdc_acm ath9k_htc ath9k ath9k_common ath9k_hw act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress ath10k_pci ath10k_core ath mt7603e mt76x2e mt76 mac80211 cfg80211 compat ledtrig_usbdev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ifb eeprom_93cx6 leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common mii aead crypto_null crypto_hash
[71402.110000] CPU: 0 PID: 12201 Comm: kworker/u8:1 Tainted: G        W       4.4.6 #1
[71402.110000] Workqueue: phy1 mt7603_mac_work [mt7603e]
[71402.110000] Stack : 80450000 804b0000 8e775e00 803c9b08 8ee662b8 000004b2 8ee651d4 0000000a
[71402.110000] 	  8f804614 8e775e18 00000088 80062fdc 00002fa9 8049365c 8f804614 8e775e18
[71402.110000] 	  803e53fc 8da77d5c 803c9b08 80060d64 80450000 804b0000 803ce3b8 8da77d74
[71402.110000] 	  8da77d5c 8e775e00 803c9b08 800409d8 8148dd7c 8da77d68 0000083b 00000000
[71402.110000] 	  00000000 00000000 00000000 00000000 00000000 00000000 31796870 00000000
[71402.110000] 	  ...
[71402.110000] Call Trace:
[71402.110000] [<800165d0>] show_stack+0x50/0x84
[71402.110000] [<801aada0>] dump_stack+0x84/0xbc
[71402.110000] [<8002baf0>] warn_slowpath_common+0xa0/0xd0
[71402.110000] [<8002bba4>] warn_slowpath_null+0x18/0x24
[71402.110000] [<8ee651d4>] mt7603_mac_work+0x204/0x280 [mt7603e]
[71402.110000] [<8003e650>] process_one_work+0x214/0x358
[71402.110000] [<8003f538>] worker_thread+0x2dc/0x444
[71402.110000] [<80043c3c>] kthread+0xdc/0xe8
[71402.110000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[71402.110000] 
[71402.210000] ---[ end trace a3ba53e46e43160e ]---
```

This error does not happen every time the device has gone into idle mode. I unfortunately do not have access to any device where I can sniff the wifi traffic, so I won't be able to provide any of those details. I hope this issue can serve as a starting point for nailing down this bug. I have tested the 2.4 wifi network with other devices too, like my laptop, and do not see the issue.

Thanks in advance for any help.

kristrev, 2016-03-29:
I was a bit wrong in my initial report. The iPhone does not send a DHCP request, it just keeps the IP it had before and it seems that L2 is OK (since the phone is never disconnected from the wifi). However, no traffic originating from the phone is received at the router (at least not visible by tcpdump).

kristrev, 2016-03-29:
So I did some more looking and I do see the traffic on the router using tcpdump. When trying to ping the router, I see both the request and reply. Howver, the request never arrives on the phone. I do not see any kernel oops in the log, so that is probably unrelated. When looking at the packets in Wireshark, all looks fine. I.e., checksums etc are looking good.

After a random time, however, I start receiving traffic again (on the phone). This seems to happen at the same time as my device is kicked from the AP.

gareth41, 2016-04-09:
Are you using any encryption, WPA/WPA2 or EAP? or open network?

I got a ZBT-WG3526 router with same wifi chips mt7603 mt7612 and using an iphone, I haven't noticed this problem.  I am however having a problem using WPA2-EAP and get a very similar error in console.

kristrev, 2016-04-09:
Yes, I am by default running with encryption (WPA2 + AES). However, I have tested disabling encryption altogether (as well as switching to WPA) and see the same result.

kristrev, 2016-07-20:
I had some time to do some more testing on this issue. I see the issue with different Macbooks (Air, Pro). I guess this smells like an iOs/OS X issue? Wifi seems to be stable on for example my Lenovo laptop.

nbd168, 2016-09-05:
Please try the latest version of LEDE

kristrev, 2016-11-02:
Hi,

Sorry for my late reply. I was finally able to test this and while it seems to work better than before (and no kernel error is to be seen), I unfortunately still see the same behaviour with both a 6s, 7 Plus and Macbook. However, now it seems to be a bit random if it happens or not.

The only thing I see in logs when device wakes up again is authenticated/associated/radius/wpa from hostapd and DHCP Request/ACK. Then device always seems to work fine for a few seconds, before it in most cases suddenly stop working

nbd168, 2016-11-03:
Please try latest LEDE, it should be more stable now

iancd, 2017-04-22:
Hi. I'm seeing the same problem with the latest LEDE snapshot (3978) 2.4Ghz psk2+ccmp. Problem happens with my iPhone 6S/iOS 10.3.1 when the phone has been idle for a while.  Let me know if I can help with testing. Same symptoms as @kristrev and nothing notable in the logs.

```iw wlan0 info
Interface wlan0
	ifindex 15
	wdev 0x2
	addr <removed>
	ssid mblan2.4
	type AP
	wiphy 0
	channel 3 (2422 MHz), width: 20 MHz, center1: 2422 MHz```
```
```
Linux version 4.4.61 (buildbot@buildslave) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3978-e209988) ) #0 SMP Tue Apr 18 10:00:51 2017
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
```

nbd168, 2017-12-03:
Should be fixed in the current version

iancd, 2017-12-05:
Seeing same problem.
I have tested with the latest December 5th LEDE snapshot 2.4Ghz psk2+ccmp. This time with 2 iPhone 6S in my house with iOS 11.2. After not using my phone for 5 mins, I woke up the phone, noticed that WiFi status showed full connectivity but there was no IP traffic allowed but nothing showing in the logs. 10mins later I woke the phone and found connectivity was normal and then immediately checked the logs to find that iPhone 6S was dissociated and re-associated due to inactivity. It does seem to having a problem with power saving modes. Please let me know if I can provide more specific tests or logs?

`LEDE Reboot SNAPSHOT r5447-e23ff06 / LuCI Master (git-17.338.50622-d2bbfa5)
SoC Type: MediaTek MT7621
`

iancd, 2017-12-15:
@nbd168 I performed some more tests using latest snapshot. Problem is easy to reproduce for me. I connect iPhone 6S to 7603 WiFi then sleep the phone for a minute. After waking the phone shows connection to WiFi with full bars but there's no connectivity. There's also no events shown in the logs.
If I sleep the phone for longer periods or turn wifi off and back on later, the phone shows as disassociated in the logs and the phone operates normally.

`Linux version 4.9.67 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5484-69d22a6
mt7603e 0000:02:00.0: Build Time: 20160107100755
SoC Type: MediaTek MT7621`

iancd, 2018-01-23:
@nbd168 Is there anything I can do to help with testing fixes for this issue?

lukasz1992, 2018-01-26:
@icand Maybe put another WiFi card into monitor mode and dump traffic?
But just guessing.

barhom, 2018-02-05:
Having the same issue on Galaxy S8 phone here on2.4ghz.

1. Connect to wifi, everything is normal
2. Sleep the Samsung Galaxy S8 for a while.
3. Wake up to find that you are still connected to wifi but the samsung is showing the "No internet" sign on the bottom right of the wifi icon that lets you know that you are connected but have no internet.

Only solution is to turn off wifi and back on. Logs show something like;

> Mon Feb  5 00:03:42 2018 daemon.info hostapd: wlan1: STA e0:f8:47:29:b3:5c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb  5 00:04:00 2018 cron.info crond[1043]: USER root pid 24496 cmd php /routerguard/www/app/artisan schedule:run >> /dev/null 2>&1

iancd, 2018-02-06:
I have identical behaviour to @barhom with my iPhone 6S/iOS v11.2.5 with the latest mt76 drivers

```
Tue Feb  6 20:33:49 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Tue Feb  6 20:33:50 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Tue Feb  6 20:33:50 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Tue Feb  6 20:33:50 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Tue Feb  6 20:33:50 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Tue Feb  6 20:33:50 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Tue Feb  6 20:33:51 2018 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

`Linux version 4.9.77 (buildbot@builds) (gcc version 5.5.0 (OpenWrt GCC 5.5.0 r5996-256477f) ) #0 SMP Mon Feb 5 11:58:27 2018
Tue Feb  6 20:04:35 2018 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3`

iancd, 2018-02-11:
@kristrev I've had some success with the latest lede snapshot build and adding disassoc_low_ack='0' to my config. Does this fix the issue for your iPhone?
`uci set wireless.default_radio0.disassoc_low_ack='0'`

```
#>uci show wireless.default_radio0
wireless.default_radio0=wifi-iface
wireless.default_radio0.device='radio0'
wireless.default_radio0.network='lan'
wireless.default_radio0.mode='ap'
wireless.default_radio0.encryption='psk2+ccmp'
wireless.default_radio0.key='xxx'
wireless.default_radio0.ssid='mblan'
wireless.default_radio0.disassoc_low_ack='0'
```

kristrev, 2018-02-11:
Hi,

Sorry for not following up in this issue. The reason I haven't been paying attention, is that I don't have problems with my iPhone (or macOS-devices) any more. I was going to close the issue, but then I saw lots of other people also reported problems so I left it open.

iancd, 2018-02-12:
@kristrev Thanks. Are your Apple devices now using 5G? In Luci > Network > Devices check your devices to see if they are wlan1 or wlan0 which corresponds to the radio with 802.11nac. Apple changed the behaviour of their wifi to prefer 5G even when it has lower signal than 2G so I was wondering this may have been to your benefit?
Also are you using the disassoc_low_ack option?
`uci show wireless.default_radio0.disassoc_low_ack`

dissent1, 2018-02-12:
It has to be in wifi-iface section
uci set wireless.@wifi-iface[0].disassoc_low_ack=‘0’ && uci commit

camelcamro, 2018-02-12:
well, i also tried newest trunk ( 12.2.2018)
and as i had some guests with iphones .... got a lot of kernel alarms ...

seems to be something very buggy :(

 `Mon Feb 12 07:29:58 2018 kern.warn kernel: [35713.818889] ------------[ cut here ]------------
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35713.823685] WARNING: CPU: 0 PID: 0 at backports-2017-11-01/net/mac80211/rx.c:4316 ieee80211_rx_napi+0x1a4/0x964 [mac80211]
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35713.834723] Rate marked as a VHT rate but data is invalid: MCS: 65, NSS: 0
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35713.841592] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35713.913011]  nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat f2fs ext4 jbd2 mbcache nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb crc32c_generic crc32_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_coMon Feb 12 07:29:58 2018 kern.warn kernel: [35714.008631] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.77 #0
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.014519] Stack : 00000000 00000000 80557b4a 00000033 8040fc04 00000000 00000000 80550000
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.022864]         804f62bc 804f5ea7 8048cd04 00000000 00000000 80553824 00000064 00000980
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.031207]         8ee165d0 8006ab98 00000001 80550000 804fc004 804fc008 80491910 8fc0dc7c
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.039550]         00000003 800a7a50 00000064 00000980 8ee165d0 00000000 00000002 00c0dc7c
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.047892]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.056232]         ...
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.058669] Call Trace:
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.061129] [<8000f714>] show_stack+0x54/0x88
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.065470] [<801e5d9c>] dump_stack+0x8c/0xd0
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.069810] [<8002adc4>] __warn+0xe4/0x118
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.073885] [<8002ae28>] warn_slowpath_fmt+0x30/0x3c
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.078885] [<8eda5c7c>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.084987] [<8ed41b78>] mt76_rx_complete+0x18c/0x278 [mt76]
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.090625] [<8ed41e24>] mt76_rx_poll_complete+0x1c0/0x260 [mt76]
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.096693] [<8ed40e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
Mon Feb 12 07:29:58 2018 kern.warn kernel: [35714.102303] ---[ end trace f2bcb7463a1e4095 ]---

Mon Feb 12 08:08:28 2018 kern.warn kernel: [ 1637.588715] ------------[ cut here ]------------
Mon Feb 12 08:08:28 2018 kern.warn kernel: [ 1637.593399] WARNING: CPU: 0 PID: 0 at /var/lib/buildbot/slaves/tictex-02/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-02-09-246d548b/mt7603_mac.c:441 mt7603_mac_fill_rx+0x178/0x520 [mt7603e]
Mon Feb 12 08:08:28 2018 kern.warn kernel: [ 1637.613672] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 [... cut ...]
Mon Feb 12 09:04:47 2018 kern.warn kernel: [ 1119.564283] [<8eda5c7c>] ieee80211_rx_napi+0x1a4/0x964 [mac80211]
Mon Feb 12 09:04:47 2018 kern.warn kernel: [ 1119.570385] [<8ecd9b78>] mt76_rx_complete+0x18c/0x278 [mt76]
Mon Feb 12 09:04:47 2018 kern.warn kernel: [ 1119.576023] [<8ecd9e24>] mt76_rx_poll_complete+0x1c0/0x260 [mt76]
Mon Feb 12 09:04:47 2018 kern.warn kernel: [ 1119.582091] [<8ecd8e64>] mt76_dma_attach+0xb94/0xcc0 [mt76]
Mon Feb 12 09:04:47 2018 kern.warn kernel: [ 1119.587708] ---[ end trace 00eb0002eb54be5e ]---

`

lukasz1992, 2018-02-12:
Assuming the WARNs come from mt7603 RX handling, I saw one thing:
RX_FLAG_VHT must be set, RX_FLAG_HT must not be set  - so I assume sband was set to 5 GHz.

Actually 5 GHz is selected if MT_RXD1_NORMAL_CH_FREQ is odd, here:
```
i = FIELD_GET(MT_RXD1_NORMAL_CH_FREQ, rxd1);
sband = (i & 1) ? &dev->mt76.sband_5g.sband : &dev->mt76.sband_2g.sband;
```

@nbd168 isn't MT7603 2GHz only? The last bit must mean something other than band selection. sband_5g for MT7603 driver is not initialized and could have random memory (not sure if this happens in kernel memory, or is it nulled?) - random memory could be reason of ieee80211_rx_napi warnings.
This also explains (!sband->channels) WARNs. What do you think about it?

I could understand "Rate marked as an HT rate but passed" warning here.

slthomason, 2018-07-09:
> @nbd168 <https://github.com/nbd168> @LorenzoBianconi
> I would like to help you clean up this issue tracker.
> This issue has most probably been fixed in recent patches.

I think this one is related to this one, unless someone can specify
otherwise: https://github.com/openwrt/mt76/issues/167

barhom, 2018-08-12:
Is this still happening?
There has been a lot of improvement lately.
Close if this doesnt happen anymore.

kristrev, 2018-08-21:
Yes, this is still very much an issue. I ended up writing a small tool which checks for iPhones/iPads on my network and kicks the client (using the "del_client" ubus command) if the device does not reply to ICMP within a given interval.

lukasz1992, 2018-08-25:
Anything better with https://github.com/openwrt/mt76/commit/980c60666eb57daa144d8725712ba303e876e677 or https://github.com/openwrt/mt76/commit/30b8371ad7233c04fa825f318954791d3796c2bf

kristrev, 2018-08-27:
I will try to make som tests with the lates version during the day today

Albis-dev, 2019-01-04:
I'm using iPhone 7 (A1778, intel modem) and can confirm the issue still persists. Here's my setup now.

`Firmware Version  OpenWrt SNAPSHOT r8982-a54129d / LuCI Master (git-19.002.70367-c7776c8) ` 
`
> opkg list-installed
...
kmod-mac80211 - 4.14.90+4.19.7-1-1
kmod-mt76 - 4.14.90+2018-12-28-3598046d-1
kmod-mt76-core - 4.14.90+2018-12-28-3598046d-1
kmod-mt7603 - 4.14.90+2018-12-28-3598046d-1
...
`

However, using "legacy" (not "n") completely solves the problem. It's been a day since I changed the MT7603 radio option, haven't noticed a single connectivity issue/disconnection.

Not sure if I can actually help something but gonna look into some codes. If anyone needs extra information from me, please reply.

Albis-dev, 2019-01-21:
I think It's safe to say this issue has been resolved. https://github.com/openwrt/mt76/commit/fc31457cd99cb85c8cea9329eedc5edd80038f29 works perfect. 

Lack of SMPS support was the issue, which was introduced in 802.11n. Everything makes sense now. 

Thanks @nbd168 !

nbd168, 2019-01-21:
Thanks for testing

closed 2019-01-21 by nbd168