openwrt/mt76 #28: (mt7603/76x8) Why do we apply cal free data when calibration exists?

noblepepper, 2016-04-11:
When using on mt7628 or mt7688 devices cal free data is applied after loading the calibration data from the factory partition.

This greatly reduced throughput and caused frequent re-associations on some boards. Commenting out the cal_free_bytes list in mt7603_apply_cal_free_data gave almost 40x improvement in throughput and eliminated the re-associations. This is obviously not the correct solution since it would break boards without calibration partitions, just evidence this is the source of the issue.
{
        static const u8 cal_free_bytes[] = {
                /*MT_EE_TEMP_SENSOR_CAL,
                MT_EE_TX_POWER_0_START_2G,
                MT_EE_TX_POWER_0_START_2G + 1,
                MT_EE_TX_POWER_1_START_2G,
                MT_EE_TX_POWER_1_START_2G + 1,
                MT_EE_CP_FT_VERSION,
                MT_EE_XTAL_FREQ_OFFSET,
                MT_EE_XTAL_WF_RFCAL,*/
        };

nbd168, 2016-06-17:
The vndor driver also applies the efuse overrides to the existing calibration data. However, I noticed that it was skipping some values compared to MT7603.
Please try 271a999

noblepepper, 2016-06-19:
I still get very poor throughput on a mt7628, iperf results are below. What is most confusing is that this varies greatly with different boards. I get reasonable but slow performance on some, others are abysmal. Some are improved by disabling the cal free data some are degraded.

The results below are from a board identical to one that gave me 10-12 Mbs with the 05-14 revisions. Unfortunately I let the magic smoke out of that one so I can't duplicate this test on it.

chaos calmer with propriety driver:

local 192.168.3.1 port 5001 connected with 192.168.3.240 port 57461
[  4]  0.0-30.0 sec   168 MBytes  47.0 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.240 port 57462
[  4]  0.0-30.0 sec   199 MBytes  55.7 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.240 port 57463
[  4]  0.0-30.0 sec   213 MBytes  59.5 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)

lede 2016-05-14

[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 49848
[  4]  0.0-30.9 sec  9.88 MBytes  2.68 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 49850
[  4]  0.0-30.2 sec  8.75 MBytes  2.43 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 49852
[  4]  0.0-30.1 sec  13.9 MBytes  3.86 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)

lede 2016-05-14 cal_free commented

[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44168
[  4]  0.0-34.8 sec  1.75 MBytes  0.42 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44170
[  4]  0.0-30.4 sec  1.89 MBytes  0.52 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44172
[  4]  0.0-30.7 sec  1.10 MBytes  0.30 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)

lede 2016-06-17

local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44628
[  4]  0.0-33.0 sec  5.63 MBytes  1.43 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44630
[  4]  0.0-32.3 sec  2.14 MBytes  0.56 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 44632
[  4]  0.0-30.2 sec  5.75 MBytes  1.60 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)

lede 2016-06-17 cal_free commented

[  4]  0.0-34.0 sec  0.18 MBytes  0.04 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 52584
[  4]  0.0-31.4 sec  2.75 MBytes  0.73 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)
[  4] local 192.168.3.1 port 5001 connected with 192.168.3.238 port 52588
[  4]  0.0-32.4 sec  4.88 MBytes  1.26 Mbits/sec
[  4] MSS size 1448 bytes (MTU 1500 bytes, ethernet)

fherbert, 2018-05-20:
I'm running  the latest (as of today) mt76 driver (eaee49eb5e8f4bb761de4e1da217b06689b2eb11), on a mt7688. The mt76 performance seems about 10 times slower than your reported chaos calmer with proprietary driver. My results are:
```
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0- 1.0 sec   481 KBytes  3.94 Mbits/sec
[  4]  1.0- 2.0 sec   461 KBytes  3.78 Mbits/sec
[  4]  2.0- 3.0 sec   499 KBytes  4.09 Mbits/sec
[  4]  3.0- 4.0 sec   287 KBytes  2.35 Mbits/sec
[  4]  4.0- 5.0 sec   587 KBytes  4.81 Mbits/sec
[  4]  5.0- 6.0 sec   279 KBytes  2.28 Mbits/sec
[  4]  6.0- 7.0 sec   549 KBytes  4.49 Mbits/sec
[  4]  7.0- 8.0 sec   329 KBytes  2.70 Mbits/sec
[  4]  8.0- 9.0 sec   529 KBytes  4.33 Mbits/sec
[  4]  9.0-10.0 sec   451 KBytes  3.70 Mbits/sec
[  4]  0.0-10.5 sec  4.58 MBytes  3.67 Mbits/sec
```
As a comparison, running iperf on the ethernet interface of the mt7688 gets pretty much maxed out for the 100Mb interface:
```
[  4]  0.0- 1.0 sec  11.2 MBytes  94.3 Mbits/sec
[  4]  1.0- 2.0 sec  11.2 MBytes  94.1 Mbits/sec
[  4]  2.0- 3.0 sec  11.2 MBytes  93.8 Mbits/sec
[  4]  3.0- 4.0 sec  11.3 MBytes  94.5 Mbits/sec
[  4]  4.0- 5.0 sec  11.2 MBytes  94.2 Mbits/sec
[  4]  5.0- 6.0 sec  11.2 MBytes  94.1 Mbits/sec
[  4]  6.0- 7.0 sec  11.2 MBytes  94.1 Mbits/sec
[  4]  7.0- 8.0 sec  11.2 MBytes  94.1 Mbits/sec
[  4]  8.0- 9.0 sec  11.2 MBytes  94.2 Mbits/sec
[  4]  9.0-10.0 sec  11.2 MBytes  94.1 Mbits/sec
[  4]  0.0-10.0 sec   112 MBytes  94.2 Mbits/sec
```

adron-s, 2018-07-25:
I confirm the problem. On Onion Omega 2(MR7688AN) everything is exactly as described above. Ping delay fluctuates heavily and is partially packets lost.

ping on an empty channel(no traffic):
`# ping 192.168.1.1 -s 1472 -A -c 100
PING 192.168.1.1 (192.168.1.1) 1472(1500) bytes of data.
1480 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=16.9 ms
1480 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=217 ms
1480 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=430 ms
1480 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=603 ms
1480 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=562 ms
1480 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=359 ms
1480 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=279 ms
1480 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=77.6 ms
1480 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=41.6 ms
1480 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=14.9 ms
1480 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=49.6 ms
1480 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=189 ms
1480 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=123 ms
1480 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=30.3 ms
1480 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=75.8 ms
1480 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=266 ms
1480 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=207 ms
1480 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=27.6 ms
1480 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=178 ms
1480 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=27.6 ms
1480 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=14.9 ms
1480 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=14.9 ms
1480 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=14.9 ms
1480 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=114 ms
1480 bytes from 192.168.1.1: icmp_seq=27 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=28 ttl=64 time=21.7 ms
1480 bytes from 192.168.1.1: icmp_seq=29 ttl=64 time=29.3 ms
1480 bytes from 192.168.1.1: icmp_seq=30 ttl=64 time=251 ms
1480 bytes from 192.168.1.1: icmp_seq=31 ttl=64 time=90.5 ms
1480 bytes from 192.168.1.1: icmp_seq=41 ttl=64 time=233 ms
1480 bytes from 192.168.1.1: icmp_seq=42 ttl=64 time=148 ms
1480 bytes from 192.168.1.1: icmp_seq=43 ttl=64 time=71.1 ms
1480 bytes from 192.168.1.1: icmp_seq=44 ttl=64 time=38.5 ms
1480 bytes from 192.168.1.1: icmp_seq=45 ttl=64 time=114 ms
1480 bytes from 192.168.1.1: icmp_seq=46 ttl=64 time=16.9 ms
1480 bytes from 192.168.1.1: icmp_seq=48 ttl=64 time=285 ms
1480 bytes from 192.168.1.1: icmp_seq=49 ttl=64 time=105 ms
1480 bytes from 192.168.1.1: icmp_seq=50 ttl=64 time=92.3 ms
1480 bytes from 192.168.1.1: icmp_seq=51 ttl=64 time=60.5 ms
1480 bytes from 192.168.1.1: icmp_seq=52 ttl=64 time=73.5 ms
1480 bytes from 192.168.1.1: icmp_seq=53 ttl=64 time=28.4 ms
1480 bytes from 192.168.1.1: icmp_seq=54 ttl=64 time=88.8 ms
1480 bytes from 192.168.1.1: icmp_seq=55 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=56 ttl=64 time=14.9 ms
1480 bytes from 192.168.1.1: icmp_seq=57 ttl=64 time=114 ms
1480 bytes from 192.168.1.1: icmp_seq=58 ttl=64 time=65.2 ms
1480 bytes from 192.168.1.1: icmp_seq=59 ttl=64 time=63.8 ms
1480 bytes from 192.168.1.1: icmp_seq=60 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=61 ttl=64 time=90.1 ms
1480 bytes from 192.168.1.1: icmp_seq=62 ttl=64 time=90.9 ms
1480 bytes from 192.168.1.1: icmp_seq=63 ttl=64 time=50.5 ms
1480 bytes from 192.168.1.1: icmp_seq=64 ttl=64 time=65.8 ms
1480 bytes from 192.168.1.1: icmp_seq=65 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=66 ttl=64 time=125 ms
1480 bytes from 192.168.1.1: icmp_seq=67 ttl=64 time=23.1 ms
1480 bytes from 192.168.1.1: icmp_seq=68 ttl=64 time=29.7 ms
1480 bytes from 192.168.1.1: icmp_seq=69 ttl=64 time=76.3 ms
1480 bytes from 192.168.1.1: icmp_seq=70 ttl=64 time=14.8 ms
1480 bytes from 192.168.1.1: icmp_seq=71 ttl=64 time=136 ms
1480 bytes from 192.168.1.1: icmp_seq=72 ttl=64 time=66.6 ms
1480 bytes from 192.168.1.1: icmp_seq=73 ttl=64 time=63.5 ms
1480 bytes from 192.168.1.1: icmp_seq=74 ttl=64 time=140 ms
1480 bytes from 192.168.1.1: icmp_seq=75 ttl=64 time=418 ms
1480 bytes from 192.168.1.1: icmp_seq=76 ttl=64 time=552 ms
[... 47 lines cut ...]
1480 bytes from 192.168.1.1: icmp_seq=2698 ttl=64 time=0.595 ms
1480 bytes from 192.168.1.1: icmp_seq=2699 ttl=64 time=0.586 ms
1480 bytes from 192.168.1.1: icmp_seq=2700 ttl=64 time=0.595 ms
1480 bytes from 192.168.1.1: icmp_seq=2701 ttl=64 time=0.608 ms
1480 bytes from 192.168.1.1: icmp_seq=2702 ttl=64 time=0.613 ms
1480 bytes from 192.168.1.1: icmp_seq=2703 ttl=64 time=0.607 ms
1480 bytes from 192.168.1.1: icmp_seq=2704 ttl=64 time=0.643 ms
1480 bytes from 192.168.1.1: icmp_seq=2705 ttl=64 time=2.53 ms
1480 bytes from 192.168.1.1: icmp_seq=2706 ttl=64 time=0.619 ms
1480 bytes from 192.168.1.1: icmp_seq=2707 ttl=64 time=0.619 ms
1480 bytes from 192.168.1.1: icmp_seq=2708 ttl=64 time=1.25 ms
1480 bytes from 192.168.1.1: icmp_seq=2709 ttl=64 time=0.598 ms
1480 bytes from 192.168.1.1: icmp_seq=2710 ttl=64 time=0.621 ms
1480 bytes from 192.168.1.1: icmp_seq=2711 ttl=64 time=0.610 ms
1480 bytes from 192.168.1.1: icmp_seq=2712 ttl=64 time=0.648 ms
1480 bytes from 192.168.1.1: icmp_seq=2713 ttl=64 time=0.608 ms
1480 bytes from 192.168.1.1: icmp_seq=2714 ttl=64 time=0.602 ms
1480 bytes from 192.168.1.1: icmp_seq=2715 ttl=64 time=0.609 ms
1480 bytes from 192.168.1.1: icmp_seq=2716 ttl=64 time=0.611 ms
1480 bytes from 192.168.1.1: icmp_seq=2717 ttl=64 time=0.589 ms
^C
--- 192.168.1.1 ping statistics ---
2717 packets transmitted, 2717 received, 0% packet loss, time 3213ms
rtt min/avg/max/mdev = 0.572/0.780/24.572/1.064 ms, pipe 3, ipg/ewma 1.183/0.688 ms
`

adron-s, 2018-07-25:
I am ready to help in testing and finding a way to solve this problem. Tell me what to do.

adron-s, 2018-07-25:
**cat /etc/config/wireless**
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'owl'
        option encryption 'psk2'
        option key '1234004321'

**iw dev**
phy#0
        Interface wlan0
                ifindex 6
                wdev 0x2
                addr 40:a3:6b:c0:9f:fb
                ssid owl
                type AP
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz

**iw phy0 info**
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x1fe
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
[... 169 lines cut ...]
[    9.026516] xt_time: kernel timezone is -0000
[    9.185737] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[    9.195758] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[    9.201852] mt76_wmac 10300000.wmac: Build Time: 20151201183641
[    9.226552] firmware init done
[    9.399793] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[    9.426978] kmodloader: done loading kernel modules from /etc/modules.d/*
[   10.721166] urandom_read: 5 callbacks suppressed
[   10.721176] random: jshn: uninitialized urandom read (4 bytes read)
[   13.434151] rt3050-esw 10110000.esw: link changed 0x00
[   16.919186] rt3050-esw 10110000.esw: link changed 0x01
[   17.539028] br-lan: port 1(eth0) entered blocking state
[   17.544341] br-lan: port 1(eth0) entered disabled state
[   17.550059] device eth0 entered promiscuous mode
[   17.565434] br-lan: port 1(eth0) entered blocking state
[   17.570810] br-lan: port 1(eth0) entered forwarding state
[   17.576513] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   18.576687] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   20.537462] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   20.548493] br-lan: port 2(wlan0) entered blocking state
[   20.553935] br-lan: port 2(wlan0) entered disabled state
[   20.559885] device wlan0 entered promiscuous mode
[   21.085948] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   21.092666] br-lan: port 2(wlan0) entered blocking state
[   21.098093] br-lan: port 2(wlan0) entered forwarding state

adron-s, 2018-07-25:
**iwconfig on client notebook**
wlp1s0    IEEE 802.11bgn  ESSID:"owl"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: 40:A3:6B:C0:9F:FB   
          Bit Rate=72.2 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=58/70  Signal level=-52 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:32  Invalid misc:459   Missed beacon:0

nbd168, 2019-07-02:
Should be a lot better now in the latest version

closed 2019-07-02 by nbd168