openwrt/mt76 #32: MCU message timed out

akent, 2016-04-28:
Occasionally see this after changing wifi settings, driver locks up and doesn't recover. Typing iwconfig for example after this just hangs forever.

[ 1120.522768] wlan0: deauthenticating from 00:60:b3:07:20:08 by local choice (Reason: 3=DEAUTH_
LEAVING)
[ 1122.325976] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1123.976553] wlan0: authenticate with 00:60:b3:07:20:08
[ 1123.996551] wlan0: send auth to 00:60:b3:07:20:08 (try 1/3)
[ 1124.114411] wlan0: send auth to 00:60:b3:07:20:08 (try 2/3)
[ 1124.234318] wlan0: send auth to 00:60:b3:07:20:08 (try 3/3)
[ 1124.354314] wlan0: authentication with 00:60:b3:07:20:08 timed out
[ 1125.364329] MCU message 8 (seq 10) timed out
[ 1126.364329] MCU message 40 (seq 11) timed out
[ 1126.373459] ------------[ cut here ]------------
[ 1126.382720] WARNING: CPU: 0 PID: 1591 at /home/adam/scratch/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.22/linux-ramips_mt7688/mt76-2016-03-04/mt7603_mac.c:1190 mt7603_mac_work+0xf0/0x274 [mt7603e]()
[ 1126.638535] CPU: 0 PID: 1591 Comm: kworker/u2:0 Not tainted 4.4.6 #19
[ 1126.651313] Workqueue: phy0 mt7603_mac_work [mt7603e]

uosbox, 2016-04-29:
[  270.700000] ------------[ cut here ]------------
[  270.700000] WARNING: CPU: 0 PID: 225 at /kos/mc/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7628/mt76-2016-04-29/mt7603_mac.c:1202 mt7603_mac_work+0x274/0x28c [mt7603e]()
[  270.700000] Modules linked in: pppoe ppp_async iptable_nat rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv4 iptable_raw iptable_mangle iptable_filter ipt_REJECT ipt_MASQUERADE ipt_ECN ip_tables xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY x_tables ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc rt2x00pci rt2x00mmio rt2x00lib nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_log_common nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack crc_itu_t crc_ccitt fuse mt7603e mt76x2e mt76 mac80211 cfg80211 compat ledtrig_usbdev ntfs eeprom_93cx6 arc4 crypto_blkcipher usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  270.700000] CPU: 0 PID: 225 Comm: kworker/u2:2 Not tainted 3.18.29 #1
[  270.700000] Workqueue: phy0 mt7603_mac_work [mt7603e]
[  270.700000] Stack : 00000001 00000000 87924c80 802a43b8 87106328 871052a8 000004b2 00000000
[  270.700000]    87804a10 87804a00 00000088 800475b4 000000e1 8003a1b0 878ff790 00000000
[  270.700000]    802c8c1c 879a7d74 879a7d74 800475b4 00000003 80037ba0 000004b2 0000011b
[  270.700000]    00000000 00000030 00000000 00000000 00000000 00000000 00000000 00000000
[  270.700000]    30796870 00000000 00000000 00000000 00000000 00000000 87bae700 87bae800
[  270.700000]    ...
[  270.700000] Call Trace:
[  270.700000] [<800141a8>] show_stack+0x50/0x84
[  270.700000] [<80024480>] warn_slowpath_common+0x84/0xb4
[  270.700000] [<80024538>] warn_slowpath_null+0x18/0x24
[  270.700000] [<871052a8>] mt7603_mac_work+0x274/0x28c [mt7603e]
[  270.700000] [<80035c0c>] process_one_work+0x1fc/0x338
[  270.700000] [<80036760>] worker_thread+0x2b0/0x40c
[  270.700000] [<80039e5c>] kthread+0xd8/0xe4
[  270.700000] [<80004878>] ret_from_kernel_thread+0x14/0x1c
[  270.700000] 
[  270.700000] ---[ end trace 6fbda36019fbbb41 ]---

languitar, 2016-05-16:
I am seeing the same log message from time to time with the result that the network of the router becomes completely unresponsive. Only a reboot helps.

ajay-sk, 2016-07-22:
We are working on MT7628  and facing the same issue. Did you get any solution for this issue.

akent, 2016-07-22:
Sadly no, we have been forced back to the closed source driver due to this and other problems.

ajay-sk, 2016-07-23:
Thanks for quick response and providing the information.

languitar, 2016-07-23:
Strangely this doesn't happen in the stable release of openwrt. Only master builds show this for me.

nbd168, 2016-07-23:
The stable release can't show this, because it has no support for MT7603 at all.

ajay-sk, 2016-07-23:
@nbd168 : Yes, I could only find in the development branch. Do you have any idea of stable Mt76 repo or way to port mtk-wifi-gpl .

nbd168, 2017-12-03:
Should be fixed in the latest version

closed 2017-12-03 by nbd168