openwrt/mt76 #35: High packet loss when changing mac address

ecsv, 2016-05-25:
This was actually a report on the IRC in #linux-wireless. No answer from nbd was received (yet) so this is just to keep track of it. The relevant part of the message is the "second thing"

> nbd: d0tslash bought two of these witi boards which you've recommended for the mt76 evaluation.
> problem right now is the extreme packet loss (the AP is basically not usable for 2.4GHz/5GHz). this seems to be related to two different problems
> first there are only two connectors (4 are required for the 2x2 setup on 5GHz/2.4GHz)
> and do you know if there is a way to disable a chain on the device so I can avoid some of the (minor) performance problems?
> The other second things seems to be related with the mac addresses in my multivap setup. I am defining multiple APs (and IBSS) devices in my default setup
> each of the devices has a different mac set. for example @option macaddr 'ac:86:74:00:00:c2'@
> but even only enabling 1 AP per device but with a changed mac seems to cause these packet losses
> an example config (sry, didn't make it really clean)  can be found under https://gist.github.com/ecsv/b6c42f07e1da6cc2ea41fe72fa5fe4d3
> is this a known problem with mt76+these chips? the used LEDE version is reboot-393-gaf1e70b. If not then I can also report it at the proper place (github mt76 issue tracker?)
> I haven't yet checked any over-the-air traffic and I doubt that I will be able to do so today (have to leave in some minutes)
> just in case it is related to the initial mac of the device. phy0 mac is  00:00:00:00:00:20 and phy1 is 00:00:00:00:00:b8. yes, this looks bogus as hell but haven't yet checked how the dts sets the source of the  mac for this device

iancd, 2018-02-12:
@ecsv Is this still an issue with the latest version? if so what is the impact and the steps to repeat. If not could you close the issue?

ecsv, 2018-02-12:
Afaik this is still a problem but we stopped using wifi hw which uses mt76 due to this and similar problems.

closed 2018-02-12 by ecsv