openwrt/mt76 #36: To add support for mt7610e.

hero18688, 2016-05-25:
I hope this was in your schedule.

nbd168, 2016-06-17:
It's currently not on my schedule

closed 2016-06-17 by nbd168

srdjanrosic, 2017-06-23:
Hi, looking for an update, is it likely that mt76 will be made to work with mt7610e?

grantg182, 2018-03-14:
Just wondering if there's been any headway on this?

ptpt52, 2018-10-12:
@nbd168 hi
I noticed that the latest code adds support for mt7610, will it support mt7610e? Is the support of soc already available. How long does it take for pcie support? Can I do something?

LorenzoBianconi, 2018-10-12:
>
> @nbd168 <https://github.com/nbd168> hi
> I noticed that the latest code adds support for mt7610, will it support
> mt7610e? Is the support of soc already available. How long does it take for
> pcie support? Can I do something?
>

Hi,

we are currently working on mt7610e/mt7650e support. For the moment just
STA mode is supported, but I will work on AP support soon.
Regards,

Lorenzo

-- 
UNIX is Sexy: who | grep -i blonde | talk; cd ~; wine; talk; touch; unzip;
touch; strip; gasp; finger; gasp; mount; fsck; more; yes; gasp; umount;
make clean; sleep

jsantala, 2018-10-12:
Please also keep adhoc and mesh in mind, thanks! :)