openwrt/mt76 #37: MT7688AN relay mode, I get oops

widora, 2016-05-28:
**Now I use mt76 to connect to a AP that is used as a relay. Find two questions:
1, the speed is very slow, I try to use Ping AP superior MT7688, and sometimes a lot of time delay:**
64 bytes from 119.75.217.109: seq=149 ttl=56 time=36.037 ms
64 bytes from 119.75.217.109: seq=150 ttl=56 time=162.243 ms
64 bytes from 119.75.217.109: seq=151 ttl=56 time=141.894 ms
64 bytes from 119.75.217.109: seq=152 ttl=56 time=275.049 ms
64 bytes from 119.75.217.109: seq=153 ttl=56 time=98.003 ms
64 bytes from 119.75.217.109: seq=154 ttl=56 time=252.172 ms
64 bytes from 119.75.217.109: seq=155 ttl=56 time=682.828 ms
64 bytes from 119.75.217.109: seq=156 ttl=56 time=9.341 ms
64 bytes from 119.75.217.109: seq=157 ttl=56 time=416.193 ms
64 bytes from 119.75.217.109: seq=158 ttl=56 time=43.680 ms
64 bytes from 119.75.217.109: seq=159 ttl=56 time=294.452 ms
64 bytes from 119.75.217.109: seq=160 ttl=56 time=3129.075 ms
64 bytes from 119.75.217.109: seq=161 ttl=56 time=2140.186 ms
64 bytes from 119.75.217.109: seq=162 ttl=56 time=1325.464 ms
**2, is very easy to appear oops, as follows:**
64 bytes from 119.75.217.109: seq=188 ttl=56 time=228.178 ms
64 bytes from 119.75.217.109: seq=189 ttl=56 time=5.745 ms
[  231.655920] ------------[ cut here ]------------
[  231.660681] WARNING: CPU: 0 PID: 6 at /home/mango/lede/source/build_dir/target-mipsel_24kec+dsp_musl-1.1.14/linux-ramips_mt7688/mt76-2016-05-14/mt7603_mac.c:1202 mt7603_mac_work+0x208/0x278 [mt7603e]()
[  231.678860] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  231.749311] CPU: 0 PID: 6 Comm: kworker/u2:0 Not tainted 4.4.11 #3
[  231.755615] Workqueue: phy0 mt7603_mac_work [mt7603e]
[  231.761488] Stack : 87803410 8781aa98 00000088 8004a06c 87829b80 80390d43 8032dfec 00000006
          80356ef8 87845d64 80390000 80048024 00000088 8004a06c 803334c4 80390000
          00000003 87845d64 80390000 80038334 00000088 87845d9c 0000010e 00000000
          00000001 00000000 87094fa8 87b83800 87b83700 30796870 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[  231.805580] Call Trace:
[  231.810507] [<8001451c>] show_stack+0x50/0x84
[  231.814940] [<80024e34>] warn_slowpath_common+0xa0/0xd0
[  231.823492] [<80024ee8>] warn_slowpath_null+0x18/0x24
[  231.830260] [<870951b0>] mt7603_mac_work+0x208/0x278 [mt7603e]
[  231.836953] [<80036288>] process_one_work+0x1f8/0x334
[  231.842771] [<80037084>] worker_thread+0x2b4/0x408
[  231.847729] [<8003b3bc>] kthread+0xdc/0xe8
[  231.851891] [<80004478>] ret_from_kernel_thread+0x14/0x1c
[  231.859220] 
[  231.861069] ---[ end trace bdf32e2dfbbf2228 ]---
ping: sendto: Network unreachable
root@lede:/# [  234.503616] MCU message 40 (seq 2) timed out
[  235.512818] MCU message 8 (seq 3) timed out
[  236.512052] MCU message 40 (seq 4) timed out
[  240.778875] MCU message 40 (seq 6) timed out
[  241.778162] MCU message 8 (seq 7) timed out
[  242.777456] MCU message 40 (seq 8) timed out
[  243.776762] MCU message 8 (seq 9) timed out
[  244.776081] MCU message 40 (seq 10) timed out
[  245.775408] MCU message 8 (seq 11) timed out
[  246.774747] MCU message 40 (seq 12) timed out
[  247.774095] MCU message 8 (seq 13) timed out
[  248.773456] MCU message 40 (seq 14) timed out
[  249.772821] MCU message 8 (seq 15) timed out

widora, 2016-06-20:
Test environment：
lede+mt76,20160617,MT7688AN,128MB DDR2
router AP--------MT7688AN（sta+AP）-----------phone
when I running large amounts of data, very prone to oops.

ajay-sk, 2016-07-22:
We are working on MT7628 openwrt, it seem the same issue still exist with latest openwrt codebase. Did you get any solution for this issue.

jimzhong, 2017-02-05:
I also encountered connectivity loss on LEDE 17.01-rc1 + lenovo Y1(MT7620A+MT7612E) with the 5G radio in client (wds) mode.

```
[15384.904399] WARNING: CPU: 0 PID: 3838 at net/core/skbuff.c:4194 0x801f150c()
[15384.911590] Modules linked in: pppoe ppp_async iptable_nat rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables sit tunnel4 ip_tunnel tun eeprom_93cx6 usb_storage sd_mod scsi_mod ext4 jbd2 mbcache usbcore nls_base usb_common crc16 crc32c_generic crypto_hash leds_gpio gpio_button_hotplug
[15384.991609] CPU: 0 PID: 3838 Comm: nc Not tainted 4.4.42 #0
[15384.997313] Stack : 8031387c 00000000 00000001 80360000 00000000 00000000 00000000 00000000
[15384.997313] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[15384.997313] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[15384.997313] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[15384.997313] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[15384.997313] 	  ...
[15385.033704] Call Trace:[<80014488>] 0x80014488
[15385.038281] [<80014488>] 0x80014488
[15385.041842] [<80024f88>] 0x80024f88
[15385.045421] [<801f150c>] 0x801f150c
[15385.049005] [<80025040>] 0x80025040
[15385.052579] [<801f150c>] 0x801f150c
[15385.056167] [<8004b5ac>] 0x8004b5ac
[15385.059731] [<8024b230>] 0x8024b230
[15385.063317] [<8024ac48>] 0x8024ac48
[15385.066903] [<800047f4>] 0x800047f4
[15385.070495] [<877907f4>] 0x877907f4 [nf_conntrack@87790000+0xc860]
[15385.076833] [<8024b230>] 0x8024b230
[15385.080419] [<8024fa40>] 0x8024fa40
[15385.084024] [<876a0370>] 0x876a0370 [ip_tables@876a0000+0x2410]
[15385.090094] [<802577a4>] 0x802577a4
[15385.093672] [<800a5d90>] 0x800a5d90
[15385.097254] [<80259ed0>] 0x80259ed0
[15385.100853] [<80231208>] 0x80231208
[15385.104448] [<80237674>] 0x80237674
[15385.108036] [<80237c2c>] 0x80237c2c
[15385.111616] [<80237518>] 0x80237518
[15385.115207] [<80237ef8>] 0x80237ef8
[15385.118781] [<8004b5ac>] 0x8004b5ac
[15385.122341] [<80201702>] 0x80201702
[15385.125929] [<8023773c>] 0x8023773c
[15385.129505] [<802003ec>] 0x802003ec
[15385.133104] [<802dd070>] 0x802dd070
[15385.136730] [<802dd5ac>] 0x802dd5ac
[15385.140321] [<87751920>] 0x87751920 [mac80211@87700000+0x5ece0]
[15385.146413] [<802dd9ac>] 0x802dd9ac
[15385.150003] [<87fd0c34>] 0x87fd0c34 [mt76x2e@87fd0000+0x96e0]
[15385.155920] [<87717ad0>] 0x87717ad0 [mac80211@87700000+0x5ece0]
[15385.161978] [<802dd630>] 0x802dd630
[15385.165554] [<80200168>] 0x80200168
[15385.169161] [<8020112c>] 0x8020112c
[15385.172731] [<801f5a1c>] 0x801f5a1c
[15385.176332] [<802011c8>] 0x802011c8
[15385.179941] [<87fc50c8>] 0x87fc50c8 [mt76@87fc4000+0x3c00]
[15385.185593] [<80201370>] 0x80201370
[15385.189188] [<80027508>] 0x80027508
[15385.192749] [<8004f094>] 0x8004f094
[15385.196327] [<8004b5ac>] 0x8004b5ac
[15385.199891] [<802f0000>] 0x802f0000
[15385.203466] [<800047f4>] 0x800047f4
[15385.207035] 
[15385.208562] ---[ end trace 468acf3a59f3e3b2 ]---
[15385.586629] mt76x2e 0000:01:00.0: MCU message 31 (seq 2) timed out
[15386.267518] mt76x2e 0000:01:00.0: MCU message 31 (seq 3) timed out
[15386.609191] br-lan: port 3(upstream) entered disabled state
[15386.953592] mt76x2e 0000:01:00.0: MCU message 30 (seq 4) timed out
[15387.299036] mt76x2e 0000:01:00.0: MCU message 30 (seq 5) timed out
[15387.676302] mt76x2e 0000:01:00.0: MCU message 30 (seq 6) timed out
[15388.021778] mt76x2e 0000:01:00.0: MCU message 30 (seq 7) timed out
[15388.364427] mt76x2e 0000:01:00.0: MCU message 30 (seq 8) timed out
[15388.709087] mt76x2e 0000:01:00.0: MCU message 30 (seq 9) timed out
[15389.052486] mt76x2e 0000:01:00.0: MCU message 30 (seq 10) timed out
[15389.398481] mt76x2e 0000:01:00.0: MCU message 30 (seq 11) timed out
[15389.741558] mt76x2e 0000:01:00.0: MCU message 30 (seq 12) timed out
[15390.087790] mt76x2e 0000:01:00.0: MCU message 30 (seq 13) timed out
[15390.429499] mt76x2e 0000:01:00.0: MCU message 30 (seq 14) timed out
[15390.776077] mt76x2e 0000:01:00.0: MCU message 30 (seq 15) timed out
[15391.118293] mt76x2e 0000:01:00.0: MCU message 30 (seq 1) timed out
[15391.465012] mt76x2e 0000:01:00.0: MCU message 30 (seq 2) timed out
[15391.807075] mt76x2e 0000:01:00.0: MCU message 30 (seq 3) timed out
[15392.163150] mt76x2e 0000:01:00.0: MCU message 30 (seq 4) timed out
[15392.505245] mt76x2e 0000:01:00.0: MCU message 30 (seq 5) timed out
[15392.850993] mt76x2e 0000:01:00.0: MCU message 30 (seq 6) timed out
[15393.194499] mt76x2e 0000:01:00.0: MCU message 30 (seq 7) timed out
[15393.539959] mt76x2e 0000:01:00.0: MCU message 30 (seq 8) timed out
[15393.883052] mt76x2e 0000:01:00.0: MCU message 30 (seq 9) timed out
[15394.229709] mt76x2e 0000:01:00.0: MCU message 30 (seq 10) timed out
[15394.573997] mt76x2e 0000:01:00.0: MCU message 30 (seq 11) timed out
```

nbd168, 2017-03-15:
Please test the latest version

sevencheng798, 2017-05-03:
it seems the same issue for mt7628 and mt7688 to use latest version.

phatina, 2017-09-26:
Same here, using latest LEDE on Onion Omega2 (which is MT7688)...

skyatua, 2017-11-08:
I have a some issue. Any news?
root@LEDE:/# uname -a
Linux LEDE 4.9.58 #0 Tue Nov 7 11:02:06 2017 mips GNU/Linux
root@LEDE:/# cat /etc/openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r5279-2b6facc'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='LEDE Reboot SNAPSHOT r5279-2b6facc'
DISTRIB_TAINTS='no-all'

nbd168, 2017-12-03:
Should be fixed in the latest version. Please test

closed 2019-01-25 by nbd168