openwrt/mt76 #38: Performance of Open Driver vs MTK Closed Driver

Foeg, 2016-06-17:
We've been looking at the performance on the MT7688 based Linkit of this open driver relative to the original closed driver provided by MTK.  

Using iperf we get ~40Mbps of tcp performance on a local network.  
DUT - wifi - AP - ether - LAPTOP

The identical network and hardware setup with the open driver gets lets than 1Mbps (if that).Sometimes with significant stalls etc.

This is with https://github.com/openwrt/mt76/issues/28 applied.

Just wondering if others see the same poor throughput performance?

nbd168, 2016-06-17:
The vndor driver also applies the efuse overrides to the existing calibration data. However, I noticed that it was skipping some values compared to MT7603.
Please try 271a9998f995

nbd168, 2016-06-17:
Sorry, that was meant to be a response to #28, not this one.
The performance issues are known, rate control status feedback is still broken and I'm still waiting for more information from MTK

elkablo, 2016-09-29:
Is this fixed yet?

psyborg55, 2017-03-16:
test with the 5GHz CH56 VHT80 (card connect at HT40):

84% max - openwrt trunk r50082
87% max - PandoraBox-ralink-mt7620-xiaomi-mini-2017-01-03-git-6c24a7a-squashfs-sysupgrade.bin
91% max - miwifi_r1cm_all_ace8a_0.6.40.bin

try to increase power to 23dBm with openwrt results in tx rate regression 270/120, 23dBm with pandorabox 270/270

nbd168, 2017-03-16:
% values are meaningless. What does this translate to in dBm?

psyborg55, 2017-03-16:
-51dBm trunk
-49dBm pandorabox
-45dBm xiaomi

nbd168, 2017-03-16:
Are these different images on the same device in the same position?

nbd168, 2017-03-16:
Which device exactly is this?

psyborg55, 2017-03-16:
yes, different images, both xiaomi mini and notebook remained in the same positions during testing

mrkiko, 2017-06-24:
This situation might be better now.

psyborg55, 2018-07-08:
@barhom your "most probably" blind guesses are irrelevant. until you can reproduce the tests i've made so far and show what's been fixed meanwhile don't spam the tracker

adron-s, 2018-07-25:
The problem is still relevant. See my post [here](https://github.com/openwrt/mt76/issues/28#issuecomment-407843021)

nbd168, 2019-01-25:
Fixed in the current version

closed 2019-01-25 by nbd168