openwrt/mt76 #43: Compile Issue with openwrt linux version 3.10.14

amulpatel, 2016-08-19:
Hello Team, I am trying to cross compile mt76 driver for ZBT device MT7621 using medeiatek SDK.

package details:
PKG_NAME:=mt76
PKG_VERSION:=2016-06-17
PKG_SOURCE_VERSION:=f06ed811cda67ab2ed9932c3e87ba03229e75f4e

But seems it is not compatible with kernal version 3.10.14.
Can some one help me out for compilation issue?

Following are the compilation error:

openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:295:14: error: comparison between 'enum ieee80211_band' and 'enum nl80211_band' [-Werror=enum-compare]
  if (c->band == NL80211_BAND_2GHZ)
              ^
In file included from openwrt-3.10.14/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/include/mac80211-backport/linux/kernel.h:3:0,
                 from openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:17,
                 from openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mmio.c:14:
openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h: In function 'mtxq_to_txq':
include/linux/kernel.h:784:27: error: dereferencing pointer to incomplete type
  const typeof( ((type *)0)->member ) *__mptr = (ptr); \
                           ^
openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:317:9: note: in expansion of macro 'container_of'
  return container_of(ptr, struct ieee80211_txq, drv_priv);
         ^
In file included from include/linux/compiler-gcc.h:103:0,
                 from include/linux/compiler.h:54,
                 from openwrt-3.10.14/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/include/mac80211-backport/linux/compiler.h:3,
                 from include/linux/linkage.h:4,
                 from include/linux/kernel.h:6,
                 from openwrt-3.10.14/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/include/mac80211-backport/linux/kernel.h:3,
                 from openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:17,
                 from openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mmio.c:14:
openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:317:34: error: invalid use of undefined type 'struct ieee80211_txq'
  return container_of(ptr, struct ieee80211_txq, drv_priv);
                                  ^
include/linux/compiler-gcc4.h:14:53: note: in definition of macro '__compiler_offsetof'
 #define __compiler_offsetof(a,b) __builtin_offsetof(a,b)
                                                     ^
include/linux/kernel.h:785:29: note: in expansion of macro 'offsetof'
  (type *)( (char *)__mptr - offsetof(type,member) );})
                             ^
openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mt76.h:317:9: note: in expansion of macro 'container_of'
  return container_of(ptr, struct ieee80211_txq, drv_priv);
         ^
cc1: all warnings being treated as errors
make[4]: *** [openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/mmio.o] Error 1
make[3]: *** [_module_openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17] Error 2
make[3]: Leaving directory `openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/linux-3.10.14-p112871'
make[2]: *** [openwrt-3.10.14/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/linux-ramips_mt7621/mt76-2016-06-17/.built] Error 2
make[2]: Leaving directory `openwrt-3.10.14/package/kernel/mt76'
make[1]: *** [package/kernel/mt76/compile] Error 2
make[1]: Leaving directory `openwrt-3.10.14'
make: *** [package/kernel/mt76/compile] Error 2
[amulpatel@Amul-PC openwrt-3.10.14]$ 

Thanks,
Amul Patel

amulpatel, 2016-08-19:
After debugging i found that the driver support 4.* kernel as there are few structure is not defined in older kernal mac80211.h e.g. struct ieee80211_txq

After fixing this.. there are more stuff which is not supported.

Looks  I either i need to upgrade kernel to 4.*  or apply patches to kernel for compatibility with mt76 driver.

Thanks.
Amul Patel

psyborg55, 2016-08-28:
i've ran into same bug on 4.4.7 kernel. did you find out how to fix it?

nbd168, 2016-08-29:
mt76 requires a newer version of compat-wireless to build against. This is not a bug in the code.

closed 2016-08-29 by nbd168

amulpatel, 2016-08-29:
Thanks for update

I had used mac80211 driver with version compat-wireless-2016-01-10 considering latest one and compiled it.
But MT76 driver refer the linux dir path e.g.  ~/net/ where almost all headers available.
Also other stuff such as "skb_free_frag" are not part of compat-wireless and its available in kernel headers and it is used in MT76 driver.

Thanks,
Amul Patel

nbd168, 2016-08-29:
LEDE uses compat-wireless 2016-06-20, and it includes backports for the relevant functions, e.g. skb_free_frag.

amulpatel, 2016-08-29:
Thanks, I am checkingwith  compat-wireless 2016-06-20.

Can you please let me know how to make use compat-wireless 2016-06-20 code use by mt76 driver instead of linux source tree?

Does it required header path update in mt76 driver to use  compat-wireless header & source tree.

Since mt76 & compat-wireless are 2 different driver & i guess some merging will be required of mt76 into compat-wireless. ??

Thanks,
Amul Patel

nbd168, 2016-08-29:
The package in OpenWrt/LEDE already uses the headers from there

amulpatel, 2016-08-29:
Thanks,

Let me explain my requirement:
Actually I am looking hostapd support in Mediatek device MT7621 but compate-wireless (mac80211) does not support it.
hence i am building mt76 driver which have support for hostapd (nl80211 with mac80211).

So just need clarification on my requirement i.e. 
Do you mean MT76 driver (mainly hostapd nl80211) is supported in compat-wireless 2016-06-20 ?

I dont need to build MT76 separate and compat-wireless 2016-06-20 (mac80211) will be enough to run hostapd? 

Thanks,
Amul