openwrt/mt76 #45: MT76 Kernel Warning (two cases)

leaders99, 2016-08-30:
I using MT7621A base board(7603E + 7612E)
It has ePA + LNA, but mt76 seems  can't support it.
(MTK openwrt can't support too. But MTK SDK can support normal through)

Is it normal? 

root@OpenWrt:/proc# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Bleeding Edge'
DISTRIB_REVISION='49918'
DISTRIB_CODENAME='designated_driver'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_DESCRIPTION='OpenWrt Designated Driver 49918'
DISTRIB_TAINTS='no-all'
root@OpenWrt:/proc# 

**No.1. mt7603_mac_work oops**

Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt sg mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables vfat fat ntfs nls_utf8 nls_iso8859_15 nls_iso8859_1 nls_cp850 nls_cp437 usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  707.740000] CPU: 2 PID: 644 Comm: kworker/u8:3 Not tainted 4.4.14 #1
[  707.760000] Workqueue: phy1 mt7603_mac_work [mt7603e]
[  707.770000] Stack : 8ed89900 80450000 803cc8a8 000004b5 8eda5258 00000000 8f804614 8ed89918
          00000088 80063514 803cc8a8 00000002 00000284 804a365c 803e80fc 8ecb3d54
          80450000 8006129c 80450000 804c0000 80452158 8045215c 803d11bc 8ecb3d54
          80450000 80040ea8 8f804614 8ecb3d8c 00000385 00000000 00000000 00cb3d74
          8eda5054 8f8f4d00 8f8f4a00 31796870 00000000 00000000 00000000 00000000
          ...
[  707.840000] Call Trace:
[  707.840000] [<800165b0>] show_stack+0x50/0x84
[  707.850000] [<801aaf58>] dump_stack+0x84/0xbc
[  707.860000] [<8002be10>] warn_slowpath_common+0xa0/0xd0
[  707.870000] [<8002bec4>] warn_slowpath_null+0x18/0x24
[  707.880000] [<8eda5258>] mt7603_mac_work+0x204/0x27c [mt7603e]
[  707.890000] [<8003eb08>] process_one_work+0x214/0x358
[  707.900000] [<8003f9f0>] worker_thread+0x2dc/0x444
[  707.910000] [<80044130>] kthread+0xdc/0xe8
[  707.920000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[  707.930000] 
[  707.930000] ---[ end trace dbd185f3a07fa571 ]---

**No.2 mt7603_mac_fill_rx oops**
[  759.320000] ------------[ cut here ]------------
[  759.320000] WARNING: CPU: 0 PID: 0 at /home/ccsp/project/openwrt/openwrt/openwrt/openwrt/build_dir/target-mipsel_1004kc+dsp_musl-1.1.14/linux-ramips_mt7621/mt76-2016-06-17/mt7603_mac.c:362 mt7603_mac_fill_rx+0x124/0x3b0 [mt7603e]()
[  759.370000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt sg mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables vfat fat ntfs nls_utf8 nls_iso8859_15 nls_iso8859_1 nls_cp850 nls_cp437 usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  759.510000] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.4.14 #1
[  759.510000] Stack : 00000000 00000000 804a6842 00000041 00000000 00000000 80450000 804c0000
          8044e040 8044dc63 803cc8a8 00000000 00000000 804a365c 00000001 8ec1d900
          00000000 80063514 80450000 804c0000 80452158 8045215c 803d11bc 8043fb94
          00000003 8006129c 00000001 8ec1d900 00000001 00000000 00000000 0043fb94
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[  759.510000] Call Trace:
[  759.510000] [<800165b0>] show_stack+0x50/0x84
[  759.510000] [<801aaf58>] dump_stack+0x84/0xbc
[  759.510000] [<8002be10>] warn_slowpath_common+0xa0/0xd0
[  759.510000] [<8002bec4>] warn_slowpath_null+0x18/0x24
[  759.510000] [<8eda3b6c>] mt7603_mac_fill_rx+0x124/0x3b0 [mt7603e]
[  759.510000] [<8eda2b0c>] mt7603_queue_rx_skb+0xb8/0x100 [mt7603e]
[  759.510000] [<8eda9174>] mt76_dma_attach+0xcb4/0xdd4 [mt76]
[  759.510000] 
[  759.680000] ---[ end trace dbd185f3a07fa572 ]---

MaFeSa, 2018-01-03:
Xiaomi Router 3G - OpenWrt SNAPSHOT r5521-9f8d282 / LuCI Master (git-17.328.04231-802d5b6)

```
[265739.627374] WARNING: CPU: 0 PID: 17696 at /home/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-12-08-e5046560/mt7603_mac.c:1088 mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[265739.644741] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uas slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit
[265739.715948]  act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs hfsplus nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common crc32c_generic
[265739.758764] CPU: 0 PID: 17696 Comm: kworker/u8:0 Tainted: G        W       4.9.67 #0
[265739.766575] Workqueue: phy0 mt7603_mac_work [mt7603e]
[265739.771688] Stack : 8e35f000 804d0000 8046e9d4 8e6e68e8 00000009 00000440 8e35f018 080ffff0
[265739.780122]         00000064 8006a858 8046e9d4 00000000 00004520 80533824 8046e940 8e267d3c
[265739.788554]         804d0000 800a7280 00000001 80530000 804dbec4 804dbec8 804735e0 8e267d3c
[265739.796986]         804d0000 80042334 8e35f018 8e267d74 0000056d 00000000 00000001 00267d5c
[265739.805418]         8e6e52fc 8f205400 8f205500 30796870 00000000 00000000 00000000 00000000
[265739.813850]         ...
[265739.816372] Call Trace:
[265739.818916] [<8000f884>] show_stack+0x54/0x88
[265739.823352] [<801d9dcc>] dump_stack+0x8c/0xd0
[265739.827777] [<8002add4>] __warn+0xe4/0x118
[265739.831939] [<8002ae9c>] warn_slowpath_null+0x1c/0x30
[265739.837059] [<8e6e5628>] mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[265739.842978] [<80040114>] process_one_work+0x200/0x348
[265739.848096] [<80041034>] worker_thread+0x2fc/0x488
[265739.852963] [<80045c10>] kthread+0xe0/0xf4
[265739.857126] [<8000a478>] ret_from_kernel_thread+0x14/0x1c
[265739.862858] ---[ end trace 6068b4b3bcc38be9 ]---
[280396.204187] ------------[ cut here ]------------
```

MaFeSa, 2018-02-06:
Same problem also with latest update
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=fe1244e04d45872f09f47b49465ca9367b4c8bd0

```
37613.746359] WARNING: CPU: 0 PID: 12824 at /home/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-01-28-8693864c/mt7603_mac.c:1168 mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[37613.764102] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack act_skbedit
[37613.834655]  act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic
[37613.876574] CPU: 0 PID: 12824 Comm: kworker/u8:0 Not tainted 4.9.77 #0
[37613.883089] Workqueue: phy0 mt7603_mac_work [mt7603e]
[37613.888115] Stack : 8e4d3e80 804c0000 80460614 8ef6eb68 00000009 00000490 8e4d3e98 080ffff2
[37613.896463]         00000064 8006a9e8 80460614 00000000 00003218 80523824 80460580 8fe6dd3c
[37613.904810]         804c0000 800a78a0 00000001 80520000 804cdec4 804cdec8 80465220 8fe6dd3c
[37613.913156]         804c0000 8004235c 8e4d3e98 8fe6dd74 0000040b 00000000 00000001 00e6dd5c
[37613.921500]         8ef6d57c 8f7bd700 8f7bd800 30796870 00000000 00000000 00000000 00000000
[37613.929846]         ...
[37613.932284] Call Trace:
[37613.934741] [<8000f714>] show_stack+0x54/0x88
[37613.939088] [<801da45c>] dump_stack+0x8c/0xd0
[37613.943428] [<8002adc4>] __warn+0xe4/0x118
[37613.947505] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[37613.952537] [<8ef6d8a8>] mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[37613.958367] [<8004013c>] process_one_work+0x200/0x348
[37613.963400] [<8004105c>] worker_thread+0x2fc/0x488
[37613.968178] [<80045c40>] kthread+0xe0/0xf4
[37613.972257] [<8000a458>] ret_from_kernel_thread+0x14/0x1c
[37613.977741] ---[ end trace e2a3f084b33b170c ]---
```

MaFeSa, 2018-02-10:
Also with latest mt76 update there is this warning

```
[78328.598527] WARNING: CPU: 1 PID: 17853 at /home/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-02-09-246d548b/mt7603_mac.c:1168 mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[78328.615965] Modules linked in: pppoe ppp_async option usb_wwan qmi_wwan pppox ppp_mppe ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter
[78328.686964]  ipt_ECN ip_tables crc_ccitt compat cdc_wdm sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip_gre gre ifb ip_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common mii crc32c_generic
[78328.735998] CPU: 1 PID: 17853 Comm: kworker/u8:1 Not tainted 4.9.77 #0
[78328.742513] Workqueue: phy0 mt7603_mac_work [mt7603e]
[78328.747541] Stack : 8e51f800 804c0000 80460614 8ea6ebc8 00000009 00000490 8e51f818 080ffff3
[78328.755892]         00000064 8006a9e8 80460614 00000001 000045bd 80523824 80460580 82791d3c
[78328.764239]         804c0000 800a78a0 00000001 80520000 804cdec4 804cdec8 80465220 82791d3c
[78328.772583]         804c0000 8004235c 8e51f818 82791d74 0000053e 00000000 00000001 00791d5c
[78328.780931]         8ea6d5dc 8f323b00 8f323a00 30796870 00000000 00000000 00000000 00000000
[78328.789278]         ...
[78328.791715] Call Trace:
[78328.794176] [<8000f714>] show_stack+0x54/0x88
[78328.798535] [<801da45c>] dump_stack+0x8c/0xd0
[78328.802878] [<8002adc4>] __warn+0xe4/0x118
[78328.806959] [<8002ae8c>] warn_slowpath_null+0x1c/0x30
[78328.811995] [<8ea6d908>] mt7603_mac_work+0x32c/0x4d4 [mt7603e]
[78328.817829] [<8004013c>] process_one_work+0x200/0x348
[78328.822860] [<8004105c>] worker_thread+0x2fc/0x488
[78328.827642] [<80045c40>] kthread+0xe0/0xf4
[78328.831720] [<8000a458>] ret_from_kernel_thread+0x14/0x1c
[78328.837207] ---[ end trace e9efb0f734778945 ]---
```

MaFeSa, 2018-05-19:
Xiaomi Mi Router 3G
OpenWrt SNAPSHOT r6909-ac70ac3 / LuCI Master (git-18.135.54932-47971d2)

```
[125765.525114] ------------[ cut here ]------------
[125765.529862] WARNING: CPU: 0 PID: 0 at /home/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-05-15-e87f9256/mt76x2_mac.c:360 mt76x2_mac_process_rx+0x1b4/0x3f0 [mt76x2e]
[125765.547454] Modules linked in: pppoe ppp_async option usb_wwan pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4
[125765.618063]  nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat cdc_wdm fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 tunnel6 tunnel4 tun af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic sha1_generic jitterentropy_rng drbg md5 hmac echainiv des_generic cbc authenc uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd
[125765.688719]  sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic
[125765.699623] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.37 #0
[125765.705692] Stack : 00000000 00000000 00000000 00000000 805e7ad2 00000034 00000000 00000000
[125765.714112]         80582d74 80582907 805121bc 00000000 00000000 00000001 8fc09d08 1cc282c5
[125765.722531]         00000000 00000000 805e0000 00007a58 00000000 0000017c 00000008 00000000
[125765.730948]         00000000 80580000 000aace7 70617773 00000000 00000000 805a0000 8e864c24
[125765.739364]         00000009 00000168 00000002 0000fe41 00000000 8028ee38 00000000 805e0000
[125765.747783]         ...
[125765.750305] Call Trace:
[125765.752850] [<800104d8>] show_stack+0x58/0x100
[125765.757370] [<80454eec>] dump_stack+0x9c/0xe0
[125765.761796] [<8002e218>] __warn+0xe0/0x114
[125765.765960] [<8002e2dc>] warn_slowpath_null+0x1c/0x30
[125765.771081] [<8e864c24>] mt76x2_mac_process_rx+0x1b4/0x3f0 [mt76x2e]
[125765.777497] [<8e860524>] mt76x2_queue_rx_skb+0x84/0xcc [mt76x2e]
[125765.783575] [<8f2b0df4>] mt76_dma_attach+0xae4/0xd50 [mt76]
[125765.789282] ---[ end trace 613115118f759953 ]---
```

closed 2019-01-26 by nbd168