openwrt/mt76 #47: build error caused by missing bitfield.h

psyborg55, 2016-09-05:
just tried compiling trunk with last mt76 commit and get this error:

> In file included from /home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.14/linux-ramips_mt7620/mt76-2016-03-04/mt76.h:25:0,
                 from /home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.14/linux-ramips_mt7620/mt76-2016-03-04/mmio.c:17:
/home/ubuntu/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.14/linux-ramips_mt7620/mt76-2016-03-04/util.h:20:28: fatal error: linux/bitfield.h: No such file or directory
compilation terminated.

nbd168, 2016-09-05:
LEDE r1513 has that header file (as part of the mac80211 package)

closed 2016-09-05 by nbd168

psyborg55, 2016-10-20:
the solution referenced above worked