openwrt/mt76 #50: Use proper commit logs

AlbanBedel, 2016-09-16:
Please, please try to improve the commit log. I'm currently looking at an issue and looking at the log is USELESS. All commit are just a short "fix X and Y" without any description of the problem fixed or what the fix is doing. I do hope the authors known what they are fixing, but this completely prevents others from tracking issues. Furthermore it make it hard for new developers to start working on the project as no explanations of the changes can be found in the log.

I known this can't be fixed for old commit, but in the future please try to adhere to the usual community standards and take a few minutes to write proper log messages. This will drastically improve the maintainability of you project and make it a lot easier for new developers to join.

blogic, 2016-09-16:
now that you vented your frustration, tell us what the bug is that you are hunting

AlbanBedel, 2016-09-16:
I'm running an open AP which sees a lot of clients (Freifunk in a public place in Hamburg) and some stations never get removed from the AP. So the amount of stations grows up to 200+ until the AP crash. This is a problem with 2.4Gz, it might also exists with 5Gz but we don't have enough 5Gz clients to judge. The AP is a ZBT-WG2626 running OpenWRT 15.04.01, so the mt76 module is out of date and I really wanted to know if some recent commits might have fixed that problem. I will need to try the current mt76 before doing a proper report, however looking at the currently open issues it doesn't seems like the current head is fit for production use :/

AlbanBedel, 2016-09-16:
Here is a graph showing the number of clients on one of the AP yesterday evening, we only had a few people in the building still it raised well above 30.

[image]

nbd168, 2016-09-16:
Please try the latest version before reporting this again. I would say that at least most of the currently open issues are not regressions, so I don't see a reason why sticking to an old version is any better.

closed 2016-09-16 by nbd168