openwrt/mt76 #53: bad reception throughput

psyborg55, 2016-10-21:
i've noticed that with recent commits reception speeds are pretty low. for comparison i've used previously compiled images and the speed drop increases as the image being tested gets newer.

for example

CC46672

>     ubuntu@vivid:~$ iperf -c 192.168.5.1 -i 1 -t 20
    ------------------------------------------------------------
    Client connecting to 192.168.5.1, TCP port 5001
    TCP window size: 85.0 KByte (default)
    ------------------------------------------------------------
    [  3] local 192.168.5.47 port 35983 connected with 192.168.5.1 port 5001
    [ ID] Interval       Transfer     Bandwidth
    [  3]  0.0- 1.0 sec  5.62 MBytes  47.2 Mbits/sec
    [  3]  1.0- 2.0 sec  5.62 MBytes  47.2 Mbits/sec
    [  3]  2.0- 3.0 sec  5.50 MBytes  46.1 Mbits/sec
    [  3]  3.0- 4.0 sec  5.75 MBytes  48.2 Mbits/sec
    [  3]  4.0- 5.0 sec  5.62 MBytes  47.2 Mbits/sec
    [  3]  5.0- 6.0 sec  7.50 MBytes  62.9 Mbits/sec
    [  3]  6.0- 7.0 sec  7.25 MBytes  60.8 Mbits/sec
    [  3]  7.0- 8.0 sec  7.38 MBytes  61.9 Mbits/sec
    [  3]  8.0- 9.0 sec  7.25 MBytes  60.8 Mbits/sec
    [  3]  9.0-10.0 sec  7.25 MBytes  60.8 Mbits/sec
    [  3] 10.0-11.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 11.0-12.0 sec  9.75 MBytes  81.8 Mbits/sec
    [  3] 12.0-13.0 sec  8.62 MBytes  72.4 Mbits/sec
    [  3] 13.0-14.0 sec  8.88 MBytes  74.4 Mbits/sec
    [  3] 14.0-15.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 15.0-16.0 sec  8.25 MBytes  69.2 Mbits/sec
    [  3] 16.0-17.0 sec  9.25 MBytes  77.6 Mbits/sec
    [  3] 17.0-18.0 sec  11.1 MBytes  93.3 Mbits/sec
    [  3] 18.0-19.0 sec  8.75 MBytes  73.4 Mbits/sec
    [  3] 19.0-20.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3]  0.0-20.0 sec   157 MBytes  65.7 Mbits/sec

vs DD49946

>     ubuntu@vivid:~$ iperf -c 192.168.5.1 -i 1 -t 20
    ------------------------------------------------------------
    Client connecting to 192.168.5.1, TCP port 5001
    TCP window size: 85.0 KByte (default)
    ------------------------------------------------------------
    [  3] local 192.168.5.47 port 36076 connected with 192.168.5.1 port 5001
    [ ID] Interval       Transfer     Bandwidth
    [  3]  0.0- 1.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  1.0- 2.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  2.0- 3.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  3.0- 4.0 sec  4.88 MBytes  40.9 Mbits/sec
    [  3]  4.0- 5.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  5.0- 6.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  6.0- 7.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  7.0- 8.0 sec  4.75 MBytes  39.8 Mbits/sec
    [  3]  8.0- 9.0 sec  4.88 MBytes  40.9 Mbits/sec
    [  3]  9.0-10.0 sec  4.88 MBytes  40.9 Mbits/sec
    [  3] 10.0-11.0 sec  4.88 MBytes  40.9 Mbits/sec
    [  3] 11.0-12.0 sec  4.00 MBytes  33.6 Mbits/sec
    [  3] 12.0-13.0 sec  3.88 MBytes  32.5 Mbits/sec
    [  3] 13.0-14.0 sec  3.62 MBytes  30.4 Mbits/sec
    [  3] 14.0-15.0 sec  3.12 MBytes  26.2 Mbits/sec
    [  3] 15.0-16.0 sec  2.88 MBytes  24.1 Mbits/sec
    [  3] 16.0-17.0 sec  3.00 MBytes  25.2 Mbits/sec
    [  3] 17.0-18.0 sec  3.12 MBytes  26.2 Mbits/sec
    [  3] 18.0-19.0 sec  3.12 MBytes  26.2 Mbits/sec
    [  3] 19.0-20.0 sec  3.62 MBytes  30.4 Mbits/sec
    [  3]  0.0-20.2 sec  83.2 MBytes  34.6 Mbits/sec

at least i can confirm transmission got some improvement:

CC46672

>     root@OpenWrt:~# iperf -c 192.168.5.47 -i 1 -t 20
    ------------------------------------------------------------
    Client connecting to 192.168.5.47, TCP port 5001
    TCP window size: 43.8 KByte (default)
    ------------------------------------------------------------
    [  3] local 192.168.5.1 port 55801 connected with 192.168.5.47 port 5001
    [ ID] Interval       Transfer     Bandwidth
    [  3]  0.0- 1.0 sec  6.50 MBytes  54.5 Mbits/sec
    [  3]  1.0- 2.0 sec  6.00 MBytes  50.3 Mbits/sec
    [  3]  2.0- 3.0 sec  6.75 MBytes  56.6 Mbits/sec
    [  3]  3.0- 4.0 sec  7.25 MBytes  60.8 Mbits/sec
    [  3]  4.0- 5.0 sec  7.38 MBytes  61.9 Mbits/sec
    [  3]  5.0- 6.0 sec  7.50 MBytes  62.9 Mbits/sec
    [  3]  6.0- 7.0 sec  7.75 MBytes  65.0 Mbits/sec
    [  3]  7.0- 8.0 sec  6.25 MBytes  52.4 Mbits/sec
    [  3]  8.0- 9.0 sec  7.00 MBytes  58.7 Mbits/sec
    [  3]  9.0-10.0 sec  7.62 MBytes  64.0 Mbits/sec
    [  3] 10.0-11.0 sec  7.50 MBytes  62.9 Mbits/sec
    [  3] 11.0-12.0 sec  7.62 MBytes  64.0 Mbits/sec
    [  3] 12.0-13.0 sec  7.62 MBytes  64.0 Mbits/sec
    [  3] 13.0-14.0 sec  6.88 MBytes  57.7 Mbits/sec
    [  3] 14.0-15.0 sec  6.00 MBytes  50.3 Mbits/sec
    [  3] 15.0-16.0 sec  6.88 MBytes  57.7 Mbits/sec
    [  3] 16.0-17.0 sec  7.25 MBytes  60.8 Mbits/sec
    [  3] 17.0-18.0 sec  7.00 MBytes  58.7 Mbits/sec
    [  3] 18.0-19.0 sec  6.88 MBytes  57.7 Mbits/sec
    [  3] 19.0-20.0 sec  6.88 MBytes  57.7 Mbits/sec
    [  3]  0.0-20.0 sec   141 MBytes  58.9 Mbits/sec

DD49946

>     root@OpenWrt:~# iperf -c 192.168.5.47 -i 1 -t 20
    ------------------------------------------------------------
    Client connecting to 192.168.5.47, TCP port 5001
    TCP window size: 43.8 KByte (default)
    ------------------------------------------------------------
    [  3] local 192.168.5.1 port 42930 connected with 192.168.5.47 port 5001
    [ ID] Interval       Transfer     Bandwidth
    [  3]  0.0- 1.0 sec  10.1 MBytes  84.9 Mbits/sec
    [  3]  1.0- 2.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3]  2.0- 3.0 sec  9.50 MBytes  79.7 Mbits/sec
    [  3]  3.0- 4.0 sec  9.50 MBytes  79.7 Mbits/sec
    [  3]  4.0- 5.0 sec  9.25 MBytes  77.6 Mbits/sec
    [  3]  5.0- 6.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3]  6.0- 7.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3]  7.0- 8.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3]  8.0- 9.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3]  9.0-10.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 10.0-11.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 11.0-12.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 12.0-13.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3] 13.0-14.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3] 14.0-15.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 15.0-16.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 16.0-17.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 17.0-18.0 sec  9.12 MBytes  76.5 Mbits/sec
    [  3] 18.0-19.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3] 19.0-20.0 sec  9.00 MBytes  75.5 Mbits/sec
    [  3]  0.0-20.0 sec   184 MBytes  77.0 Mbits/sec

psyborg55, 2017-01-05:
part of the problem: https://github.com/openwrt/mt76/issues/70

nbd168, 2017-12-03:
Please try the latest version, it should work better now

closed 2018-10-02 by nbd168