openwrt/mt76 #56: per rate power control not working

psyborg55, 2016-10-24:
similar to this test https://github.com/openwrt/openwrt/issues/146

i've tried changing rates on mt7612e from 54M to 6M

all rates give the same output ~ -62dBm

xiaomi mini running openwrt 49946 with latest mt76 commit

nbd168, 2016-10-24:
All rates giving the same power output is the normal behavior. What's the bug?

psyborg55, 2016-10-24:
Usually lower rates should give higher output power. Maybe the bug with parsing target rate power values from eeprom, or setting these values into chip?

nbd168, 2016-10-24:
If the amplifier is good enough and can make the target power work for higher rates as well, then lower rates will not have higher power. It also depends on the configured tx power.

psyborg55, 2016-10-24:
sorry but i can't agree on this one. the device has external PA tied to mt7620a and as you can see from a test referenced in first post it still has slightly better range at lower rates despite the bug with 6 and 9 mbps rates. mt7612e operates with internal PA and my attempt to raise power by modifying eeprom to output 22dBm instead of 17 gave slightly better range but the tx rate dropped from 240-300 to 90-120 mbps affecting the real throughput. and going beyond 22dBm had no any effect on output power or range.

nbd168, 2016-10-24:
Per-rate power offsets are stored in the EEPROM based on the target power that the device is calibrated for. Hacking the EEPROM data to increase overall power will not adjust per-rate power limits accordingly, only a recalibration of the device for a higher target power will.
I still don't see any indication of a driver bug, and mt7620a behavior is not relevant here, because of big differences in hardware design.

closed 2017-10-13 by nbd168