openwrt/mt76 #57: Set WiFi 5G channel to 52 or later will disable wifi

imjayge, 2016-10-24:
Hi, my board wifi chip is using mt7621,7603.

I've updated my openwrt mt76 driver to "mt76x2: dfs: remove unnecessary memset(62abac9ea4fd8d0e9a651767b345e99c1fc7febc)"

and add patch for bitfield.h so that I can build succeessfully.

I test wifi 5G channel by the order: 36,40,44,48,52,56 ...

However, when I changed to channel 52, wireless webGUI shows "Wireless is disabled or not associated".

Does the dfs malfunction?

Below is my dts configuration:

&pcie {
        status = "okay";

        pcie0 {
                mt76@0,0 {
                        reg = <0x0000 0 0 0 0>;
                        device_type = "pci";
                        mediatek,mtd-eeprom = <&factory 0x8000>;
                        mediatek,2ghz = <0>;                       
                };
        };

        pcie1 {
                mt76@1,0 {
                        reg = <0x0000 0 0 0 0>;
                        device_type = "pci";
                        mediatek,mtd-eeprom = <&factory 0x0000>;
                        mediatek,5ghz = <0>;                        
                };
        };
};

and the console log just like:

oot@OpenWrt:/# [  447.490000] br-lan: port 2(wlan0) entered disabled state
[  447.560000] device wlan0 left promiscuous mode
[  447.560000] br-lan: port 2(wlan0) entered disabled state
[  448.450000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  448.460000] device wlan0 entered promiscuous mode
[  449.090000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  449.100000] br-lan: port 2(wlan0) entered forwarding state
[  449.110000] br-lan: port 2(wlan0) entered forwarding state
[  451.110000] br-lan: port 2(wlan0) entered forwarding state
[  610.610000] device wlan0 left promiscuous mode
[  610.610000] br-lan: port 2(wlan0) entered disabled state
[  611.860000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  611.880000] device wlan0 entered promiscuous mode
[  611.890000] br-lan: port 2(wlan0) entered forwarding state
[  611.900000] br-lan: port 2(wlan0) entered forwarding state
[  611.980000] br-lan: port 2(wlan0) entered disabled state
[  612.220000] device wlan0 left promiscuous mode
[  612.220000] br-lan: port 2(wlan0) entered disabled state
[  612.660000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.480000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  623.500000] device wlan0 entered promiscuous mode
[  623.790000] device wlan0 left promiscuous mode
[  623.790000] br-lan: port 2(wlan0) entered disabled state
[  624.230000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  635.010000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  635.030000] device wlan0 entered promiscuous mode
[  635.320000] device wlan0 left promiscuous mode
[  635.320000] br-lan: port 2(wlan0) entered disabled state
[  635.760000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  646.580000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  646.600000] device wlan0 entered promiscuous mode
[  646.880000] device wlan0 left promiscuous mode
[  646.880000] br-lan: port 2(wlan0) entered disabled state
[  647.330000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

If you need some other information please tell my. Thanks!

corny, 2016-11-22:
The [Digineo AC1200 pro](https://www.digineo.de/produkte/ac1200-pro) is also affected.

LorenzoBianconi, 2016-11-23:
Hi,

could you please paste your wireless config file (/etc/config/wireless) and
regdb configuration?
I tested channel 52@80MHz using Lede trunk, country code US and it seems to
work fine.

Regards,
Lorenzo

corny, 2016-12-06:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '100'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'VHT80'
        option country 'DE'

config wifi-iface 'lan_radio0'
        option device 'radio0'
        option mode 'ap'
        option network 'lan'
        option ssid 'LEDE'
```
Output of `logread`:
```
Tue Dec  6 21:55:31 2016 daemon.notice netifd: radio0 (2521): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Dec  6 21:55:31 2016 daemon.notice netifd: radio0 (2521): wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Tue Dec  6 21:55:31 2016 daemon.notice netifd: radio1 (2608): Configuration file: /var/run/hostapd-phy1.conf
Tue Dec  6 21:55:33 2016 kern.info kernel: [ 1775.400000] br-lan: port 2(wlan0) entered disabled state
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: interface state HT_SCAN->DFS
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): DFS start_dfs_cac() failed, -1
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): Interface initialization failed
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: interface state DFS->DISABLED
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: AP-DISABLED 
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: interface state DISABLED->DISABLED
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): wlan0: AP-DISABLED 
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Dec  6 21:55:33 2016 daemon.notice netifd: radio0 (2521): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

LorenzoBianconi, 2016-12-10:
Hi Julian,

I am running Lede-trunk on Buffalo WSR1166 and it works properly:

root@lede:~# cat /etc/config/wireless
config wifi-device 'radio0'
       option type 'mac80211'
       option channel '100'
       option hwmode '11a'
       option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
       option htmode 'VHT80'
       option country DE

config wifi-iface 'default_radio0'
       option device 'radio0'
       option network 'lan'
       option mode 'ap'
       option ssid 'LEDE-5g'
       option encryption 'psk2'
       option key 'xxxxx'

config wifi-device 'radio1'
       option type 'mac80211'
       option channel '11'
       option hwmode '11g'
       option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
       option htmode 'HT20'
       option disabled '1'

config wifi-iface 'default_radio1'
       option device 'radio1'
       option network 'lan'
       option mode 'ap'
       option ssid 'LEDE'
       option encryption 'none'

root@lede:~# iw reg get
global
country DE: DFS-ETSI
       (2400 - 2483 @ 40), (N/A, 20), (N/A)
       (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
       (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
       (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
       (5725 - 5875 @ 80), (N/A, 14), (N/A)
       (57000 - 66000 @ 2160), (N/A, 40), (N/A)

Could you please paste the output of 'iw ret get'? Moreover could you
please paste regdb.c
(<lede-dir>/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/compat-wireless-2016-10-08/net/wireless/)?
Is it properly compiled?

Couple of tests to carry out:
1- Could you please apply the patch 999-dfs_debug.patch to hostapd
(just copy it in <lede-dir>/package/network/services/hostapd/patches/)
and set CONFIG_WPA_MSG_MIN_PRIORITY to 0 from menuconfig? Could you
please paste the output of logread?
2- Could you please apply the patch 999-mac80211_start_cac_debug.patch
to mac80211 (just copy it in
<lede-dir>/package/kernel/mac80211/patches/)? Could you please paste
the output of dmesg?

Regards,
Lorenzo

Brother-Lal, 2016-12-10:
I might be able to test this, where could we find that patch?

mbaumga, 2016-12-10:
[regdb.zip](https://github.com/openwrt/mt76/files/643899/regdb.zip)

root@DLINK860L:~# iw reg get
global
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 14), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

I too would like to have the patches to carry out your requested tests.

LorenzoBianconi, 2016-12-11:
I added both patches as attached to the mail, but they are not
forwarded by GitHub. You can download them using these links:

- https://dl.dropboxusercontent.com/u/13975547/999-dfs_debug.patch
(hostapd. Please set CONFIG_WPA_MSG_MIN_PRIORITY to 0 from menuconfig)
- https://dl.dropboxusercontent.com/u/13975547/999-mac80211_start_cac_debug.patch
(mac80211)

I just added some debug logs in order to figure out what goes wrong.

Regards,
Lorenzo

mbaumga, 2016-12-11:
Built the firmware with your patches and got the following messages after setting the device on Channel 104.

dmesg: [https://gist.github.com/mbaumga/f354935eb6eaa61cc14f949c6c64b781](url)
logread: [https://gist.github.com/mbaumga/eb32b34d32c39a02de79bd5a13510557](url)

Hope it helps.

Brother-Lal, 2016-12-11:
Digineo AC1200 output is here: 

[DMESG](https://gist.github.com/Brother-Lal/72fa1ca111efdeaae6780ba67d6a4b60)
[logread](https://gist.github.com/Brother-Lal/2ccee106aca6d5a3c66f8b7e8cad7360)

LorenzoBianconi, 2016-12-11:
I guess the issue is cfg80211_check_combinations() which return
-EBUSY. Are you using just a single interface in AP mode or more than
one for a given wifi card? how many SSID?

Lorenzo

Brother-Lal, 2016-12-11:
One SSID for the 2.4ghz, 2 SSIDs for the 5Ghz.
All AP Mode.

mbaumga, 2016-12-11:
[image]

Should be 1 Interface in AP mode with one SSID at the DLink.

LorenzoBianconi, 2016-12-11:
Could you try to create just a single SSID for the AP @ 5GHz? Same issue?

Lorenzo

Brother-Lal, 2016-12-11:
Same issue, same error.

```
[ 8081.300000] device wlan0 entered promiscuous mode
[ 8081.550000] nl80211_start_radar_detection-7237
[ 8081.550000] ieee80211_start_radar_detection-2733
[ 8081.560000] ieee80211_vif_use_channel-1588: interface combination not supported (err=-16)
[ 8081.570000] ieee80211_start_radar_detection-2749: ERROR=-16
[ 8081.580000] ieee80211_start_radar_detection-2760: ret=-16
[ 8081.630000] device wlan0 left promiscuous mode
[ 8081.630000] br-lan: port 3(wlan0) entered disabled state
[ 8082.100000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 8082.150000] device wlan0 entered promiscuous mode

```

LorenzoBianconi, 2016-12-11:
I updated mac80211 patch adding more logs regarding iface combination.
Can you give it a whirl?
Thanks

https://dl.dropboxusercontent.com/u/13975547/999-mac80211_start_cac_debug.patch

Lorenzo

Brother-Lal, 2016-12-11:
Ok, another round:

[Logread](https://gist.github.com/Brother-Lal/eb0e36f3fe3b2c1f1b1d85b037304128)

Thanks for looking into it!

LorenzoBianconi, 2016-12-11:
Have you applied the patch 'mt76x2: init: unlock DFS channels', commit
b9f3193a2438ecc6807fe1600272536fb7ff5745?
Can you look at if_comb in mt76x2_init.c? how is radar_detect_widths set?

Lorenzo

mbaumga, 2016-12-11:
I would say: Works for me!

[https://gist.github.com/mbaumga/e98a1c54eb1f84207a72cf48aeaac8e8](https://gist.github.com/mbaumga/e98a1c54eb1f84207a72cf48aeaac8e8)

radar_detect_widths hasn't been set anywhere. Seems like adding it did the trick. Would be nice, if the patch becomes part of the standard LEDE build. 

Thank you a lot for this fix! The world got a bit better now. :)

Brother-Lal, 2016-12-12:
The same here, also fixed in the Digineo.
I actually downgraded the mt76 package once before to see i f it helps, because i thought that "mt76-2016-08-25" downloaded by lede is the version from 25.9(with this fix). Next time just try the newest git without looking at some dates in files...
Thanks a lot for the debugging, guess the mt76 version in LEDE should be bumped.

LorenzoBianconi, 2016-12-12:
Could you please try the latest LEDE version? (just updated)

Regards,
Lorenzo

Brother-Lal, 2016-12-12:
> Mon Dec 12 11:44:07 2016 daemon.notice netifd: radio0 (961): wlan0: DFS-CAC-COMPLETED success=1 freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0

LEDE current works like charm!
very nice that it was updated in such a short time.

nbd168, 2016-12-12:
Thanks for testing

closed 2016-12-12 by nbd168

bertjwregeer, 2017-01-02:
How can I check that I have lede current? I downloaded the latest snapshots for my device from the firmware downloads page, but wlan0 gets disabled when I set my channel to higher than 52.

schallertd, 2017-01-17:
Same here on my DIR-860L B1 with the  MT7612E chip on Designated Driver r50078.
Will patch with the radar_detect_widths fix and re-build now...

```
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: interface state COUNTRY_UPDATE->DFS
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=0, width=0, seg0=0, seg1=0, cac_time=60s
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): DFS start_dfs_cac() failed, -1
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): Interface initialization failed
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: interface state DFS->DISABLED
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: AP-DISABLED
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: Unable to setup interface.
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: interface state DISABLED->DISABLED
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): wlan0: AP-DISABLED
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Jan 17 14:40:14 2017 daemon.notice netifd: radio0 (1090): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

hdamico, 2019-11-23:
Does anyone still have the patches? Links are broken and i can't find them