openwrt/mt76 #58: WiTiBoard 2GHz wifi too slow

elkablo, 2016-11-07:
Current LEDE master compiled for WiTiBoard gives slow 2GHz wifi (MT7602E)
I have configured N mode through Luci on 20MHz (channel 11), but download speed is 2.5 MB/s at most.

nbd168, 2016-11-07:
What does "Current" mean? Please post a revision number of commit hash

elkablo, 2016-11-07:
LEDE commit 578f7b9c597b0ea9bcf0726cc681713c0672edfe
LEDE .config: http://blackhole.sk/~kabel/tmp/witi-lede-config

nbd168, 2016-11-07:
What kind of client are you using? Did you try different channels (maybe the one you're using is busy)? Please post the rate control stats from /sys/kernel/debug/ieee80211/phy*/netdev:wlan*/stations/*/rc_stats

elkablo, 2016-11-08:
You can find the rc_stats files on http://blackhole.sk/~kabel/tmp/witislow2/
names are in the form rc_stats.n_C_F_A.txt where C is channel number (I tried for 9 and 5 in this test, the default, 11, was also slow), F is frequency (20mhz or 40mhz as set in luci) and A is the average download speed (downloading a file from the router's SSD via wget).
I am testing this on my Thinkpad X230 with iwl7260 ac card.

elkablo, 2016-11-11:
Can you reproduce this error? Or should I do more tests?

nbd168, 2016-11-23:
Is the second antenna port for the 2 GHz wifi soldered and connected properly on your device? It seems that single-stream rates work well and dual-stream rates don't.
As far as I know, WiTi boards used to ship with only 2 of 4 antenna connectors soldered onto the board.

elkablo, 2016-11-23:
Yes, I have soldered the second connector and bought antenna for it. Should I remove the antenna, or do I need to unsolder the connector as well?

nbd168, 2016-12-07:
You should not remove the antenna, but I'm thinking that maybe the connector is not soldered properly

neheb, 2017-01-30:
2.4ghz on the witi does not work right for me. Adding a second antenna causes 2.4ghz to completely die. Strictly speaking that has nothing to do with mt76. Just thought I'd mention it.

nbd168, 2017-03-15:
Seems like a hardware issue to me so I'll close this report.

closed 2017-03-15 by nbd168