openwrt/mt76 #59: if the mt76 driver support mt7602e?

dtwangquan, 2016-11-16:
if the mt76 driver support mt7602e?

why my mt7602e device not work?

nbd168, 2016-11-16:
I have no idea. You'd have to provide some details.

dtwangquan, 2016-11-16:
```
 cat proc/mtd 
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "factory"
mtd3: 00fb0000 00010000 "firmware"
mtd4: 00127c9b 00010000 "kernel"
mtd5: 00e88365 00010000 "rootfs"
mtd6: 00b20000 00010000 "rootfs_data"
```
```
mtd2 is MT7602E_EEPROM.bin
# hexdump /dev/mtd2
0000000 7662 0001 6240 6463 6665 7602 14c3 0000
0000010 0000 7602 14c3 0000 0000 ffff ffff ffff
0000020 ffff d837 009d 7fff 9bfd ffff ffff ffff
0000030 ffff ffff ff22 2000 ffff 01b4 0000 0000
0000040 0000 d3ff 0000 0000 0000 0000 0000 01e0
0000050 0000 0000 ba00 0a84 001a 0000 0b83 001a
0000060 0000 067a 001a 7a00 1a06 0000 0878 001a
0000070 7e00 1a02 0000 1676 001a 7500 1a1c 0000
0000080 0a79 001a 7900 1a0a 0000 067b 001a 7700
0000090 1a14 0000 057e 001a 7900 1a11 0000 0000
00000a0 c3c3 c4c4 0000 c2c2 8282 c2c2 8282 0000
00000b0 0000 c2c2 0000 0000 0000 0000 0000 8383
00000c0 ffff ffff ffff ffff ffff ffff ffff ffff
*
00000f0 ffff ffff ffff 0000 ff00 ffff ffff ffff
0000100 ca0f c574 07e8 3d30 b001 2608 0e00 1504
0000110 8a00 4000 0000 0800 9d00 0008 c012 0000
0000120 2008 2a04 0090 2400 0401 0854 a0d0 2028
0000130 ffff ffff ffff 08ff ffff ffff ffff 0000
0000140 ffff ffff ffff ffff ffff ffff ffff ffff
*
00001e0 81c0 c382 4504 0746 0908 ffff ffff ffff
00001f0 ffff ffff ffff ffff ffff ffff ffff ffff
*
0010000
```

```
 ifconfig -a     Have wlan0 mac is right 

wlan0     Link encap:Ethernet  HWaddr 40:62:63:64:65:66  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

My dts file, cup is mt7621a
 ```
pcie@1e140000 {
        status = "okay";

        pcie0 {
            mt76@0,0 {
                reg = <0x0000 0 0 0 0>;
                device_type = "pci";
                mediatek,mtd-eeprom = <&factory 0x0000>;
                mediatek,2ghz = <0>;
            };
        };
    };
```

iw phy phy0 info
Wiphy phy0
[... 96 lines cut ...]
                          DFS state: usable (for 141 sec)
                          DFS CAC time: 60000 ms
                        * 5680 MHz [136] (20.0 dBm) (no IR, radar detection)
                          DFS state: usable (for 141 sec)
                          DFS CAC time: 60000 ms
                        * 5700 MHz [140] (20.0 dBm) (no IR, radar detection)
                          DFS state: usable (for 141 sec)
                          DFS CAC time: 60000 ms
                        * 5745 MHz [149] (20.0 dBm) (no IR)
                        * 5765 MHz [153] (20.0 dBm) (no IR)
                        * 5785 MHz [157] (20.0 dBm) (no IR)
                        * 5805 MHz [161] (20.0 dBm) (no IR)
                        * 5825 MHz [165] (20.0 dBm) (no IR)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 8,
                   total <= 8, #channels <= 1, STA/AP BI must match
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing

why My 7602e is 5G wifi?

nbd168, 2016-11-16:
You set the capability in the .dts file wrong.
`mediatek,2ghz = <0>;` 
This line disables 2.4 GHz support.

dtwangquan, 2016-11-17:
Thank you very much, the problem is solved

closed 2016-11-17 by dtwangquan

github-easyway, 2017-07-28:
i want to know how to solved this problem, i meet same problem,thanks!