openwrt/mt76 #60: MT7603 driver crash.

pparent76, 2016-11-16:
Hi,

The driver MT7603 provokes random network crashes, on ZBT-WG3526. When using the wifi chip it randomly provokes complete network failure, including wired network. After a crash I cannot connect the the device in SSH anymore because the Network does not work anymore.

I often get warning messages like this one, during normal use:

```
[  229.420000] ------------[ cut here ]------------
[  229.420000] WARNING: CPU: 0 PID: 28 at /store/buildbot/slave/ramips.mt7621/build/build_dir/target-mipsel_1004kc+dsp_musl-1.1.15/linux-ramips_mt7621/mt76-2016-06-17/mt7603_mac.c:400 mt7603_mac_fill_rx+0x2c8/0x3b0 [mt7603e]()
[  229.420000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress mt7603e mt76x2e mt76 mac80211 cfg80211 compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  229.420000] CPU: 0 PID: 28 Comm: kworker/u8:2 Tainted: G        W       4.4.14 #1
[  229.420000] Workqueue: phy1 mt7603_mac_work [mt7603e]
[  229.420000] Stack : 8f918100 80470000 803f5fa8 00000190 8eddbd10 00000001 00000001 8f31d900
[  229.420000]    00000000 80063514 803f5fa8 00000000 0000001c 804d365c 80411948 8f94baa4
[  229.420000]    80470000 8006129c 80480000 804f0000 8047e378 8047e37c 803fa8bc 8f94baa4
[  229.420000]    80470000 80040ea8 00000001 8f94badc 00000897 00000038 00000000 0094bac4
[  229.420000]    8eddd054 8fb18000 8fb18900 31796870 00000000 00000000 00000000 00000000
[  229.420000]    ...
[  229.420000] Call Trace:
[  229.420000] [<80016698>] show_stack+0x50/0x84
[  229.420000] [<801b14a4>] dump_stack+0x84/0xbc
[  229.420000] [<8002be10>] warn_slowpath_common+0xa0/0xd0
[  229.420000] [<8002bec4>] warn_slowpath_null+0x18/0x24
[  229.420000] [<8eddbd10>] mt7603_mac_fill_rx+0x2c8/0x3b0 [mt7603e]
[  229.420000] [<8eddab0c>] mt7603_queue_rx_skb+0xb8/0x100 [mt7603e]
[  229.420000] [<8ed69174>] mt76_dma_attach+0xcb4/0xdd4 [mt76]
[  229.420000] 
[  229.420000] ---[ end trace 4b46c3a11acc2699 ]---
[  229.420000] ------------[ cut here ]------------
```

I can give more details, or do more tests if you wish.

Pierre.

corny, 2016-12-19:
#62 describes the same issue

pparent76, 2017-01-16:
Hi,

The compagny Comminter ( http://www.societe.com/societe/communication-interactive-441996279.html ) would like to offer 250 euros (via paypal) for anyone who succeeds in solving issues related to the mt7603 driver on wg3526. 

Namely,
https://github.com/openwrt/mt76/issues/66
https://github.com/openwrt/mt76/issues/63
https://github.com/openwrt/mt76/issues/62
https://github.com/openwrt/mt76/issues/60

We can also offer free wg3526 hardware, if needed.

Regards.
Pierre.

jordipalet, 2017-02-12:
I think I'm having the same issue, or related.

Using LEDE Reboot 17.01.0-rc2 r3131-42f3c1f / LuCI lede-17.01 branch (git-17.032.85243-5f80115).

Almost every day I get this:

[ 3502.270000] ------------[ cut here ]------------
[ 3502.270000] WARNING: CPU: 1 PID: 2741 at /build/lede-17.01/slaves/phase1/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7621/mt76-2017-01-31-3c8caafc/mt7603_mac.c:1207 0x8edfd31c mt7603e@8edf8000+0x68a0
[ 3502.290000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_mppe ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_cake nf_conntrack sch_teql em_nbyte sch_pie sch_red sch_gred sch_dsmark cls_basic act_ipt sch_prio em_meta em_text sch_codel sch_sfq act_police em_cmp sch_fq act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables msdos ip_gre gre ifb sit tunnel6 tunnel4 ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb mmc_block usb_storage sdhci_pltfm sdhci mtk_sd mmc_core ohci_platform ohci_hcd ehci_platform ehci_hcd ext4 jbd2 mbcache crc32c_generic leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[ 3502.430000] CPU: 1 PID: 2741 Comm: kworker/u8:0 Not tainted 4.4.47 #0
[ 3502.440000] Workqueue: phy1 0x8edfd11c [mt7603e@8edf800+0x68a0]
[ 3502.440000] Stack : 8d32ab00 80420000 803a193c 000004b7 00000000 00000000 00000000 00000000
[ 3502.440000] 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3502.440000] 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3502.440000] 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3502.440000] 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3502.440000] ...
[ 3502.480000] Call Trace:[<80016448>] 0x80016448
[ 3502.480000] [<80016448>] 0x80016448
[ 3502.490000] [<801b4138>] 0x801b4138
[ 3502.490000] [<8002bf50>] 0x8002bf50
[ 3502.490000] [<8edfd31c>] 0x8edfd31c [mt7603e@8edf800+0x68a0]
[ 3502.500000] [<8002c008>] 0x8002c008
[ 3502.500000] [<8edfd31c>] 0x8edfd31c [mt7603e@8edf800+0x68a0]
[ 3502.510000] [<8000bea0>] 0x8000bea0
[ 3502.510000] [<8003f100>] 0x8003f100
[ 3502.520000] [<8003fffc>] 0x8003fffc
[ 3502.520000] [<8003fd24>] 0x8003fd24
[ 3502.520000] [<8003fd24>] 0x8003fd24
[ 3502.530000] [<800447b4>] 0x800447b4
[ 3502.530000] [<800446dc>] 0x800446dc
[ 3502.530000] [<80005478>] 0x80005478
[ 3502.540000]
[ 3502.540000] ---[ end trace f807dc80e13a4e0e ]---

rotanid, 2017-02-12:
@pparent76 i doubt that 250 euro will change anything. According to @corny (who has the info from the driver developer) the current driver is just an early prototype.

nbd168, 2017-03-15:
Please try the latest version

nbd168, 2017-03-15:
Duplicate of #32

closed 2017-03-15 by nbd168

jordipalet, 2017-03-15:
I'm using 17.01.0 since several weeks ago, and still having the same issue.

Or do you mean snapshot build Wed Mar 15 04:46:10 2017 ?If you mean something else, please provide the link so I can try it.

nbd168, 2017-03-15:
I just pushed the fixes, a snapshot hasn't been built yet. You need to test the master branch r3749-64170ab925 or newer

jordipalet, 2017-03-15:
> Responder a: openwrt/mt76 <reply@reply.github.com>
> Fecha: miércoles, 15 de marzo de 2017, 21:00

I'm using 17.01.0 since several weeks ago, and still having the same issue.

Or do you mean snapshot build Wed Mar 15 04:46:10 2017 ? If you mean something else, please provide the link so I can try it.

Saludos,
Jordi
 

-----Mensaje original-----

      
      
    

IPv4 is over
Are you ready for the new Internet ?
http://www.consulintel.es
The IPv6 Company

This electronic message contains information which may be privileged or confidential. The information is intended to be for the use of the individual(s) named above. If you are not the intended recipient be aware that any disclosure, copying, distribution or use of the contents of this information, including attached files, is prohibited.

camelcamro, 2017-03-17:
pls can someone update me, when a new trunk version was built with new fixes so, that i can test ?
in newer lede built also included ?

nbd168, 2017-03-17:
The fix is only included in LEDE, not in OpenWrt. A new snapshot build is available, so you can test now.

jordipalet, 2017-03-17:
I've tried it with the latest trunk from today, and in just a couple of hours, I already got the same problem. I will try anyway, during all the weekend.

I'm happy to test anything you need.

Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.180000] ------------[ cut here ]------------
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.180000] WARNING: CPU: 1 PID: 386 at /var/lib/buildbot/slaves/tictex-02/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-15-c91e6609/mt7603_mac.c:1228 mt7603_mac_work+0x1bc/0x284 [mt7603e]()
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.200000] Modules linked in: pppoe ppp_async pppox ppp_mppe ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 uFri Mar 17 19:57:29 2017 kern.warn kernel: [   72.330000] CPU: 1 PID: 386 Comm: kworker/u8:3 Tainted: G        W       4.4.53 #0
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.340000] Workqueue: phy0 mt7603_mac_work [mt7603e]
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] Stack : 8fed5900 80480000 80404944 000004cc 8f3ed3d8 00000000 8fc04614 8fed5918
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] 	  00000088 80064f7c 80404944 00000001 00000182 804e367c 80420404 8f701d54
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] 	  80480000 80062cc8 80490000 80500000 8048e288 8048e28c 80409258 8f701d54
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] 	  80480000 800424d0 8fc04614 8f701d8c 000005e5 00000000 00000000 00701d74
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] 	  8f3ed21c 8fe21700 8fe21800 30796870 00000000 00000000 00000000 00000000
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.350000] 	  ...
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.380000] Call Trace:
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.380000] [<8001692c>] show_stack+0x50/0x84
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.390000] [<801b7144>] dump_stack+0x84/0xbc
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.390000] [<8002cf50>] warn_slowpath_common+0xa0/0xd0
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.400000] [<8002d008>] warn_slowpath_null+0x18/0x24
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.400000] [<8f3ed3d8>] mt7603_mac_work+0x1bc/0x284 [mt7603e]
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.410000] [<80040100>] process_one_work+0x214/0x358
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.410000] [<80040ffc>] worker_thread+0x2d8/0x440
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.420000] [<800457ac>] kthread+0xd8/0xec
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.420000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.430000] 
Fri Mar 17 19:57:29 2017 kern.warn kernel: [   72.430000] ---[ end trace 5aede76181c5a682 ]---

camelcamro, 2017-03-18:
i guess it is a different bug ...
better would be to open a new ticket for it ....