openwrt/mt76 #61: MT7628 Compile error

mdp19900707, 2016-11-17:
Hi,

I downloaded OpenWrt on Nov 11,2016.So linux kernel version is 4.4.14.
`git clone https://github.com/openwrt/openwrt.git`
I updated mt76/Makefile yesterday，but i can not compile.....
package/kernel/mt76/Makefile
PKG_VERSION:=2016-11-05
PKG_SOURCE_VERSION:=892543eb7942990a06129c554125402cfb8e358c

`mt76-2016-11-05/util.h:20:28: fatal error: linux/bitfield.h: No such file or directory**`

```js
rm -f /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/.configured_*
touch /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/.configured_yyyyyyyy
make   -C "/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/linux-4.4.14" ARCH="mips" CROSS_COMPILE="mipsel-openwrt-linux-musl-" SUBDIRS="/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05" NOSTDINC_FLAGS="-I/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05 -I/op/openwrt-master/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include/mac80211-backport/uapi -I/op/openwrt-master/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include/mac80211-backport -I/op/openwrt-master/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include/mac80211/uapi -I/op/openwrt-master/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/linux-4.4.14
  CC [M]  /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/mmio.o
In file included from /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/mt76.h:25:0,
                 from /op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/mmio.c:17:
**/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/util.h:20:28: fatal error: linux/bitfield.h: No such file or directory**
compilation terminated.
make[5]: *** [/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/mmio.o] Error 1
make[4]: *** [_module_/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05] Error 2
make[4]: Leaving directory '/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/linux-4.4.14'
make[3]: *** [/op/openwrt-master/build_dir/target-mipsel_24kec+dsp_musl-1.1.15/linux-ramips_mt7628/mt76-2016-11-05/.built] Error 2
make[3]: Leaving directory '/op/openwrt-master/package/kernel/mt76'
make[2]: *** [package/kernel/mt76/compile] Error 2
make[2]: Leaving directory '/op/openwrt-master'
make[1]: *** [/op/openwrt-master/staging_dir/target-mipsel_24kec+dsp_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/op/openwrt-master'
make: *** [world] Error 2
```

I can give more details, or do more tests if you wish.

mdp19900707, 2016-11-17:
My drive is WRTnode_2P,wifi chip is MT7628.

nbd168, 2016-11-23:
OpenWrt needs a backport of bitfield support from LEDE.

closed 2016-11-23 by nbd168