openwrt/mt76 #62: ZBT-WG3526 - WIFI kernel/driver crash on high usage (mt7603_mac.c:1205 mt7603_mac_work)

camelcamro, 2016-11-25:
Steps to reproduce the error:

easy to reproduce within 1-2 minute:
1.) install trunk
2.) install iperf
3.) activate wifi and use 2,4Ghz on client connection (default config, also the same with different wifi settings)
4. generate traffic with “iperf” and you will get kernel/driver crash within 1-2 minute
(server): iperf -s -i 1 -w 6M
(client): iperf -c 192.168.1.19 -i 1 -w 6M -l 2M
`Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.350000] ------------[ cut here ]------------
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.350000] WARNING: CPU: 1 PID: 3850 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-08-25/mt7603_mac.c:1208 mt7603_mac_work+0x200/0x27c [mt7603e]()
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.370000] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_fWed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.440000] CPU: 1 PID: 3850 Comm: kworker/u8:2 Not tainted 4.4.32 #0
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000] Workqueue: phy1 mt7603_mac_work [mt7603e]
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000] Stack : 8e973880 80480000 803fdcfc 000004b8 8f21d324 00000000 8fc04614 8e973898
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000]         00000088 80063e90 803fdcfc 00000001 00000f0a 804d367c 80419780 8ea89d54
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000]         80480000 80061bdc 80480000 804f0000 80488298 8048829c 80402610 8ea89d54
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000]         80480000 80041414 8fc04614 8ea89d8c 00000447 00000000 00000000 00a89d74
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000]         8f21d124 8ff49f00 8ff49e00 31796870 00000000 00000000 00000000 00000000
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.450000]         ...
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.490000] Call Trace:
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.490000] [<800167a8>] show_stack+0x50/0x84
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.490000] [<801b5834>] dump_stack+0x84/0xbc
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.500000] [<8002bf58>] warn_slowpath_common+0xa0/0xd0
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.500000] [<8002c010>] warn_slowpath_null+0x18/0x24
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.510000] [<8f21d324>] mt7603_mac_work+0x200/0x27c [mt7603e]
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.510000] [<8003f040>] process_one_work+0x214/0x358
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.520000] [<8003ff3c>] worker_thread+0x2d8/0x440
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.520000] [<800446f4>] kthread+0xd8/0xec
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.530000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.530000]
Wed Nov 23 09:30:07 2016 kern.warn kernel: [ 2507.540000] ---[ end trace db1272735385a729 ]---

System
Hostname	lede
Model	ZBT-WG3526
Firmware Version	LEDE Reboot r0+2292 / LuCI Master (git-16.324.51057-1c27f6b) from 22.11.2016
Kernel Version	4.4.32
lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgra..> 22-Nov-2016 12:56 3932323

OpenWrt version: LEDE TRUNK

dmesg and/or logread output
What is expected vs. what actually happens

source code:
`void mt7603_mac_work(struct work_struct *work)
{
	struct mt7603_dev *dev = container_of(work, struct mt7603_dev, mac_work.work);
	struct sk_buff *skb;
	int time = MT7603_WATCHDOG_TIME;

	spin_lock_bh(&dev->status_lock);
	while ((skb = skb_peek(&dev->status_list)) != NULL) {
		struct mt7603_cb *cb = mt7603_skb_cb(skb);
		if (time_is_after_jiffies(cb->jiffies + MT7603_STATUS_TIMEOUT))
			break;

		__skb_unlink(skb, &dev->status_list);
		mt7603_skb_done(dev, skb,
				MT7603_CB_TXS_FAILED | MT7603_CB_TXS_DONE);
	}
	spin_unlock_bh(&dev->status_lock);

	mutex_lock(&dev->mutex);

	if (WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->tx_dma_check,
					       RESET_CAUSE_TX_BUSY,
					       mt7603_tx_dma_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->rx_dma_check,
					       RESET_CAUSE_RX_BUSY,
					       mt7603_rx_dma_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->tx_hang_check,
					       RESET_CAUSE_TX_HANG,
					       mt7603_tx_hang)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->rx_pse_check,
					       RESET_CAUSE_RX_PSE_BUSY,
					       mt7603_rx_pse_busy)) ||
	    WARN_ON_ONCE(mt7603_watchdog_check(dev, &dev->beacon_check,
					       RESET_CAUSE_BEACON_STUCK,
THIS LINE IS: 1208 ->
					       NULL))) {
		dev->beacon_check = 0;
		dev->tx_dma_check = 0;
		dev->tx_hang_check = 0;
		dev->rx_dma_check = 0;
		dev->rx_pse_check = 0;
		dev->rx_dma_idx = ~0;
		memset(dev->tx_dma_idx, 0xff, sizeof(dev->tx_dma_idx));
		mt7603_mac_watchdog_reset(dev);
	}`

pparent76, 2016-11-28:
Sounds similar to the bug I reported. I also experience crashes with 2.4Ghz on wg3526.

camelcamro, 2016-11-28:
i did today a lot of tests with 5GHZ wifi .... incl high load with QOS and SQM,etc....
5Ghz is stable - no kernel crash or warnings at all

so, the problem is only on 2,4Ghz Wifi.

hmm, how can we proceed now that devels can get more information to fix it ?

pparent76, 2016-11-28:
Yes I can do some test, but for now I'm not sure I have the skills to help with the development.

Do not hesitate to ask me to do any test.

corny, 2016-12-19:
I have the same Problem with the current LEDE:
```
[ 3754.000000] ------------[ cut here ]------------
[ 3754.000000] WARNING: CPU: 2 PID: 2445 at $HOME/workspace/lede/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-12-15/mt7603_mac.c:1207 mt7603_mac_work+0x200/0x274 [mt7603e]()
[ 3754.020000] Modules linked in: iptable_nat ip6table_nat nf_nat_ipv6 nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 br_netfilter em_cmp em_nbyte sch_pie sch_gred em_meta act_ipt sch_dsmark sch_teql cls_basic em_text sch_tbf sch_codel sch_red sch_htb sch_sfq act_police sch_fq sch_prio act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress mt7603e mt76x2e mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack ip6t_rpfilter ip6t_rt ip6t_frag ip6t_hbh ip6t_eui64 ip6t_mh ip6t_ah ip6t_ipv6header ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables tun leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[ 3754.160000] CPU: 2 PID: 2445 Comm: kworker/u8:2 Not tainted 4.4.38 #0
[ 3754.170000] Workqueue: phy1 mt7603_mac_work [mt7603e]
[ 3754.170000] Stack : 8eeb5980 80450000 803d771c 000004b7 8f27d31c 00000002 8fc04614 8eeb5998
[ 3754.170000]    00000088 8006354c 803d771c 00000002 0000098d 804b367c 803f2b88 8fc73d54
[ 3754.170000]    80450000 80061298 80460000 804d0000 8045e118 8045e11c 803dbfe0 8fc73d54
[ 3754.170000]    80450000 80041214 8fc04614 8fc73d8c 00000433 00000000 00000000 00c73d74
[ 3754.170000]    8f27d11c 8ff60400 8ff60900 31796870 00000000 00000000 00000000 00000000
[ 3754.170000]    ...
[ 3754.210000] Call Trace:
[ 3754.210000] [<80016684>] show_stack+0x50/0x84
[ 3754.220000] [<801abf5c>] dump_stack+0x84/0xbc
[ 3754.220000] [<8002be9c>] warn_slowpath_common+0xa0/0xd0
[ 3754.230000] [<8002bf54>] warn_slowpath_null+0x18/0x24
[ 3754.230000] [<8f27d31c>] mt7603_mac_work+0x200/0x274 [mt7603e]
[ 3754.240000] [<8003eeac>] process_one_work+0x214/0x358
[ 3754.240000] [<8003fd3c>] worker_thread+0x2d8/0x440
[ 3754.250000] [<80044034>] kthread+0xd8/0xec
[ 3754.250000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[ 3754.260000] 
[ 3754.260000] ---[ end trace 3aa9bad13c1fc06e ]---
```
```

pparent76, 2017-01-16:
Hi,

The compagny Comminter ( http://www.societe.com/societe/communication-interactive-441996279.html ) would like to offer 250 euros (via paypal) for anyone who succeeds in solving issues related to the mt7603 driver on wg3526. 

Namely,
https://github.com/openwrt/mt76/issues/66
https://github.com/openwrt/mt76/issues/63
https://github.com/openwrt/mt76/issues/62
https://github.com/openwrt/mt76/issues/60

We can also offer free wg3526 hardware, if needed.

Regards.
Pierre.

nbd168, 2017-03-15:
I've made some fixes, please test the latest version.

nbd168, 2017-03-15:
Duplicate of #32

closed 2017-03-15 by nbd168

pparent76, 2017-05-22:
Ok we tested it on LEDE snapshot release, it seems to work!

But is it compatible with openwrt, or will it be some day? Or is it only compatible with LEDE. If it is only compatible with LEDE. Or is there a way to cross-compile it for a stable version of LEDE?

If you want the reward please send an email to amasson@comminter.com and pparent@comminter.com, with your paypal account.

airend, 2017-05-22:
> But is it compatible with openwrt, or will it be some day? Or is it only compatible with LEDE. If it is only compatible with LEDE. Or is there a way to cross-compile it for a stable version of LEDE?

http://lists.infradead.org/pipermail/lede-dev/2017-May/007668.html, so yes since it sounds like the LEDE tree will become the new foundation.

pparent76, 2017-05-22:
Ok, in the mean time, I've succeeded to cross-compile the package for LEDE 17.01.1, It will still need some testing.

Unfortunately it seems unlikely that we can compile the driver for openwrt 15.05.1, right?

camelcamro, 2017-05-22:
are there the new changes/fixes in the new LEDE version ?
i can help as power tester ... as having ZBT3526 with 16 and 32MB flash.
on which LEDE version were new changes done ?
because tested with version from 1 week ago, andstill having a lot of issues (kernel panics)

airend, 2017-05-22:
> Unfortunately it seems unlikely that we can compile the driver for openwrt 15.05.1, right?

It should be possible, but it's likely that `mac80211` compat stack is outdated; LEDE's is pretty recent (`2017-01-31`), while OpenWrt's is stuck on `2016-01-10`.

pparent76, 2017-05-30:
After more testing, it seems a lot more stable than before but i'm not sure that it is really stable yet.

I sometimes get random reboots without warning or error messages before.

Sometimes I get

> [ 1518.450000] rcu_sched kthread starved for 1010 jiffies! g21367 c21366 f0x0 s3 ->state=0x1
> [ 1547.330000] INFO: rcu_sched detected stalls on CPUs/tasks:
> [ 1547.330000]  2-...: (0 ticks this GP) idle=bfa/0/0 softirq=34881/34881 fqs=0 
> [ 1547.340000]  (detected by 1, t=1003 jiffies, g=21368, c=21367, q=119412)
> [ 1547.340000] Task dump for CPU 2:
> [ 1547.350000] swapper/2       R running      0     0      1 0x00100000
> [ 1547.350000] Stack : 00000000 ae9dd275 000001ac ffffffff 0000013c 77aba2c0 804762a4 80420000
>                    8042475c 00000001 00000000 804245e0 80424724 80420000 00000000 80013554
>                    00000000 00000002 8fc72000 8fc73ec0 80420000 8005dc24 1100fc03 00000002
>                    00000000 80420000 804762a4 8005dc1c 80420000 8001acc0 1100fc03 00000000
>                    00000004 804244a0 000000a0 8001acc8 ffbdbf6f ffafb2ef d7f37b3e fb7e3fbf
>                    ...
>  [ 1547.390000] Call Trace:[<80013554>] 0x80013554
>  [ 1547.390000] [<8005dc24>] 0x8005dc24
> 

I'm not sure whether it's related or not with the driver.

All this is on wg3526 with a mt7603e driver running.

pparent76, 2017-06-27:
Today I got this on a fresh snapshot with kernel 4.4.74:

```
[ 3796.890000] ------------[ cut here ]------------
[ 3796.890000] WARNING: CPU: 3 PID: 19712 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-06-16-70d22512/mt7603_mac.c:1237 mt7603_mac_work+0x204/0x288 [mt7603e]()
[ 3796.910000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
[ 3797.050000] CPU: 3 PID: 19712 Comm: kworker/u8:0 Not tainted 4.4.74 #0
[ 3797.060000] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 3797.060000] Stack : 8dd14d80 80490000 80408a2c 000004d5 8f2ed50c 00000000 8fc04614 8dd14d98
	  00000088 80065020 80408a2c 00000003 00004d00 804f367c 80424510 8e53fd54
	  80490000 80062d6c 80490000 80510000 804942c8 804942cc 8040d34c 8e53fd54
	  80490000 80042548 8fc04614 8e53fd8c 000019e8 00000000 00000000 0053fd74
	  8f2ed308 8fff3900 8fff3800 30796870 00000000 00000000 00000000 00000000
	  ...
[ 3797.100000] Call Trace:
[ 3797.100000] [<80016920>] show_stack+0x50/0x84
[ 3797.100000] [<801bbabc>] dump_stack+0x84/0xbc
[ 3797.110000] [<8002cfac>] warn_slowpath_common+0xa0/0xd0
[ 3797.110000] [<8002d064>] warn_slowpath_null+0x18/0x24
[ 3797.120000] [<8f2ed50c>] mt7603_mac_work+0x204/0x288 [mt7603e]
[ 3797.120000] [<80040178>] process_one_work+0x214/0x358
[ 3797.130000] [<80041074>] worker_thread+0x2d8/0x440
[ 3797.130000] [<80045824>] kthread+0xd8/0xec
[ 3797.140000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[ 3797.140000] 
[ 3797.140000] ---[ end trace 6464ff107782198d ]---
[ 3861.340000] MCU message 40 (seq 15) timed out
[ 3861.340000] ------------[ cut here ]------------
[ 3861.340000] WARNING: CPU: 1 PID: 21161 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-06-16-70d22512/mt7603_mac.c:1225 mt7603_mac_work+0xe8/0x288 [mt7603e]()
[ 3861.360000] Modules linked in: xt_coova pppoe ppp_async rtl8192cu rtl8192c_common rtl_usb pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY slhc rtlwifi nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_police act_ipt act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ifb tun usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common crc32c_generic [last unloaded: xt_coova]
[ 3861.500000] CPU: 1 PID: 21161 Comm: kworker/u8:1 Tainted: G        W       4.4.74 #0
[ 3861.510000] Workqueue: phy0 mt7603_mac_work [mt7603e]
[ 3861.520000] Stack : 8d840500 80490000 80408a2c 000004c9 8f2ed3f0 00000000 8fc04614 8d840518
	  00000088 80065020 80408a2c 00000001 000052a9 804f367c 80424510 8e4b9d54
	  80490000 80062d6c 80490000 80510000 804942c8 804942cc 8040d34c 8e4b9d54
	  80490000 80042548 8fc04614 8e4b9d8c 00001b84 00000000 00000000 004b9d74
[... cut ...]
[ 3861.560000] [<8002cfac>] warn_slowpath_common+0xa0/0xd0
[ 3861.570000] [<8002d064>] warn_slowpath_null+0x18/0x24
[ 3861.570000] [<8f2ed3f0>] mt7603_mac_work+0xe8/0x288 [mt7603e]
[ 3861.580000] [<80040178>] process_one_work+0x214/0x358
[ 3861.580000] [<80041074>] worker_thread+0x2d8/0x440
[ 3861.590000] [<80045824>] kthread+0xd8/0xec
[ 3861.590000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[ 3861.600000] 
[ 3861.600000] ---[ end trace 6464ff107782198e ]---
```
It did not reboot but the wifi stopped working until next reboot.

Lotus253, 2017-08-25:
I changed local code at mt76  ( LEDE ImageBulder ( master )) and  build  kmod-mt7603_4.9.44+2017-07-17-3c4c9a64-1_arm_cortex-a7_neon-vfpv4.ipk   but can't test it because I haven't device. How can I test it ?