openwrt/mt76 #63: ZBT-WG3526 - kernel crash on multiple SSID Wifi - mt7603_mac.c:1205 mt7603_mac_work+0x1bc/0x27

camelcamro, 2016-11-25:
if I’m creating multiple SSID on 2,4Ghz .. kernel crash after a short while if I’m using it immediately ..

then kernel CPU stall ... and reboot

Thu Nov 24 13:26:23 2016 user.notice root: PPP - Adding route for lostreselementos to be reachable through pptp client 192.168.3.101
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.620000] ————[ cut here ]———— Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.620000] WARNING: CPU: 2 PID: 22 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-08-25/mt7603_mac.c:1205 mt7603_mac_work+0x1bc/0x27c [mt7603e]()
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.640000] Modules linked in: pppoe ppp_async option iptable_nat usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMThu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.800000] CPU: 2 PID: 22 Comm: kworker/u8:1 Tainted: G W 4.4.32 #0
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.810000] Workqueue: phy1 mt7603_mac_work [mt7603e]
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] Stack : 8fcd5080 80480000 803fdcfc 000004b5 8ecbd2e0 00000000 8fc04614 8fcd5098
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] 00000088 80063e90 803fdcfc 00000002 00000016 804d367c 80419780 8fcedd54
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] 80480000 80061bdc 80480000 804f0000 80488298 8048829c 80402610 8fcedd54
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] 80480000 80041414 8fc04614 8fcedd8c 00000743 00000000 00000000 00cedd74
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] 8ecbd124 8f6b6d00 8f6b6600 31796870 00000000 00000000 00000000 00000000
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.820000] ...
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.850000] Call Trace:
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.850000] [<800167a8>] show_stack+0×50/0×84 Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.860000] [<801b5834>] dump_stack+0×84/0xbc
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.860000] [<8002bf58>] warn_slowpath_common+0xa0/0xd0
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.870000] [<8002c010>] warn_slowpath_null+0×18/0×24 Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.870000] [<8ecbd2e0>] mt7603_mac_work+0x1bc/0x27c [mt7603e]
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.880000] [<8003f040>] process_one_work+0×214/0×358 Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.880000] [<8003ff3c>] worker_thread+0x2d8/0×440 Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.890000] [<800446f4>] kthread+0xd8/0xec
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.890000] [<80005478>] ret_from_kernel_thread+0×14/0x1c
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.900000]
Thu Nov 24 13:26:31 2016 kern.warn kernel: [ 126.900000] —[ end trace 90fc0aaf1a63872e ]— Thu Nov 24 13:26:32 2016 kern.warn kernel: [ 127.970000] ————[ cut here ]———— Thu Nov 24 13:26:32 2016 kern.warn kernel: [ 127.970000] WARNING: CPU: 0 PID: 0 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-08-25/mt7603_mac.c:365 mt7603_mac_fill_rx+0×124/0x3b4 [mt7603e]()

System
Hostname	lede
Model	ZBT-WG3526
Firmware Version	LEDE Reboot r0+2292 / LuCI Master (git-16.324.51057-1c27f6b) from 22.11.2016
Kernel Version	4.4.32
lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgra..> 22-Nov-2016 12:56 3932323

camelcamro, 2016-12-12:
i tested with newest trunk from 12.12.2016 13:45 ZBT3526 ... still existing ...

this is for now (more or less the last showstopper before i can use this router for production:

still getting the same issue, if I'm activating another (multiple) SSID on 2,4Ghz wifi:
```
Mon Dec 12 20:37:09 2016 kern.info kernel: [  290.440000] device wlan1 left promiscuous mode
Mon Dec 12 20:37:09 2016 kern.info kernel: [  290.440000] br-lan: port 3(wlan1) entered disabled state
Mon Dec 12 20:37:09 2016 daemon.notice netifd: Network device 'wlan1' link is down
Mon Dec 12 20:37:09 2016 daemon.notice netifd: radio1 (23703): Configuration file: /var/run/hostapd-phy1.conf
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.260000] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.270000] device wlan1 entered promiscuous mode
Mon Dec 12 20:37:11 2016 daemon.notice netifd: radio1 (23703): wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Dec 12 20:37:11 2016 daemon.notice netifd: radio1 (23703): Using interface wlan1 with hwaddr xxxxxxx and ssid "Rxxxxxxxxxx-INT"
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.630000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.630000] br-lan: port 3(wlan1) entered forwarding state
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.640000] br-lan: port 3(wlan1) entered forwarding state
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.650000] device wlan1-1 entered promiscuous mode
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.660000] IPv6: ADDRCONF(NETDEV_UP): wlan1-1: link is not ready
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.660000] br-wlan_guest: port 2(wlan1-1) entered forwarding state
Mon Dec 12 20:37:11 2016 kern.info kernel: [  292.670000] br-wlan_guest: port 2(wlan1-1) entered forwarding state
Mon Dec 12 20:37:11 2016 daemon.notice netifd: radio1 (23703): Using interface wlan1-1 with hwaddr xxxxxxxxxx and ssid "Rxxxxxxxxx-Gast"
Mon Dec 12 20:37:11 2016 kern.info kernel: [  293.030000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
Mon Dec 12 20:37:11 2016 daemon.notice netifd: radio1 (23703): wlan1: interface state COUNTRY_UPDATE->ENABLED
Mon Dec 12 20:37:11 2016 daemon.notice netifd: radio1 (23703): wlan1: AP-ENABLED
Mon Dec 12 20:37:12 2016 daemon.notice netifd: Network device 'wlan1' link is up
Mon Dec 12 20:37:12 2016 daemon.notice netifd: Network device 'wlan1-1' link is up
Mon Dec 12 20:37:12 2016 daemon.info hostapd: wlan1: STA e8:ab:fa:4b:a1:42 IEEE 802.11: authenticated
Mon Dec 12 20:37:12 2016 daemon.info hostapd: wlan1: STA e8:ab:fa:4b:a1:42 IEEE 802.11: associated (aid 1)
Mon Dec 12 20:37:12 2016 daemon.info hostapd: wlan1: STA e8:ab:fa:4b:a1:42 WPA: pairwise key handshake completed (RSN)
Mon Dec 12 20:37:13 2016 kern.info kernel: [  294.640000] br-lan: port 3(wlan1) entered forwarding state
Mon Dec 12 20:37:13 2016 kern.info kernel: [  294.670000] br-wlan_guest: port 2(wlan1-1) entered forwarding state
Mon Dec 12 20:37:14 2016 kern.warn kernel: [  296.080000] ------------[ cut here ]------------
Mon Dec 12 20:37:14 2016 kern.warn kernel: [  296.080000] WARNING: CPU: 2 PID: 22 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-11-05/mt7603_mac.c:1208 mt7603_mac_work+0x200/0x27c [mt7603e]()
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.100000] Modules linked in: pppoe ppp_async option iptable_nat usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMMon Dec 12 20:37:15 2016 kern.warn kernel: [  296.270000] CPU: 2 PID: 22 Comm: kworker/u8:1 Not tainted 4.4.36 #0
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.280000] Workqueue: phy1 mt7603_mac_work [mt7603e]
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000] Stack : 8fcd5080 80480000 803fdd1c 000004b8 8ec75324 00000001 8fc04614 8fcd5098
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000]         00000088 80063e90 803fdd1c 00000002 00000016 804d367c 804197a0 8fcedd54
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000]         80480000 80061bdc 80480000 804f0000 80486298 8048629c 80402630 8fcedd54
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000]         80480000 80041414 8fc04614 8fcedd8c 00000982 00000000 00000000 00cedd74
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000]         8ec75124 8fe61300 8fe61a00 31796870 00000000 00000000 00000000 00000000
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.290000]         ...
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.320000] Call Trace:
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.320000] [<800167a8>] show_stack+0x50/0x84
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.330000] [<801b5844>] dump_stack+0x84/0xbc
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.330000] [<8002bf58>] warn_slowpath_common+0xa0/0xd0
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.340000] [<8002c010>] warn_slowpath_null+0x18/0x24
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.340000] [<8ec75324>] mt7603_mac_work+0x200/0x27c [mt7603e]
[... cut ...]
Mon Dec 12 20:37:15 2016 kern.warn kernel: [  296.370000] ---[ end trace 1dbe47ebfc9941fb ]---
Mon Dec 12 20:37:18 2016 daemon.info hostapd: wlan1: STA a0:0b:ba:88:40:0a IEEE 802.11: authenticated
Mon Dec 12 20:37:18 2016 daemon.info hostapd: wlan1: STA a0:0b:ba:88:40:0a IEEE 802.11: associated (aid 2)
Mon Dec 12 20:37:18 2016 daemon.info hostapd: wlan1: STA a0:0b:ba:88:40:0a WPA: pairwise key handshake completed (RSN)

```
i spent a lot of time .....but still no ideas ....
if there is any idea howto solved .... to motivate  => i will donate with 50euro bonus for solving that issue :)

camelcamro, 2016-12-15:
is anyone having an idea what i can test more in detail or eg: to compile a debug LEDE version and testing with that to provide more details what is here the problem ?

psyborg55, 2016-12-21:
there was a similar bug on MT7620A maybe the cause is same.. https://dev.openwrt.org/ticket/20338

camelcamro, 2016-12-21:
maybe ... but we need someone from the devels to look into. my knowledge is not enough for it.
i guess, that is 1 of the highest prior on the MT76 
:(

psyborg55, 2016-12-21:
it is the same problem, the solution for 7620 will work if properly implemented in mt76 driver

camelcamro, 2016-12-22:
so, we need a hero which can implement that on the same principle ?

pparent76, 2017-01-16:
Hi,

The compagny Comminter ( http://www.societe.com/societe/communication-interactive-441996279.html ) would like to offer 250 euros (via paypal) for anyone who succeeds in solving issues related to the mt7603 driver on wg3526. 

Namely,
https://github.com/openwrt/mt76/issues/66
https://github.com/openwrt/mt76/issues/63
https://github.com/openwrt/mt76/issues/62
https://github.com/openwrt/mt76/issues/60

We can also offer free wg3526 hardware, if needed.

Regards.
Pierre.

amulpatel, 2017-02-06:
Hi 
I am also trying to create multiple ssid on 2.4 Gz band
I have ZBT device with MT7621 chip.
Is it possible to create multiple ssid on board??
If yes then can you please share wireless config file??

Thanks,
Amul Patel

nbd168, 2017-03-15:
This issue should be fixed in the latest version, please test

sevencheng798, 2017-03-22:
@nbd168 Can you give me a link which the lastest version to fix the problem?

camelcamro, 2017-03-22:
? 
did you never read the LEDE pages ?
see:
then you would find: 
https://lede-project.org/downloads
and there is: link to: "Development Snapshot builds"

and then you can go to your architecture to DL ...
https://downloads.lede-project.org/snapshots/targets/ramips/mt7621/

there is then the image: 
https://downloads.lede-project.org/snapshots/targets/ramips/mt7621/lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgrade.bin

sevencheng798, 2017-03-22:
@camelcamro @nbd168  i using mt7628 chip, i just test the new lastest version for mt76: the link - https://github.com/openwrt/mt76.git, but i found the cash issue still exist.

camelcamro, 2017-03-22:
which hardware are you using ?

sevencheng798, 2017-03-22:
I using WrtNode MT7628 modules. the openwrt version - 1505 r49389

camelcamro, 2017-03-22:
maybe you are here wrong ....
as this is related to the mt7621 (eg: router: ZBT 3526)
pls open a new ticket for it
mt7621 is now stable (tested with ZBT 3526 - also with multiple SSID

1 hint for you: what is crashing ?
are you having 2,4GHZ in use ?
are you having 5GHZ in use ?
are you having multiple SSID on 2,4GHZ network configured ? => if so, pls try with only 1 SSID network
are you having multiple SSID on 5GHZ network configured ? => if so, pls try with only 1 SSID network

most crashed were relatd if multiple SSID were used ....
(with only 1 SSID mine hardware didn't crashed before, too)

on 5GHZ network I didn'T have that issue as all. also with multiple SSID it was stable before, too

cu Erwin

camelcamro, 2017-03-22:
@camelcamro @nbd168 i using mt7628 chip, i just test the new lastest version for mt76: the link - https://github.com/openwrt/mt76.git, but i found the cash issue still exist.
=> you need to test LEDE version, as it is only fixed there !!!
https://downloads.lede-project.org/snapshots/targets/ramips/mt7628/

camelcamro, 2017-03-22:
it is:
https://downloads.lede-project.org/snapshots/targets/ramips/mt7628/lede-ramips-mt7628-wrtnode2p-squashfs-sysupgrade.bin

milaebrothers, 2017-05-05:
@camelcamro do you know if they are affected both WG3526 16Mb Flash and 32Mb Flash? Because I'd like to buy one of this device, and 32Mb version cost only 1,5USD more

camelcamro, 2017-05-05:
Hello,
i strongly recommend to let the finger from the 32MB device, because you can't reboot them !!!!
as long as this is not fixed with a workaround, you need to do a reboot physically.
better, buy the 16mb version and sue a USB stick (syncing monitoring to USB, lograotion to USB, etc ...) 
multiple SSID are working mostly fine (not very often a kernel crash)

if you need installation script ... i have everything inclusive:
.) huwaei modem support (this cost around 80 dollar but very good for 2/3/4g)
.) SMS GUI
.) SMS2EMAIL
.) USSD GUI
.) LOG ROTATE to USB stick
.) MULTIPLE SSID
.) VPN server (pptp)
.) FTP server
.) Load balancing
.) Traffic shaping on interfaces (multiple SSID)
.) firewall inclusive blacklist download from RBL public servers + merging)
.) monitoring incl. IP monitoring rules
.) install scripts to have not that important tools on USB stick (mc, iptraf, speedtest.net, tcpdump, etc..)
.) a lot of other features (switching modem, signal monitoring, etc..)

if interested ... i can send you ...

milaebrothers, 2017-05-08:
Hello, I ordered 16Mb Flash version and now I'll be waiting 20/30 days before receiving it. 
At now no modem needed I had to connect it to my vdsl Router via WAN port, but I'd appreciate your scripts. It would be good try a method to let start working SD module card ( maybe it would be better to open a post on Lede project Forum?)

camelcamro, 2017-05-09:
SD card is not working on the WG3526 - (is only working on ZBT 826)
but the SD card is that slow, that you can forget it ...
eg:
2MB/sec on USB stick (which cost in the small version for 32GB 6eur)
2MB/sec on USB stick with SD-card (which cost in the small version for 4eur)
eg: amazon)
0,2MB/sec on SD card on ZBT826 ...
so forget about the SD interface for now, as long as this is not fixed correct.

SD card on ZBT3526 is possible to read, but if you try to "write" you will get kernel panics and it would be hanging/crashing ...

camelcamro, 2018-01-24:
i guess this ticket can be closed.
cu camel

barhom, 2018-03-09:
@camelcamro cant you close it?

camelcamro, 2018-03-10:
Hmm, I will try...I thought only adMin can close it.

closed 2018-03-10 by camelcamro