openwrt/mt76 #64: ZBT-WG3526 - WIFI 5Ghz - only channel 36 to 48 are working

camelcamro, 2016-11-25:
easy to reproduce within 1-2 minute:
1.) install trunk
2.) activate wifi and use 5Ghz on client connection (default config, also the same with different wifi settings)
3.) change channel to 52

also playing around with other language settings and “NOSCAN 1”, but also not all channels are working.
tried DE, US, CHINA, WORLD
with some trick channel 149 is possible (eg: CHINA country) but connection then not possible.
missing other channels, too. eg: 100,102 which is working on normal manufacture firmware.
logread shows:
root@lede:~# logread -f
Wed Nov 23 10:03:03 2016 kern.info kernel: [ 1737.850000] device wlan0 left promiscuous mode
Wed Nov 23 10:03:03 2016 kern.info kernel: [ 1737.850000] br-lan: port 3(wlan0) entered disabled state
Wed Nov 23 10:03:03 2016 daemon.notice netifd: Network device ‘wlan0’ link is down
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): Configuration file: /var/run/hostapd-phy0.conf
Wed Nov 23 10:03:04 2016 kern.info kernel: [ 1739.260000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: interface state UNINITIALIZED→COUNTRY_UPDATE
Wed Nov 23 10:03:04 2016 kern.info kernel: [ 1739.270000] device wlan0 entered promiscuous mode
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: interface state COUNTRY_UPDATE→HT_SCAN
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: interface state HT_SCAN→DFS
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): DFS start_dfs_cac() failed, -1
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): Interface initialization failed
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: interface state DFS→DISABLED
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: AP-DISABLED
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: interface state DISABLED→DISABLED
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): wlan0: AP-DISABLED
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): hostapd_free_hapd_data: Interface wlan0 wasn’t started
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Nov 23 10:03:04 2016 kern.info kernel: [ 1739.610000] device wlan0 left promiscuous mode
Wed Nov 23 10:03:04 2016 kern.info kernel: [ 1739.610000] br-lan: port 3(wlan0) entered disabled state
Wed Nov 23 10:03:04 2016 daemon.notice netifd: radio0 (3067): ELOOP: remaining socket: sock=19 eloop_data=0x7712ce20 user_data=0 handler=0x4208b9
Wed Nov 23 10:03:05 2016 kern.info kernel: [ 1740.070000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:15 2016 daemon.notice netifd: radio0 (3163): Configuration file: /var/run/hostapd-phy0.conf
Wed Nov 23 10:03:16 2016 kern.info kernel: [ 1750.960000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: interface state UNINITIALIZED→COUNTRY_UPDATE
Wed Nov 23 10:03:16 2016 kern.info kernel: [ 1750.970000] device wlan0 entered promiscuous mode
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: interface state COUNTRY_UPDATE→HT_SCAN
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: interface state HT_SCAN→DFS
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): DFS start_dfs_cac() failed, -1
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): Interface initialization failed
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: interface state DFS→DISABLED
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: AP-DISABLED
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: interface state DISABLED→DISABLED
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): wlan0: AP-DISABLED
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): hostapd_free_hapd_data: Interface wlan0 wasn’t started
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Nov 23 10:03:16 2016 kern.info kernel: [ 1751.290000] device wlan0 left promiscuous mode
Wed Nov 23 10:03:16 2016 kern.info kernel: [ 1751.290000] br-lan: port 3(wlan0) entered disabled state
Wed Nov 23 10:03:16 2016 daemon.notice netifd: radio0 (3163): ELOOP: remaining socket: sock=19 eloop_data=0x7711ae20 user_data=0 handler=0x4208b9
Wed Nov 23 10:03:16 2016 kern.info kernel: [ 1751.770000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:27 2016 daemon.notice netifd: radio0 (3267): Configuration file: /var/run/hostapd-phy0.conf
Wed Nov 23 10:03:27 2016 kern.info kernel: [ 1762.650000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:27 2016 daemon.notice netifd: radio0 (3267): wlan0: interface state UNINITIALIZED→COUNTRY_UPDATE
Wed Nov 23 10:03:27 2016 kern.info kernel: [ 1762.660000] device wlan0 entered promiscuous mode
Wed Nov 23 10:03:27 2016 daemon.notice netifd: radio0 (3267): wlan0: interface state COUNTRY_UPDATE→HT_SCAN
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: interface state HT_SCAN→DFS
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): DFS start_dfs_cac() failed, -1
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): Interface initialization failed
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: interface state DFS→DISABLED
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: AP-DISABLED
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: interface state DISABLED→DISABLED
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): wlan0: AP-DISABLED
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): hostapd_free_hapd_data: Interface wlan0 wasn’t started
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Nov 23 10:03:28 2016 kern.info kernel: [ 1762.990000] device wlan0 left promiscuous mode
Wed Nov 23 10:03:28 2016 kern.info kernel: [ 1762.990000] br-lan: port 3(wlan0) entered disabled state
Wed Nov 23 10:03:28 2016 daemon.notice netifd: radio0 (3267): ELOOP: remaining socket: sock=19 eloop_data=0x776f0e20 user_data=0 handler=0x4208b9
[... 2 lines cut ...]
Wed Nov 23 10:03:39 2016 kern.info kernel: [ 1774.350000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: interface state UNINITIALIZED→COUNTRY_UPDATE
Wed Nov 23 10:03:39 2016 kern.info kernel: [ 1774.360000] device wlan0 entered promiscuous mode
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: interface state COUNTRY_UPDATE→HT_SCAN
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: interface state HT_SCAN→DFS
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): DFS start_dfs_cac() failed, -1
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): Interface initialization failed
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: interface state DFS→DISABLED
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: AP-DISABLED
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: interface state DISABLED→DISABLED
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): wlan0: AP-DISABLED
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): hostapd_free_hapd_data: Interface wlan0 wasn’t started
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Nov 23 10:03:39 2016 kern.info kernel: [ 1774.690000] device wlan0 left promiscuous mode
Wed Nov 23 10:03:39 2016 kern.info kernel: [ 1774.690000] br-lan: port 3(wlan0) entered disabled state
Wed Nov 23 10:03:39 2016 daemon.notice netifd: radio0 (3365): ELOOP: remaining socket: sock=19 eloop_data=0x772fce20 user_data=0 handler=0x4208b9
Wed Nov 23 10:03:40 2016 kern.info kernel: [ 1775.190000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

System
Hostname	lede
Model	ZBT-WG3526
Firmware Version	LEDE Reboot r0+2292 / LuCI Master (git-16.324.51057-1c27f6b) from 22.11.2016
Kernel Version	4.4.32
lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgra..> 22-Nov-2016 12:56 3932323

LorenzoBianconi, 2016-11-26:
could you please paste the output of following commands/config files?
1 - iw list
2 - iw reg get
3 - hostapd-phy0.conf

I guess the issue is 'DFS start_dfs_cac() failed, -1', and in
particular hostapd_start_dfs_cac() return -1. Have you enabled ieee80211h
in hostapd config file?
could you please enable hostapd debug log and repeat the test?

Regards,
Lorenzo

camelcamro, 2016-11-26:
here the output ...
`root@router:~# iw list
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x1f2
                        HT20/HT40
                        Static SM Power Save
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (30.0 dBm)
                        * 2417 MHz [2] (30.0 dBm)
                        * 2422 MHz [3] (30.0 dBm)
                        * 2427 MHz [4] (30.0 dBm)
                        * 2432 MHz [5] (30.0 dBm)
                        * 2437 MHz [6] (30.0 dBm)
                        * 2442 MHz [7] (30.0 dBm)
                        * 2447 MHz [8] (30.0 dBm)
                        * 2452 MHz [9] (30.0 dBm)
                        * 2457 MHz [10] (30.0 dBm)
                        * 2462 MHz [11] (30.0 dBm)
                        * 2467 MHz [12] (disabled)
                        * 2472 MHz [13] (disabled)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
[... 152 lines cut ...]
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=xxxxxxxxxxxxxxxxxxxxxx
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=LEDE-OPENWRT-WLAN-INT
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=78:a3:51:13:69:a0

root@router.LEDE-OPENWRT.com:/tmp/run$
`

the debug test and when I'm trying to activate a 2nd SSID on 2,4Ghz net .. i will send in few min ..

ignisf, 2016-11-26:
@camelcamro just FYI,
you can use 

    ```
    preformatted text here
    ```
to format your config snippets. https://help.github.com/articles/basic-writing-and-formatting-syntax/#quoting-code

camelcamro, 2016-11-26:
here my test:

i started the "guest network, and in debug mode: (also tested with ht20 and ht40 - same result)
see:

`Sat Nov 26 13:00:17 2016 daemon.debug hostapd: wlan1: STA a0:0b:ba:88:40:0a MLME: MLME-DEAUTHENTICATE.indication(a0:0b:ba:88:40:0a, 1)
Sat Nov 26 13:00:17 2016 daemon.debug hostapd: wlan1: STA e8:ab:fa:4b:a1:42 MLME: MLME-DEAUTHENTICATE.indication(e8:ab:fa:4b:a1:42, 1)
Sat Nov 26 13:00:17 2016 kern.info kernel: [  139.930000] device wlan1 left promiscuous mode
Sat Nov 26 13:00:17 2016 kern.info kernel: [  139.930000] br-lan: port 2(wlan1) entered disabled state
Sat Nov 26 13:00:17 2016 daemon.notice netifd: Network device 'wlan1' link is down
Sat Nov 26 13:00:18 2016 daemon.notice netifd: radio1 (4974): Configuration file: /var/run/hostapd-phy1.conf
Sat Nov 26 13:00:19 2016 kern.info kernel: [  141.710000] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Sat Nov 26 13:00:19 2016 kern.info kernel: [  141.720000] device wlan1 entered promiscuous mode
Sat Nov 26 13:00:19 2016 daemon.notice netifd: radio1 (4974): wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Nov 26 13:00:19 2016 daemon.notice netifd: radio1 (4974): Using interface wlan1 with hwaddr 78:a3:51:13:69:a0 and ssid "LEDE-OPENWRT-WLAN-INT"
Sat Nov 26 13:00:19 2016 kern.info kernel: [  142.080000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sat Nov 26 13:00:19 2016 kern.info kernel: [  142.090000] br-lan: port 2(wlan1) entered forwarding state
Sat Nov 26 13:00:19 2016 kern.info kernel: [  142.090000] br-lan: port 2(wlan1) entered forwarding state
Sat Nov 26 13:00:19 2016 kern.info kernel: [  142.110000] IPv6: ADDRCONF(NETDEV_UP): wlan1-1: link is not ready
Sat Nov 26 13:00:19 2016 daemon.notice netifd: radio1 (4974): Using interface wlan1-1 with hwaddr 7a:a3:51:13:69:a0 and ssid "LEDE-OPENWRT-Gast"
Sat Nov 26 13:00:20 2016 kern.info kernel: [  142.460000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
Sat Nov 26 13:00:20 2016 daemon.notice netifd: radio1 (4974): wlan1: interface state COUNTRY_UPDATE->ENABLED
Sat Nov 26 13:00:20 2016 daemon.notice netifd: radio1 (4974): wlan1: AP-ENABLED
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Network device 'wlan1' link is up
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Network device 'wlan1-1' link is up
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Interface 'wlan_guest' is enabled
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Interface 'wlan_guest' has link connectivity
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Interface 'wlan_guest' is setting up now
Sat Nov 26 13:00:20 2016 daemon.notice netifd: Interface 'wlan_guest' is now up
Sat Nov 26 13:00:20 2016 user.notice firewall: Reloading firewall due to ifup of wlan_guest (wlan1-1)
Sat Nov 26 13:00:21 2016 kern.info kernel: [  144.090000] br-lan: port 2(wlan1) entered forwarding state
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.560000] ------------[ cut here ]------------
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.560000] WARNING: CPU: 0 PID: 22 at /home/buildbot/slave-lede-local/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/mt76-2016-08-25/mt7603_mac.c:1208 mt7603_mac_work+0x200/0x27c [mt7603e]()
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.580000] Modules linked in: pppoe ppp_async option iptable_nat usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] CPU: 0 PID: 22 Comm: kworker/u8:1 Not tainted 4.4.32 #0
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] Workqueue: phy1 mt7603_mac_work [mt7603e]
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] Stack : 8fcd5080 80480000 803fdcfc 000004b8 8f345324 00000001 8fc04614 8fcd5098
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000]         00000088 80063e90 803fdcfc 00000000 00000016 804d367c 80419780 8fcedd54
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000]         80480000 80061bdc 80480000 804f0000 80488298 8048829c 80402610 8fcedd54
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000]         80480000 80041414 8fc04614 8fcedd8c 0000058f 00000000 00000000 00cedd74
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000]         8f345124 8f126400 8f126500 31796870 00000000 00000000 00000000 00000000
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000]         ...
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] Call Trace:
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<800167a8>] show_stack+0x50/0x84
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<801b5834>] dump_stack+0x84/0xbc
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<8002bf58>] warn_slowpath_common+0xa0/0xd0
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<8002c010>] warn_slowpath_null+0x18/0x24
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<8f345324>] mt7603_mac_work+0x200/0x27c [mt7603e]
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<8003f040>] process_one_work+0x214/0x358
Sat Nov 26 13:00:23 2016 kern.warn kernel: [  145.750000] [<8003ff3c>] worker_thread+0x2d8/0x440
[... cut ...]
Sat Nov 26 13:00:38 2016 kern.warn kernel: [  160.770000] ---[ end trace f50f4ff354c1f4d7 ]---
Sat Nov 26 13:00:39 2016 daemon.debug hostapd: wlan1: STA e8:ab:fa:4b:a1:42 WPA: EAPOL-Key timeout
Sat Nov 26 13:00:39 2016 daemon.debug hostapd: wlan1: STA e8:ab:fa:4b:a1:42 WPA: PTKINITNEGOTIATING: Retry limit 4 reached
Sat Nov 26 13:00:39 2016 daemon.debug hostapd: wlan1: STA e8:ab:fa:4b:a1:42 WPA: event 3 notification
Sat Nov 26 13:00:39 2016 daemon.debug hostapd: wlan1: STA e8:ab:fa:4b:a1:42 IEEE 802.1X: unauthorizing port
`

LorenzoBianconi, 2016-11-26:
> here the output ...
> `root@router:# iw list
> Wiphy phy1
> max # scan SSIDs: 4
> max scan IEs length: 2257 bytes
> max # sched scan SSIDs: 0
> max # match sets: 0
> max # scan plans: 1
> max scan plan interval: -1 [...]
why channel 36?

> ieee80211n=1
> ht_coex=0
> ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
> vht_oper_chwidth=1
> vht_oper_centr_freq_seg0_idx=42
> ieee80211ac=1
> vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][
> MAX-A-MPDU-LEN-EXP0]
>
> interface=wlan0
> ctrl_interface=/var/run/hostapd
> ap_isolate=1
> disassoc_low_ack=1
> preamble=1
> wmm_enabled=1
> ignore_broadcast_ssid=0
> uapsd_advertisement_enabled=1
> wpa_passphrase=xxxxxxxxxxxxxxxxxxxxx
> auth_algs=1
> wpa=2
> wpa_pairwise=CCMP
> ssid=LEDE-OPENWRT-WLAN-INT-5
> bridge=br-lan
> wpa_key_mgmt=WPA-PSK
> okc=0
> disable_pmksa_caching=1
> bssid=78:a3:51:13:69:a1
>
> root@router.LEDE-OPENWRT.com:/tmp/run$ cat hostapd-phy1.conf
> driver=nl80211
> logger_syslog=127
> logger_syslog_level=2
> logger_stdout=127
> logger_stdout_level=2
> country_code=US
> ieee80211d=1
> hw_mode=g
> channel=8
>
> ieee80211n=1
> ht_coex=0
> ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
>
> interface=wlan1
> ctrl_interface=/var/run/hostapd
> ap_isolate=1
> disassoc_low_ack=1
> preamble=1
> wmm_enabled=1
> ignore_broadcast_ssid=0
> uapsd_advertisement_enabled=1
> wpa_passphrase=xxxxxxxxxxxxxxxxxxxxxx
> auth_algs=1
> wpa=2
> wpa_pairwise=CCMP
> ssid=LEDE-OPENWRT-WLAN-INT
> bridge=br-lan
> wpa_key_mgmt=WPA-PSK
> okc=0
> disable_pmksa_caching=1
> bssid=78:a3:51:13:69:a0
>
> root@router.LEDE-OPENWRT.com:/tmp/run$
> `
>
> the debug test and when I'm trying to activate a 2nd SSID on 2,4Ghz net ..
> i will send in few min ..
>
> <https://github.com/openwrt/mt76/issues/64#issuecomment-263058901>, or mute
> the thread

camelcamro, 2016-11-26:
hmm, i didn'T see , that i have to mark/select first the range, and then press "code :)
thx for the hint :)

camelcamro, 2016-11-26:
> why channel 36?
because it is free .. on 5Ghz it is not a problem
and in austria i can only use with that firmware 36 to 48.
all other channels not possible here or wifi then not working anymore.
if wanted i can also change to 40 or whatever ....
i tested every channel with iperf in both direction, i had no difference at all (ok, channel 48 was slower, but that is maybe related to HT40/80 +-, which is then taking the channels related on the "+/-"

LorenzoBianconi, 2016-11-27:
I guess we should consider a problem a time in order to not squash
different issues, let's consider DFS channel one (as outlined in
thread title), ok?

1- Are you using latest version of mac80211/mt76/hostapd from Lede repository?
2- Could you please use the simplest configuration, just one interface
configured in AP mode, channel 5500@80MHz?
3- Could you please start hostapd in interactive mode (not use -B
option) with wpa_debug_level set to 'debug level' and
CONFIG_MSG_MIN_PRIORITY = 0 from menuconfig. Moreover you can use the
following patch for hostapd in daemon mode and CONFIG_MSG_MIN_PRIORITY
= 0

--- a/src/utils/wpa_debug.c
+++ b/src/utils/wpa_debug.c
@@ -29,7 +29,7 @@ static FILE *wpa_debug_tracing_file = NU
 #endif /* CONFIG_DEBUG_LINUX_TRACING */

-int wpa_debug_level = MSG_INFO;
+int wpa_debug_level = MSG_DEBUG;
 int wpa_debug_show_keys = 0;
 int wpa_debug_timestamp = 0;

could you please paste the output of logread?

4- Could you please dump mt76 eeprom? (cat
/sys/kernel/debug/ieee80211/phy0/mt76/eeprom)

Regards,
Lorenzo

camelcamro, 2016-11-28:
see:

camelcamro, 2016-11-28:
sorry, i missed the topic, and i thought it is related to "buggy 2,4Ghz wifi driver" ....

here i changed to channel 100 (5500), and then i got in logs:

```
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state COUNTRY_UPDATE->DFS
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): DFS start_dfs_cac() failed, -1
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): Interface initialization failed
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state DFS->DISABLED
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: AP-DISABLED
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: Unable to setup interface.
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state DISABLED->DISABLED
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan_g5: AP-DISABLED
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): hostapd_free_hapd_data: Interface wlan_g5 wasn't started
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: AP-DISABLED
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): hostapd_free_hapd_data: Interface wlan0 wasn't started
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): ELOOP: remaining socket: sock=20 eloop_data=0x77258ed0 user_data=0 handler=0x41a3f5
Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): Device setup failed: HOSTAPD_START_FAILED
Mon Nov 28 16:36:15 2016 user.notice SQM: Stopping SQM on wlan_g5
Mon Nov 28 16:36:15 2016 user.notice SQM: ERROR: wlan_g5 does currently not exist, not even trying to start SQM on nothing.
Mon Nov 28 16:36:15 2016 user.notice SQM: ERROR: wlan_g5 does currently not exist, not even trying to start SQM on nothing.
Mon Nov 28 16:36:18 2016 daemon.info dnsmasq-dhcp[27239]: DHCPREQUEST(br-lan) 192.168.2.51 00:90:f5:d9:9a:fa
Mon Nov 28 16:36:18 2016 daemon.info dnsmasq-dhcp[27239]: DHCPACK(br-lan) 192.168.2.51 00:90:f5:d9:9a:fa notebook_lan_erwin
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Configuration file: /var/run/hostapd-phy0.conf
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state COUNTRY_UPDATE->DFS
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): DFS start_dfs_cac() failed, -1
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Interface initialization failed
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state DFS->DISABLED
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: AP-DISABLED
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: Unable to setup interface.
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state DISABLED->DISABLED
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan_g5: AP-DISABLED
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): hostapd_free_hapd_data: Interface wlan_g5 wasn't started
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: AP-DISABLED
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): hostapd_free_hapd_data: Interface wlan0 wasn't started
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): ELOOP: remaining socket: sock=20 eloop_data=0x77850ed0 user_data=0 handler=0x41a3f5
Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Device setup failed: HOSTAPD_START_FAILED
 
```

LorenzoBianconi, 2016-11-28:
>
>
> sorry, i missed the topic, and i thought it is related to "buggy 2,4Ghz
> wifi driver" ....

no worries :)

> here i changed to channel 100 (5500), and then i got in logs:
>
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state COUNTRY_UPDATE->DFS
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): DFS start_dfs_cac() failed, -1
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): Interface initialization failed
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state DFS->DISABLED
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: AP-DISABLED
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: Unable to setup interface.
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: interface state DISABLED->DISABLED
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan_g5: AP-DISABLED
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): hostapd_free_hapd_data: Interface wlan_g5 wasn't started
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): wlan0: AP-DISABLED
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): hostapd_free_hapd_data: Interface wlan0 wasn't started
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): ELOOP: remaining socket: sock=20 eloop_data=0x77258ed0 user_data=0 handler=0x41a3f5
> Mon Nov 28 16:36:13 2016 daemon.notice netifd: radio0 (27262): Device setup failed: HOSTAPD_START_FAILED
> Mon Nov 28 16:36:15 2016 user.notice SQM: Stopping SQM on wlan_g5
> Mon Nov 28 16:36:15 2016 user.notice SQM: ERROR: wlan_g5 does currently not exist, not even trying to start SQM on nothing.
> Mon Nov 28 16:36:15 2016 user.notice SQM: ERROR: wlan_g5 does currently not exist, not even trying to start SQM on nothing.
> Mon Nov 28 16:36:18 2016 daemon.info dnsmasq-dhcp[27239]: DHCPREQUEST(br-lan) 192.168.2.51 00:90:f5:d9:9a:fa
> Mon Nov 28 16:36:18 2016 daemon.info dnsmasq-dhcp[27239]: DHCPACK(br-lan) 192.168.2.51 00:90:f5:d9:9a:fa notebook_lan_erwin
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Configuration file: /var/run/hostapd-phy0.conf
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state COUNTRY_UPDATE->DFS
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): DFS start_dfs_cac() failed, -1
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Interface initialization failed
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state DFS->DISABLED
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: AP-DISABLED
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: Unable to setup interface.
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: interface state DISABLED->DISABLED
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan_g5: AP-DISABLED
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): hostapd_free_hapd_data: Interface wlan_g5 wasn't started
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): wlan0: AP-DISABLED
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): hostapd_free_hapd_data: Interface wlan0 wasn't started
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): nl80211: deinit ifname=wlan0 disabled_11b_rates=0
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): ELOOP: remaining socket: sock=20 eloop_data=0x77850ed0 user_data=0 handler=0x41a3f5
> Mon Nov 28 16:36:24 2016 daemon.notice netifd: radio0 (27709): Device setup failed: HOSTAPD_START_FAILED
>
>
> <https://github.com/openwrt/mt76/issues/64#issuecomment-263304461>, or mute
> the thread

you have not enabled the debug log yet. It will allow us to get a more
precise idea why CAC start fails ;)
To do it in a 'quick and dirty' way just apply the patch and
set CONFIG_MSG_MIN_PRIORITY = 0 from menuconfig

Moreover could you please dump mt76 eeprom in this way:

cat /sys/kernel/debug/ieee80211/phy0/mt76/eeprom > /tmp/eeprom

Regards,
Lorenzo

camelcamro, 2016-11-29:
i didn'T recompile and install a own built. also tried to install hostapd. but no difference

eeprom attached

camelcamro, 2016-11-29:
simply rename it, as ZIP not possible to upload, so, i renamed the bin EEPROm to PNG

[image]

camelcamro, 2016-12-07:
@LorenzoBianconi : anything else what i can provide  to dig into ?

LorenzoBianconi, 2016-12-10:
Could you please paste regdb.c
(<lede-dir>/build_dir/target-mipsel_24kc_musl-1.1.15/linux-ramips_mt7621/compat-wireless-2016-10-08/net/wireless/)?
Is it properly compiled?

Couple of tests to carry out:
1- Could you please apply the patch 999-dfs_debug.patch to hostapd
(just copy it in <lede-dir>/package/network/services/hostapd/patches/)
and set CONFIG_WPA_MSG_MIN_PRIORITY to 0 from menuconfig? Could you
please paste the output of logread?
2- Could you please apply the patch 999-mac80211_start_cac_debug.patch
to mac80211 (just copy it in
<lede-dir>/package/kernel/mac80211/patches/)? Could you please paste
the output of dmesg?

Regards,
Lorenzo

camelcamro, 2016-12-10:
hmm, can you build such kind of package ?
and also the patches are missing or where to DL ?

LorenzoBianconi, 2016-12-10:
>
> hmm, can you build such kind of package ?
> and also the patches are missing or where to DL ?

Patches are attached to the mail. Are you running Lede trunk? Have you
compiled the fw once?

Lorenzo

> <https://github.com/openwrt/mt76/issues/64#issuecomment-266205158>, or mute
> the thread

LorenzoBianconi, 2016-12-10:
Can you provide me your email address? Github does not allow me to send the
fw image since the attached file exceed max allowed size.

Lorenzo

camelcamro, 2016-12-11:
Running lede Trunk from 9.12.2016
Do currently use not own compiled.
I already compiled a lede version for mt7601 - zbt3526 only (as on the lede trunk, the sd card patch not fully added).

camelcamro, 2016-12-12:
tried newest LEDE trunk, and seems to be working.
thx to LorenzoBianconi which supported me to test and for fixing the issues :)

not 100% clear when it is working.
but all channels now working (exclusive 116 to 140)

of course as based on the architecture:
 -> range 1: channel: 34 to 50 => OK
 -> range 2 : channel: 98 to 114 => OK

 -> range 3: channel 116 to 140 is not working - maybe related to country: "AT" or hardware (ZBT:3526) ?

```

Mon Dec 12 19:49:41 2016 kern.info kernel: [  426.340000] device wlan0 left promiscuous mode
Mon Dec 12 19:49:41 2016 kern.info kernel: [  426.340000] br-lan: port 2(wlan0) entered disabled state
Mon Dec 12 19:49:41 2016 daemon.notice netifd: Network device 'wlan0' link is down
Mon Dec 12 19:49:41 2016 daemon.info dnsmasq[5646]: read /etc/hosts - 4 addresses
Mon Dec 12 19:49:41 2016 daemon.info dnsmasq[5646]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Mon Dec 12 19:49:41 2016 daemon.info dnsmasq-dhcp[5646]: read /etc/ethers - 0 addresses
Mon Dec 12 19:49:42 2016 daemon.notice netifd: radio0 (28272): Configuration file: /var/run/hostapd-phy0.conf
Mon Dec 12 19:49:42 2016 kern.info kernel: [  427.730000] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mon Dec 12 19:49:42 2016 kern.info kernel: [  427.740000] device wlan0 entered promiscuous mode
Mon Dec 12 19:49:42 2016 daemon.notice netifd: radio0 (28272): wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Dec 12 19:49:42 2016 daemon.notice netifd: radio0 (28272): wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Mon Dec 12 19:49:43 2016 daemon.notice netifd: radio0 (28272): Using interface wlan0 with hwaddr xxxxxxx and ssid "xxxxxxx"
Mon Dec 12 19:49:44 2016 kern.info kernel: [  429.340000] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mon Dec 12 19:49:44 2016 kern.info kernel: [  429.350000] br-lan: port 2(wlan0) entered forwarding state
Mon Dec 12 19:49:44 2016 kern.info kernel: [  429.350000] br-lan: port 2(wlan0) entered forwarding state
Mon Dec 12 19:49:44 2016 daemon.notice netifd: radio0 (28272): wlan0: interface state HT_SCAN->ENABLED
Mon Dec 12 19:49:44 2016 daemon.notice netifd: radio0 (28272): wlan0: AP-ENABLED
Mon Dec 12 19:49:44 2016 daemon.notice netifd: Network device 'wlan0' link is up
Mon Dec 12 19:49:46 2016 kern.info kernel: [  431.350000] br-lan: port 2(wlan0) entered forwarding state

```

closed 2016-12-13 by nbd168