openwrt/mt76 #65: ZBT-WG3526 - WIFI 5Ghz - throughput is very poor

camelcamro, 2016-11-25:
easy to reproduce within 1-2 minute:

1.) install trunk
2.) activate wifi and use 5Ghz on client connection (default config, also the same with different wifi settings)
3.) generate traffic with “iperf” and you will get low traffic throughput
(server): iperf -s -i 1 -w 6M
(client): iperf -c 192.168.1.19 -i 1 -w 6M -l 2M

on openwrt it shows for this client:
300.0 Mbit/s, 40MHz, MCS 15, Short GI
300.0 Mbit/s, 40MHz, MCS 15, Short GI

on openwrt interface it shows: Channel: 36 (5.180 GHz) | Bitrate: 270 Mbit/s

client to server:

C:\Users\Camel>iperf -c 192.168.1.1 -i 1 -w 6M -l 2M

Client connecting to 192.168.1.1, TCP port 5001
TCP window size: 6.00 MByte

[ 3] local 192.168.1.196 port 16821 connected with 192.168.1.1 port 5001
[ ID] Interval Transfer Bandwidth
[ 3] 0.0- 1.0 sec 24.0 MBytes 201 Mbits/sec
[ 3] 1.0- 2.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 2.0- 3.0 sec 16.0 MBytes 134 Mbits/sec
[ 3] 3.0- 4.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 4.0- 5.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 5.0- 6.0 sec 16.0 MBytes 134 Mbits/sec
[ 3] 6.0- 7.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 7.0- 8.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 8.0- 9.0 sec 16.0 MBytes 134 Mbits/sec
[ 3] 9.0-10.0 sec 18.0 MBytes 151 Mbits/sec
[ 3] 0.0-10.0 sec 180 MBytes 150 Mbits/sec

server to client:

root@lede:~# iperf -c 192.168.1.196 -i 1 -w 6M -l 2M

Client connecting to 192.168.1.196, TCP port 5001
TCP window size: 320 KByte (WARNING: requested 6.00 MByte)

[ 3] local 192.168.1.1 port 47600 connected with 192.168.1.196 port 5001
[ ID] Interval Transfer Bandwidth
[ 3] 0.0- 1.0 sec 12.0 MBytes 101 Mbits/sec
[ 3] 1.0- 2.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 2.0- 3.0 sec 12.0 MBytes 101 Mbits/sec
[ 3] 3.0- 4.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 4.0- 5.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 5.0- 6.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 6.0- 7.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 7.0- 8.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 8.0- 9.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 9.0-10.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 0.0-10.2 sec 100 MBytes 82.6 Mbits/sec

System
Hostname	lede
Model	ZBT-WG3526
Firmware Version	LEDE Reboot r0+2292 / LuCI Master (git-16.324.51057-1c27f6b) from 22.11.2016
Kernel Version	4.4.32
lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgra..> 22-Nov-2016 12:56 3932323

camelcamro, 2016-11-28:
based on the missing hardware of MIMO hardware on my devices .... maybe it is normal to have oly that throughput ... hmm

nbd168, 2017-03-15:
Throughput should be a lot better with the latest LEDE version

closed 2017-03-15 by nbd168