openwrt/mt76 #66: ZBT-WG3526 - WIFI 2,4Ghz - throughput is very poor

camelcamro, 2016-11-25:
easy to reproduce within 1-2 minute:
1.) install trunk
2.) install iperf
3.) activate wifi and use 2,4Ghz on client connection (default config, also the same with different wifi settings)
4. generate traffic with “iperf” and you will get kernel/driver crash within 1-2 minute
(server): iperf -s -i 1 -w 6M
(client): iperf -c 192.168.1.19 -i 1 -w 6M -l 2M

on openwrt it shows for this client:
130.0 Mbit/s, 20MHz, MCS 15
117.0 Mbit/s, 20MHz, MCS 14

on openwrt interface it shows: Channel: 6 (2.437 GHz) | Bitrate: 130 Mbit/s

client to server:

C:\Users\Camel>iperf -c 192.168.1.1 -i 1 -w 6M -l 2M

Client connecting to 192.168.1.1, TCP port 5001
TCP window size: 6.00 MByte

[ 3] local 192.168.1.196 port 15896 connected with 192.168.1.1 port 5001
[ ID] Interval Transfer Bandwidth
[ 3] 0.0- 1.0 sec 14.0 MBytes 117 Mbits/sec
[ 3] 1.0- 2.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 2.0- 3.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 3.0- 4.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 4.0- 5.0 sec 6.00 MBytes 50.3 Mbits/sec
[ 3] 5.0- 6.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 6.0- 7.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 7.0- 8.0 sec 10.0 MBytes 83.9 Mbits/sec
[ 3] 8.0- 9.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 9.0-10.0 sec 8.00 MBytes 67.1 Mbits/sec
[ 3] 0.0-10.4 sec 86.0 MBytes 69.4 Mbits/sec

server to client:

root@lede:~# iperf -c 192.168.1.196 -i 1 -w 6M -l 2M

Client connecting to 192.168.1.196, TCP port 5001
TCP window size: 320 KByte (WARNING: requested 6.00 MByte)

[ 3] local 192.168.1.1 port 47592 connected with 192.168.1.196 port 5001
[ ID] Interval Transfer Bandwidth
[ 3] 0.0- 1.0 sec 2.21 MBytes 18.6 Mbits/sec
[ 3] 1.0- 2.0 sec 700 KBytes 5.73 Mbits/sec
[ 3] 2.0- 3.0 sec 1.37 MBytes 11.5 Mbits/sec
[ 3] 3.0- 4.0 sec 891 KBytes 7.30 Mbits/sec
[ 3] 4.0- 5.0 sec 1.18 MBytes 9.90 Mbits/sec
[ 3] 5.0- 6.0 sec 1.43 MBytes 12.0 Mbits/sec
[ 3] 6.0- 7.0 sec 891 KBytes 7.30 Mbits/sec
[ 3] 7.0- 8.0 sec 1.68 MBytes 14.1 Mbits/sec
[ 3] 8.0- 9.0 sec 2.42 MBytes 20.3 Mbits/sec
[ 3] 9.0-10.0 sec 2.49 MBytes 20.9 Mbits/sec
[ 3] 0.0-10.1 sec 15.2 MBytes 12.6 Mbits/sec
root@lede:~#

System
Hostname	lede
Model	ZBT-WG3526
Firmware Version	LEDE Reboot r0+2292 / LuCI Master (git-16.324.51057-1c27f6b) from 22.11.2016
Kernel Version	4.4.32
lede-ramips-mt7621-zbt-wg3526-squashfs-sysupgra..> 22-Nov-2016 12:56 3932323

psyborg55, 2016-12-04:
not sure but i would guess all mt76xx chips that work on 2.4GHz are using rt2x00 driver from trunk, not mt76 driver. that is why 2.4GHz wifi is weak on all these boards.

kofec, 2016-12-04:
Do you know how to force use mt76 driver beside rt2x00 ?

nbd168, 2016-12-04:
There's no need to force anything, since there's no overlap in chipset support between rt2x00 and mt76. MT7603 and MT7602 are only supported by mt76, not rt2x00.

camelcamro, 2016-12-04:
hmm, strange, that mt76 is then so buggy ... and multiple SSID would crash kernel very quick

nbd168, 2016-12-04:
mt7603 and mt76x2 are different drivers of the mt76 family. mt76x2 tends to be more stable than mt7603

psyborg55, 2016-12-04:
strange there is no better reception possible with any of the two, is it due to the lack of code or something to do with newly introduced register RFRX?

the average with his mt7603 12.6 Mbits/sec

and with mt7620a i got around 30 Mbits/sec

pparent76, 2017-01-16:
Hi,

The compagny Comminter ( http://www.societe.com/societe/communication-interactive-441996279.html ) would like to offer 250 euros (via paypal) for anyone who succeeds in solving issues related to the mt7603 driver on wg3526. 

Namely,
https://github.com/openwrt/mt76/issues/66
https://github.com/openwrt/mt76/issues/63
https://github.com/openwrt/mt76/issues/62
https://github.com/openwrt/mt76/issues/60

We can also offer free wg3526 hardware, if needed.

Regards.
Pierre.

MaFeSa, 2017-04-06:
Same problem also with DIR-860L B1

barhom, 2018-08-12:
There has been a lot of improvement on mt7603 since 25 Nov 2016.
Please close this ticket.

closed 2018-11-28 by nbd168