openwrt/mt76 #67: ZBT-WG3526 - mt7621 - SD card (MMC) support - [solved] but not already added to TRUNK ??

camelcamro, 2016-11-28:
Hello,
text copied from a source which solved it already with a patch.
this i also tested and it is working fine. ZBT-826 ZBT 3526
there that is already solved, but still not in trunk and under dev.openwrt is not possible to register.
maybe your team can merge it into trunk ?

see
https://dev.openwrt.org/ticket/21834

text from source:
```

Changed 3 weeks ago by shmygov
↳ Reply
I prepared the attached two patches which seem to solve the problem.
My ZBT-WG3526 router is working fine with these patches, including SD card function.
Here is how to use the patches:
Get the OpenWrt source as usual:
git clone ​git://github.com/openwrt/openwrt.git
I tested the patches using OpenWrt trunk, version r49373 (commit ce116bc6f997d8d6e6b976cacce5d4c60d705fc6)
Download the attached patches:
001-openwrt-ramips-mt7621-set-ralink-soc.patch
002-openwrt-ramips-mt7621-fix-wg3526-sd-card.patch
and copy them to the root directory of OpenWrt source.
Apply the patches:
patch -p1 -i 001-openwrt-ramips-mt7621-set-ralink-soc.patch
patch -p1 -i 002-openwrt-ramips-mt7621-fix-wg3526-sd-card.patch
If you have already compiled OpenWrt (there is a build_dir in your source tree), then you should run the following command to apply the patches to the kernel code in the build_dir:
make target/linux/{clean,prepare} V=s
Compile OpenWrt the usual way:
./scripts/feeds update -a
Clear the /tmp folder
make menuconfig
In the configuration menu, you need to select the following options (along with other options of your choice):
Target System (Ralink RT288x/RT3xxx)
Subtarget (MT7621 based boards)
Target Profile (Default Profile)
Kernel modules --> Other modules ---> <*> kmod-sdhci-mt7620
Finally, build the firmware:
make
```
patch files:
[https://dev.openwrt.org/raw-attachment/ticket/21834/001-openwrt-ramips-mt7621-set-ralink-soc.patch]
[https://dev.openwrt.org/raw-attachment/ticket/21834/002-openwrt-ramips-mt7621-fix-wg3526-sd-card.patch]
#################################################
FYI:
as in openwrt ticketing system it is not possible to post ...
pls can you help to put that into trunk, as it is not possible to post on openwrt (Captcha Error)
see:
https://dev.openwrt.org/ticket/21834
and also see:
https://bugs.lede-project.org/index.php?do=details&task_id=307&order=id&sort=desc

camelcamro, 2017-01-12:
@Devels: can anyone implemement that fix ? should be not that complicated, right ?

nbd168, 2017-01-12:
This has nothing to do with mt76, you opened up the bug report in the wrong repository

closed 2017-01-12 by nbd168

camelcamro, 2017-01-12:
hmm, lede is telling the same ?
maybe someone can take care which repo is responsible for it ... :(