openwrt/mt76 #68: mt76x2: changing tx power has no effect

MrBan, 2016-12-07:
I've been doing some test changing the radio TX power (`iw dev wlan0 set txpower ...`) and measuring the output power with a spectrum analyzer.

I've noticed that even setting the TX power level at very different values (e.g. 1 dBm vs 20 dBm) the output power doesn't change accordingly.

Analyzing the code in `mt76x2_phy_set_txpower()` I found something that's not clear to me. The code that deals with the *AGC* is straightforward, but I don't understand the portion concerning the *rate power*:

```c
mt76x2_add_rate_power_offset(&t, txp.chain[0].target_power +
			     txp.chain[0].delta);
mt76x2_limit_rate_power(&t, dev->txpower_conf);
mt76x2_add_rate_power_offset(&t, -(txp.chain[0].target_power +
			     txp.chain[0].delta + delta));
```

in particular I can't understand why `delta` is subtracted to the `rate powers` in the last line.

For example, a negative `delta` means that the user is trying to set a TX power lower than the target power. In this case the *AGC* values are decreased (as expected), instead here we are increasing the *rate powers*.

Is there something I'm missing?

psyborg55, 2017-01-05:
isn't delta just some sort of txpower offset, if your target power is 17dBm and you get regression at MCS15 normally you'll fill a negative value to get e.g 15dBm with MCS15?

the txpower change itself can still be problematic, first i've noticed it doesn't work then after some testing i figured out it wouldn't load set value just on upgrade/reboot but i had to manually change txpower from luci for the change to take effect.

jkalmus, 2017-01-25:
How did you manage to change the txpower  ? 

I tried the following:
`iw dev wlan0 set txpower fixed 2000`
`iw phy phy0 set txpower fixed 2000`

I also tried changing it from Luci, but none of the above works. 
On one device the txpower is fixed to 11dBm, on another 8dBm, and another device shows 7dBm..!! and I cannot change it. 

Device: ZBT WG3526 (LEDE Reboot r2013)

psyborg55, 2017-01-26:
i don't remember exactly. i've formatted drive containign
 buildroot that i used to build images when i was testing that...

psyborg55, 2017-02-28:
update: unless the code is broken somewhere due to various changes, txpower change will be limited by eeprom target power. this limit is present at addresses: 0x64, 0x69... and 0xF8

nbd168, 2018-02-24:
Fixed in the latest version

closed 2018-02-24 by nbd168