openwrt/mt76 #72: Cannot set up secure 802.11s mesh

jkalmus, 2017-01-11:
My hardware:
ZBT WG3526 WiFi router with 5GHz MT7612E WiFi module. 
I built and installed the LEDE trunk version.

I am trying to set up a secure mesh using wpa_supplicant the following way: 

1. I created the following wpa_supplicant.conf configuration file: 
```
# secure mesh network
network={
        ssid="test_mesh" 
        mode=5
        frequency=5180
        key_mgmt=SAE
        psk=”12345678”
}
```

2. I set up a mesh using the following commands on two nodes:
```
iw phy phy0 interface add wlan0 type mp
ifconfig wlan0 up
wpa_supplicant -B -i wlan0 -c ./wpa_supplicant.conf
```

3. Output of `iw dev wlan0 station dump`:
```
root@lede:~# iw dev wlan0 station dump
Station 78:a3:51:26:43:7d (on wlan0)
	inactive time:	730 ms
	rx bytes:	5066
	rx packets:	53
	tx bytes:	910
	tx packets:	6
	tx retries:	1
	tx failed:	0
	signal:  	-31 [-31, -37] dBm
	signal avg:	-32 [-32, -38] dBm
	Toffset:	24576540 us
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	6.5 MBit/s MCS 0
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	connected time:	24 seconds
```

The nodes seem to detect each other but I cannot ping the nodes. 
Anyone has got the same issue ?

bittorf, 2017-01-11:
do you have authsae support installed?
please give output of 'uci show wireless' and 'uci show network'

jkalmus, 2017-01-11:
I have the wpad-mesh package install, which I guess is enough to support 802.11s secure mesh.
I have tried the same configuration on a TP-Link Archer C7 and the secure mesh works with wpad-mesh. The only different is the wireless driver: ath10k on the TP-Link, mt76 on the ZBT. 

Here is my config: 
root@lede:~# uci show wireless
```
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='11'
wireless.radio1.hwmode='11g'
wireless.radio1.path='pci0000:00/0000:00:00.0/0000:01:00.0'
wireless.radio1.htmode='HT20'
wireless.radio1.disabled='1'
```
root@lede:~# uci show network
```
network.loopback=interface
network.loopback.ifname='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd2e:9fd7:d7aa::/48'
network.lan=interface
network.lan.ifname='eth0.1'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan_dev=device
network.lan_dev.name='eth0.1'
network.lan_dev.macaddr='78:a3:51:26:45:82'
network.wan=interface
network.wan.ifname='eth0.2'
network.wan.proto='static'
network.wan.ipaddr='192.168.0.131'
network.wan.netmask='255.255.255.0'
network.wan.gateway='192.168.0.1'
network.wan.broadcast='192.168.0.255'
network.wan.dns='192.168.0.1'
network.wan_dev=device
network.wan_dev.name='eth0.2'
network.wan_dev.macaddr='78:a3:51:26:45:83'
network.wan6=interface
network.wan6.ifname='eth0.2'
network.wan6.proto='dhcpv6'
network.@switch[0]=switch
network.@switch[0].name='switch0'
network.@switch[0].reset='1'
network.@switch[0].enable_vlan='1'
network.@switch_vlan[0]=switch_vlan
network.@switch_vlan[0].device='switch0'
network.@switch_vlan[0].vlan='1'
network.@switch_vlan[0].ports='0 1 2 3 6t'
network.@switch_vlan[1]=switch_vlan
network.@switch_vlan[1].device='switch0'
network.@switch_vlan[1].vlan='2'
network.@switch_vlan[1].ports='4 6t'
```

bittorf, 2017-01-11:
you have both radios disabled and there is no mesh-password set...what are you doing?

jkalmus, 2017-01-12:
LoL
For your information, there are different ways to set up a mesh network in OpenWRT: one is through the /etc/config/wireless file (which I think you must be familiar with), another is by using "iw ... mesh join ..." commands, and another is using wpa_supplicant. 

If you read my very first post carefully, you would understand that I have chosen wpa_supplicant, which allows to set up a secure mesh. The password is inside my wpa_supplicant.conf file. 

You can read more about open80211s here: https://github.com/o11s/open80211s/wiki/HOWTO

bittorf, 2017-01-12:
@jkalmus - i understand that you tried the manual way, but want to make sure you first try the "builtin" uci-way.

jkalmus, 2017-01-12:
OK, I didn't get your point. 

I tried the following 4 cases of which only the open mesh set up using wpa_supplicant (case 3) worked:

1. Set up an **open mesh** the uci-way: No ping

```
root@lede:~# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='0'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].device='radio0'
wireless.@wifi-iface[0].mode='mesh'
wireless.@wifi-iface[0].mesh_id='test_mesh'
wireless.@wifi-iface[0].encryption='none'
```
```
root@lede:~# iw dev wlan0 station dump
Station 78:a3:51:26:43:7d (on wlan0)
	inactive time:	110 ms
	rx bytes:	23485
	rx packets:	610
	tx bytes:	0
	tx packets:	0
	tx retries:	0
	tx failed:	0
	signal:  	-27 [-27, -34] dBm
	signal avg:	-26 [-26, -33] dBm
	tx bitrate:	6.0 MBit/s
	rx bitrate:	6.0 MBit/s
	mesh llid:	0
	mesh plid:	0
	mesh plink:	LISTEN
	mesh local PS mode:	UNKNOWN
	mesh peer PS mode:	UNKNOWN
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	321 seconds
```

2. Set up a **secure mesh** the uci-way: No ping between nodes

```
root@lede:~# uci show wireless
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='pci0000:00/0000:00:01.0/0000:02:00.0'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='0'
wireless.@wifi-iface[0]=wifi-iface
wireless.@wifi-iface[0].device='radio0'
wireless.@wifi-iface[0].mode='mesh'
wireless.@wifi-iface[0].mesh_id='test_mesh'
wireless.@wifi-iface[0].encryption='psk2/aes'
wireless.@wifi-iface[0].key='12345678'
```
```
root@lede:~# iw dev wlan0 station dump
Station 78:a3:51:26:43:7d (on wlan0)
	inactive time:	460 ms
	rx bytes:	11574
	rx packets:	135
	tx bytes:	974
	tx packets:	7
	tx retries:	0
	tx failed:	0
	signal:  	-27 [-27, -31] dBm
	signal avg:	-27 [-27, -36] dBm
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	6.5 MBit/s MCS 0
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	connected time:	64 seconds
```

3. Set up an **open mesh** using wpa_supplicant: Ping successful

wpa_supplicant.conf
```
# open mesh network
network={
        ssid="test_mesh"
        mode=5
        frequency=5180
        key_mgmt=NONE
}
```
```
root@lede:~# iw dev wlan0 station dump
Station 78:a3:51:26:43:7d (on wlan0)
	inactive time:	380 ms
	rx bytes:	3048
	rx packets:	37
	tx bytes:	310
	tx packets:	3
	tx retries:	0
	tx failed:	0
	signal:  	-27 [-27, -32] dBm
	signal avg:	-28 [-28, -32] dBm
	Toffset:	13313098 us
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	6.5 MBit/s MCS 0
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	13 seconds
```

```
--- 192.168.100.2 ping statistics ---
60 packets transmitted, 60 packets received, 0% packet loss
round-trip min/avg/max = 0.773/1.215/2.769 ms
```

4. Set up a **secure mesh** using wpa_supplicant: No ping
wpa_supplicant.conf
```
# secure mesh network
network={
        ssid="test_mesh" 
        mode=5
        frequency=5180
        key_mgmt=SAE
        psk=”12345678”
}
```
```
root@lede:~# iw dev wlan0 station dump
Station 78:a3:51:26:43:7d (on wlan0)
	inactive time:	730 ms
	rx bytes:	5066
	rx packets:	53
	tx bytes:	910
	tx packets:	6
	tx retries:	1
	tx failed:	0
	signal:  	-31 [-31, -37] dBm
	signal avg:	-32 [-32, -38] dBm
	Toffset:	24576540 us
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	6.5 MBit/s MCS 0
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	connected time:	24 seconds

```

bittorf, 2017-01-12:
thank you, will replicate that and report

bittorf, 2017-01-14:
we have to wait till this is done https://github.com/lede-project/source/commit/fd718c50256536ed082852d735dc7690d7604418 - also an idea: have you tried without "VHT80" and have you tried with "psk2+aes" (instead of a slash)

jkalmus, 2017-01-16:
Have tried without "VHT80" and with "psk2+aes", but doesn't work either. 
Thanks for your effort.

sages, 2017-02-01:
What wifi modules are loaded?
I've had the same issue with a A5-V11 and with a WT3020 device. Both ralink wifi.
I had to make the following change:
in /etc/modules.d/rt2800-soc
change it from:
rt2800soc
to:
rt2800soc nohwcrypt=1

I now have an encrypted mesh link with pings between the A5-V11, Nexx WT3020 and a tplink Wr703N
The TP link uses an ath9k driver and didn't require any other changes apart from configuring the wireless mesh.

jkalmus, 2017-02-01:
The following WiFI module is loaded : mt7603e
Will try with nohwcrypt=1 parameter and report.

I confirm that the issue does not happen in case of at9k and ath10k.

jkalmus, 2017-02-01:
I've checked the following directory:  /sys/module/mt76x2e/  and I don't see any trace of "nohwcrypt" parameter.

I case of the ath9k module the nohwcrypt parameter value can be changed in the following directory: 
/sys/module/ath9k/parameters/nohwcrypt

sages, 2017-02-01:
Do a ls on /etc/modules.d
My WT3020 had the mt drivers but was using rt28xx for the actual wifi

sages, 2017-02-01:
On my Nexx WT3020
root@LEDE-Nexx:~# ls /sys/module
8250                    mt7603e                 sch_tbf
act_mirred              mt76x2e                 slhc
act_skbedit             nf_conntrack            spurious
block                   nf_conntrack_ipv4       tcp_cubic
cfg80211                nf_conntrack_ipv6       workqueue
cls_flow                nf_conntrack_rtcache    x_tables
cls_fw                  nf_defrag_ipv4          xt_CLASSIFY
cls_route               nf_defrag_ipv6          xt_CT
cls_tcindex             nf_log_common           xt_DSCP
cls_u32                 nf_log_ipv4             xt_HL
compat                  nf_log_ipv6             xt_LOG
crc_ccitt               nf_nat                  xt_REDIRECT
crc_itu_t               nf_nat_ipv4             xt_TCPMSS
eeprom_93cx6            nf_nat_masquerade_ipv4  xt_comment
em_u32                  nf_nat_redirect         xt_connbytes
firmware_class          nf_reject_ipv4          xt_connlimit
gpio_button_hotplug     nf_reject_ipv6          xt_connmark
ifb                     ppp_async               xt_conntrack
ip6_tables              ppp_generic             xt_dscp
ip6t_REJECT             pppoe                   xt_ecn
ip6table_filter         pppox                   xt_helper
ip6table_mangle         printk                  xt_hl
ip_tables               rng_core                xt_length
ipt_ECN                 rt2800lib               xt_limit
ipt_MASQUERADE          rt2800mmio              xt_mac
ipt_REJECT              rt2800pci               xt_mark
iptable_filter          rt2800soc               xt_multiport
iptable_mangle          rt2x00lib               xt_nat
iptable_nat             rt2x00mmio              xt_recent
ipv6                    rt2x00pci               xt_state
kernel                  rt2x00soc               xt_statistic
leds_gpio               sch_cake                xt_tcpmss
mac80211                sch_hfsc                xt_tcpudp
module                  sch_htb                 xt_time
mt76                    sch_ingress
root@LEDE-Nexx:~# ls /sys/module/mt76x2e
coresize   drivers    holders    initsize   initstate  refcnt     taint      uevent
root@LEDE-Nexx:~# ls /etc/modules.d
20-eeprom-93cx6         ipt-conntrack-extra     nf-ipt6
30-gpio-button-hotplug  ipt-core                nf-nat
34-ifb                  ipt-ipopt               ppp
42-ip6tables            ipt-nat                 pppoe
50-mt76                 lib-crc-ccitt           rt2800-pci
60-leds-gpio            lib-crc-itu-t           rt2800-soc
70-sched-core           nf-conntrack            rt2x00-pci
75-sched-cake           nf-conntrack6
ipt-conntrack           nf-ipt

The mt76x2e module on mine also doesn't have any nohwcrypt parameter. But the rt2800soc does.

jkalmus, 2017-02-01:
Yep, the nohwcrypt trick doesn't work on the mt76x2e module. 

BTW, your Naxx WT3020 uses the MediaTek RT5390 WiFi chipset. Hence the rt2800soc module for WiFi.
https://wiki.openwrt.org/toh/nexx/wt3020

In my case the MT7612 chipset is supported by the mt76x2e module.

sages, 2017-02-01:
Worth a try anyway. Will keep an eye on the issue. Good luck.

bittorf, 2017-02-13:
can you please test again after the latest changes in LEDE?

jkalmus, 2017-03-16:
1. open mesh using uci-way works
2. secure mesh using uci-way does not work
3. open mesh using wpa_supplicant works
4. secure mesh using wpa_supplicant does not work

bittorf, 2017-03-17:
thanks for testing. can you report this to linux-wireless mailinglist. maybe @nbd168 has an opinion about that too...felix?

nbd168, 2017-03-17:
I just pushed a fix, please test r3770-315afb92eb or newer

jkalmus, 2017-03-17:
Thanks for the update. With r3770-315afb92eb the nodes still cannot ping each other in a secure mesh. However, I noticed that ARP messages appear in tcpdump on the mesh interface, which wasn't the case in my previous tests.

nbd168, 2017-12-03:
Secure mesh is fixed in the current version

closed 2017-12-03 by nbd168

oavaldezi, 2018-10-27:
I'm having the same problem with OpenWrt 18.06.1, r7258-5eb055306f running on several [ZBT WE1226 routers](https://openwrt.org/toh/hwdata/zbt/zbt_we1226) set up with /etc/config/wireless files.
I can ping the nodes when the mesh is unencrypted, and stations can associate to the nodes.
But if I set
```
option encryption 'psk2+ccmp'
option key '<my key here>'
option sae_password '<my key here>
```'
I can't ping the nodes anymore, and stations can't associate to the nodes.

strusty, 2018-10-29:
Is that a defunct authsae option under wpad-mesh?  This may be a clue - https://github.com/libremesh/lime-packages/pull/416

oavaldezi, 2018-10-29:
> Is that a defunct authsae option under wpad-mesh?

@strusty , earlier you referred to [this thread](https://github.com/libremesh/lime-packages/pull/416), from which I infer that Lime ([the LibreMesh project](https://libremesh.org/)) only uses `option ieee80211s_encryption 'psk2/aes'`, and not `option sae_password`. I will give it a try.

Why does Lime set `option ieee80211s_mesh_fwding '0'` instead of `'1'`? I had thought `'1'` was necessary for the mesh to work.

And where are all of these options properly documented?

oavaldezi, 2018-10-30:
Well, I tried with
```
option encryption 'psk2'
option key '<my key here>'
```
and with `psk2+ccmp`, `psk2+aes`, `psk2/aes`,  and the result was always the same: with encryption I can't ping the nodes, and stations can't associate to the nodes.

strusty, 2018-10-30:
Remember, you are now on the bleeding edge here, and most people are not eager to share the details of making 802.11s into an open commercial product.  Even getting 802.11s to mesh at all, for example, on 5Ghz, is not possible with current QCA firmware on IPQ-4019 and required ath10k-ct driver by private company to fix the bug.  But, they have QCA SDK and QCA charges a lot of money for this and has many layers of NDA so only key people like Google can make it work and offer 802.11s mesh product.  Once you can encrypt the mesh and the station limit is lifted, the cat is out of the bag, so to speak.  All mesh traffic running via SSL is end-to-end encrypted between client and target server, so MiM attack is not as easy as in olden days.  You can always use ipsec and vpn in the meantime if it is an emergency; my bet from following your threads is that it plain doesn't work with the public firmware and needs bugs fixed that were left in the radio firmware accidentally on purpose by the manufacturer.  Once we all figure this out we can properly document it ourselves, someone will help us solve this soon, it is the next thing on the public agenda for 802.11s.  Also, remember that there is no current 2018 HOWTO or repository on the current state of the industry, even OpenWISP docs are deprecated as the software around this is changing rapidly.   It seems like this was already working at the authsae stage of things, but now wpad-mesh developers probably need to be asked your question to get some relief.  Please keep us all posted, when you solve this, we all solve this.

markbirss, 2018-11-28:
@strusty @oavaldezi 
Wanted to discuss ways of confirming if in fact the mesh when up is encrypted or not, ways of checking?

It was suggested on openwrt wiki to use Android App like "Wifi Analyzer"

I suggested TCPDump and check if any plaintext is being passed.

I also tried quite a number of different ways. The Android App now shows me both a Open and a encrypted adhoc network which leaves me bit unsure

i used LibreMesh firmware and changed wireless config to include these 

```
        option ieee80211s_encryption 'psk2/aes'
        option encryption 'psk2/aes'
        option sae_password 'password'
```

and also changes /etc/modules.d/rt2800-soc with "nohwcrypt-1" but this does not seem to change whether i add it or not

strusty, 2018-11-28:
https://forum.openwrt.org/t/setting-up-authenticated-mesh-with-wpad-mesh/12399/49  has reported:

Got mine working with with these settings. I had to activate hwnocrypt=1 for my ath9k device (wzr-hp-g300nh), connected with a mt7621 device (xiaomi router 3g) on 2.4Ghz network.
Like mjs said, Web Ui tells me encryption is "none" when the mesh is conected otherwise WPA2.

/etc/config/wireless:-

config wifi-iface 'mesh0'
       option device 'radio0'
       option mode 'mesh'
       option mesh_fwding '1'
       option mesh_id 'My Mesh'
       option encryption 'psk2/aes' (psk2+ccmp works too)
       option key 'MyPassword'          
       option network 'lan'
/etc/config/network:-

config interface 'lan'
        option ifname 'eth0.1'

oavaldezi, 2018-11-28:
@markbirss @strusty I haven't tried `option ieee80211s_encryption 'psk2/aes'`. I'll give it a try this weekend. With `option encryption 'psk2'`, the [Wifi Analyzer app](https://play.google.com/store/apps/details?id=com.farproc.wifi.analyzer) (which is quite helpful) showed the mesh network as unencrypted, and tcpdump showed plaintext messages.

markbirss, 2019-02-11:
@oavaldezi ok, it now appears on 802.11s encryption not working with 18.06.1/2, however im able to run mesh and ap same time (even on mt7620/ralink2800 usb dongles)

strusty, 2019-02-11:
Yes, mesh and AP work together, although mesh seems to steal about 1/2 the
bandwidth even if it is not pushing data.  You will notice on some of the
mt/ralink routers that it works for a little while, then the AP will
randomly drop connection with a silent failure.  It would be nice if mesh
encryption was working, although I thought I saw somewhere that someone had
achieved this.  However, if your traffic is using https and ssh, etc. it
shouldn't pose too much of a risk.

strusty, 2019-02-11:
however, have you tried this ->

wpad-mesh-openssl - 2018-12-02-c2c6c01b-1 - This package contains a minimal
IEEE 802.1x/WPA Authenticator and Supplicant (with 802.11s mesh and SAE
support).

oavaldezi, 2019-02-11:
> however, have you tried this -> wpad-mesh-openssl - 2018-12-02-c2c6c01b-1

I'll give it a try as soon as I can.

oavaldezi, 2019-02-12:
> have you tried this -> wpad-mesh-openssl - 2018-12-02-c2c6c01b-1

_wpad-mesh-openssl - 2018-05-21-62566bc2-5_ was available for my architecture, and it did the trick. The mesh seems to be up and authenticated, according to **iw dev wlan0 station dump**

I also needed to set **option network** to **'lan'** in order to bridge the mesh interfaces into the rest of my lan.

markbirss, 2019-02-12:
i will also have a look soon, thank you

oavaldezi, 2019-02-12:
@strusty, thanks for the tip. I'm curious: how did you learn about wpad-mesh-openssl ?

BIGFAAT, 2019-02-12:
@oavaldezi Can you please post your network and wireless config as an example? Thank you in advance.

oavaldezi, 2019-02-12:
@BIGFAAT, my pleasure.

**/etc/config/network:**
```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'some ula_prefix'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option ipaddr '192.168.0.151'
	option netmask '255.255.255.0'
	option gateway '192.168.0.1'
	list dns '200.12.232.4'
	list dns '200.12.229.1'
	list dns '8.8.8.8'
	option ip6assign '60'

config device 'lan_dev'
	option name 'eth0.1'
	option macaddr 'macaddrX'

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'

config device 'wan_dev'
	option name 'eth0.2'
	option macaddr 'macaddrY'

config interface 'wan6'
	option ifname 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 6t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '4 6t'
```
**/etc/config/wireless:**
```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '6'
	option hwmode '11g'
	option path 'platform/10300000.wmac'
	option htmode 'HT20'
	option disabled '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'ssid_name'
	option encryption 'psk2'
	option key 'ssid_password'

config wifi-iface 'mesh'
	option device 'radio0'
	option network 'lan'
	option mode 'mesh'
	option mesh_id 'mesh_name'
	option mesh_fwding '1'
	option encryption 'psk2/aes'
	option key 'mesh_password'
```
I hope this helps others.

Now that I got it working on very cheap [ZBT WE1226](https://openwrt.org/toh/hwdata/zbt/zbt_we1226) AP's, I'd like to try it on dual radio AP's, having the mesh on one frequency, and the AP's on the other.