openwrt/mt76 #77: wrong RSSI calculation

psyborg55, 2017-01-14:
looks like a driver does some wrong calculation on RSSI value.

when having zeroed out offset 0x45 in eeprom (this was popular with xiaomi mini until http://nbd.name/mt76-lnagain.patch was made) i get RSSI value ~ -65dBm, if i keep original value (+2dBm) at 0x45 the RSSI is calculated in wrong direction showing ~69dBm while it should be closer to -60dBm

nbd168, 2017-03-17:
Please try this patch: http://nbd.name/mt76-rssi-offset.patch

psyborg55, 2017-03-17:
no RSSI value change.

nbd168, 2017-03-17:
I just reviewed the docs and your descrption again. If I understand it correctly, it seems to me that the behavior you're describing matches the intention of the code and there is no bug. If you set the LNA Gain to +2 dBm (0x82), it means that the LNA path adds 2 dBm to the signal, which the driver should then subtract from the raw RSSI value to get the real signal strength.
Am I missing something here?

psyborg55, 2017-03-17:
right, without LNA gain i have -65dBm, +2 means the chip has to increase gain for 2dB. and increased gain manifests by getting RSSI closer to 0, or better described in positive direction. simple math -65+2=-63, not -67 (except that in real life test this has difference of about 4-5dB)

i think that maybe raw RSSI value should already have this gain included since the LNA should amplify it by 2dB.

worst thing here is this strange design; using eLNA gain parameter to tune iLNA chip, however with their default value +2dB the reception performance is the best