openwrt/mt76 #80: tx path config bug

psyborg55, 2017-02-12:
when set chainmask to 0x101 in mt76x2_init rx path is properly limited to one stream while tx path still uses both streams

RX rate was max 150Mbps and never exceeded, TX rate 240-270Mbps

nbd168, 2017-03-17:
Not a bug. The driver does not support antenna selection yet, and for tx the stream configuration is set in the TXWI

closed 2017-03-17 by nbd168

psyborg55, 2017-03-19:
is there a way to move configuration to mt76x2_phy.c ? similar how mt76x2_set_rx_chains works. or is there txwi register for 1 stream ? i guess 0x93 is 2 stream.

i've added some code that based on chainmask limits RX path according to NIC_CONF_2/mirror from NIC_CONF_0, if the same could be used for TX path too it would be great.

psyborg55, 2017-03-20:
tried changin 0x93 to 0x83 and 0x0 which did not make any differnece. maybe this register affect HT20 bug described in https://dev.openwrt.org/ticket/20335#comment:8