openwrt/mt76 #82: mt76x2_mac.c:254

kofec, 2017-02-14:
[19780.088032] ------------[ cut here ]------------
[19780.092840] WARNING: CPU: 0 PID: 0 at /home/kofec/openwrt/LEDEXiaomi/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2017-02-03-a5eaeb27/mt76x2_mac.c:254 mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]()
[19780.111841] Modules linked in: pppoe ppp_async rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_hashlimit wireguard slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt mt7603e mt76x2e mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel zram lzo_decompress lzo_compress lz4_decompress lz4_compress zsmalloc eeprom_93cx6 usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform ehci_hcd sd_mod scsi_mod gpio_button_hotplug f2fs ext4 jbd2 mbcache usbcore nls_base usb_common crc16 crc32c_generic crypto_hash
[19780.204087] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.47 #0
[19780.209955] Stack : 803483f4 00000000 00000001 80390000 8038d09c 8038cd23 80329d44 00000000
[19780.209955] 	  803d3438 8736a000 00000001 00000000 8777d878 8004ad54 8032f24c 80390000
[19780.209955] 	  00000003 8736a000 8032d650 87c0ddf4 8777d878 80048cc0 8777d460 00000000
[19780.209955] 	  00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[19780.209955] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[19780.209955] 	  ...
[19780.246344] Call Trace:
[19780.248851] [<80014840>] show_stack+0x50/0x84
[19780.253318] [<80025344>] warn_slowpath_common+0xa0/0xd0
[19780.258657] [<800253fc>] warn_slowpath_null+0x18/0x24
[19780.263849] [<876f40b4>] mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]
[19780.270277] [<876f047c>] mt76x2_queue_rx_skb+0x84/0xcc [mt76x2e]
[19780.276449] [<8771d09c>] mt76_dma_attach+0xdcc/0xeec [mt76]
[19780.282141] 
[19780.283674] ---[ end trace dac107a912c3ac07 ]---

kofec, 2017-04-02:
today again:
[18338.703277] ------------[ cut here ]------------
[18338.708071] WARNING: CPU: 0 PID: 0 at /home/kofec/openwrt/LEDEXiaomiDVB/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2017-03-17-712b8e8f/mt76x2_mac.c:254 mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]()
[18338.727339] Modules linked in: pppoe ppp_async rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_hashlimit wireguard usbhid slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables hid_generic crc_itu_t crc_ccitt hid evdev input_core mt76x2e mt7603e i2c_dev i2c_core mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables ip6_udp_tunnel udp_tunnel tun vfat fat cifs nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 sha256_generic md5 md4 hmac ecb des_generic usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache crc16 aead crypto_null cryptomgr crc32c_generic crypto_hash
[18338.826811] CPU: 0 PID: 0 Comm: swapper Not tainted 4.4.56 #0
[18338.832685] Stack : 803717f4 00000000 00000001 803c0000 803bd09c 803bcd23 803527cc 00000000
[18338.832685] 	  80403438 871e6800 00000001 00000000 876bd878 8004ad5c 80357ccc 803c0000
[18338.832685] 	  00000003 871e6800 803560d0 87c0ddf4 876bd878 80048cc8 876bd460 00000000
[18338.832685] 	  00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[18338.832685] 	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[18338.832685] 	  ...
[18338.869088] Call Trace:
[18338.871596] [<80014838>] show_stack+0x50/0x84
[18338.876049] [<8002534c>] warn_slowpath_common+0xa0/0xd0
[18338.881404] [<80025404>] warn_slowpath_null+0x18/0x24
[18338.886586] [<871740c4>] mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]
[18338.893021] [<8717047c>] mt76x2_queue_rx_skb+0x84/0xcc [mt76x2e]
[18338.899209] [<8773109c>] mt76_dma_attach+0xdcc/0xeec [mt76]
[18338.904897] 
[18338.906419] ---[ end trace 790bc0c71f3ff8b0 ]---

nbd168, 2017-12-03:
Fixed in the latest version

closed 2017-12-03 by nbd168