openwrt/mt76 #83: doesn't work for japan

psyborg55, 2017-02-14:
set any of japanes channels will not work the output is HT40 channel pair not allowed

nbd168, 2017-02-14:
Please include more detailed steps to reproduce. This description is simply too vague

psyborg55, 2017-02-14:
you are missing these channels in mac80211 file whatever its named i don't know exactly. even then it is not detected unless running HT20. in some cases it will not start at all (dpeneding on channel and mode HT/VHT). the only case where it started on HT40 was in N mode and did not work because my card did not read any beacons, signal/rssi levels are same as on EU channels so no real interest here to dig why it was not detected.

from my memory the message has been: HT40 channel pair (184, 192) not allowed