openwrt/mt76 #84: HT/VHT modes have weak range

psyborg55, 2017-02-23:
connecting to mt7612e wifi from greater distance makes the connection very slow in N/AC modes.
with tx rate HT40 30 Mbps that sometimes change to 54, 60 or 90 Mbps max speed was 2.5/1.5 Mbps.

changing to legacy mode with the notebook in the same place and tx rate 18 Mbps the speed was 8.5/4.5 Mbps

throughput measured with speedtest.net, in N/AC modes it had 6 ms larger ping response time.

nbd168, 2017-10-13:
Current versions should behave much better now, there were some rate control related bugs

closed 2017-10-13 by nbd168

psyborg55, 2017-12-20:
unfortunately, still the same. i have notebook connecting from 5m distance behind thin wall and glass doors, syncs at 30-78Mbps (12-36Mbps in legacy mode) but no data passes through. page loading starts and hangs. in the original test the notebook was placed 2 floors below the router and signal was going through 2 ~50cm walls

psyborg55, 2018-01-03:
this bug may be related to txpower issue: https://github.com/openwrt/mt76/issues/16