openwrt/mt76 #86: Scanning not possible when sta disabled?

fherbert, 2017-03-06:
I'm not sure if this is a bug or a feature.  I have an event driven script which disables the sta wifi-iface if it's not able to connect. This adds `option disabled '1'` to the sta wifi-iface section of the config file so my wireless config looks like:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'ap'
        option device 'radio0'
        option network 'aplan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'
        option ifname 'wlan0'

config wifi-iface 'sta'
        option device 'radio0'
        option network 'wlan'
        option mode 'sta'
        option ifname 'apcli0'
        option ssid 'AP'
        option encryption 'psk2'
        option key 'appassword'
        option disabled '1'
```

After wifi down/up is run, I'm unable to use just iwinfo to scan for access points:
```
iwinfo wlan0 scan
Scanning not possible
```

I can work around this by first running `iw dev wlan0 scan`, once this has run, running `iwinfo wlan0 scan` displays the available access points.

If the sta wifi-iface is not disabled (ie option disabled '0') then iwinfo wlan0 scan returns the available access points without having the run iw dev wlan0 scan first.

I'm using a MT7688 (linkit7688) running [LEDE v17.01](https://github.com/lede-project/source/commit/09a8183ce849984b725d1c8be21ea93ce93e7f09) and have tried mt76 commits at 3c8caafc5e150db79f714b958a51cee8f242f309 and a5eaeb27d6a80c012ffc8b5bb5549983e446b4ef

So is this a bug or a feature?

psyborg55, 2017-03-08:
the default behavior was that if the wifi interface is disabled completely, hitting scan button in luci would return available APs. i would assume luci used iw dev command for scanning which creates tmp wifi interface that is not destroyed after luci scan finish and could probably be reused for iwinfo scan.

fherbert, 2017-03-08:
It looks like I may have been using the incorrect device name to use for the scan, when running`iwinfo radio0 scan` it scan's successfully (while still having the sta wifi-iface disabled).
From reading the [docs,](https://wiki.openwrt.org/doc/howto/wireless.utilities) Luci uses iwinfo.

nbd168, 2017-03-15:
This does not seem like a mt76 bug to me, so this is the wrong place for this report. It's more likely an iwinfo issue.

closed 2017-03-15 by nbd168