openwrt/mt76 #88: Log getting spammed with "daemon.notice hostapd: ACS: Survey is missing noise floor"

torstefi, 2017-03-16:
So, every time I reboot my router (which I have to do periodically because of a unrelated Issue I haven't been able to diagnose), the log gets spammed with lines like this:

`Thu Mar 16 04:01:25 2017 daemon.notice hostapd: ACS: Survey is missing noise floor`

267 lines last reboot. O_o

Also, there's no noise floor measurement under 'associated stations'. I'm wondering if pull request #85 would fix this?

Soberia, 2020-01-05:
Same here (19.07.0-rc2)

4k3or3et, 2020-01-11:
I have the same issue on stable 19.07.0.

brentonv, 2020-03-11:
Same on 19.07.1 and 19.07.2

mxdh, 2020-03-12:
Same on 19.07.1 and 19.07.2

brentonv, 2020-03-12:
My "ACS: Survey is missing noise floor" error was resolved by manually setting radio channel as mentioned in this post: https://forum.openwrt.org/t/rt-n56-no-reactivation-of-5ghz-except-reboot/12244/5

tmedicci, 2020-06-12:
Same here in 19.07.3 (stable). Device: MT7688.

I can confirm that manually setting the channel fixes the error. However, I need to set the option to auto because there a plenty of such devices near each other it would be desirable that they could select the best channel to AP interface.

kroeckx, 2021-02-21:
Not that ACS stands for "Automatic Channel Selection", so it's normal that if you set a fixed channel that you don't get the messages since ACS is not running.

Note that iwinfo is also reporting: "Noise: unknown", in luci at cgi-bin/luci/admin/network/wireless you will see " --- dBm" for the radio, and if a device is associated, it just shows the signal "-52 dBm", and not the combination of signal and noise "-52 / -94 dBm".

So my question really is why is there no noise reported.

uniartisan, 2021-03-05:
I have met the same probelm.
MI route 3G/3Pro mt7621

57382, 2021-06-01:
I have the same problem.

suryadana, 2021-07-02:
> My "ACS: Survey is missing noise floor" error was resolved by manually setting radio channel as mentioned in this post: https://forum.openwrt.org/t/rt-n56-no-reactivation-of-5ghz-except-reboot/12244/5

I am solved with this just set the channel to 11 and increase the txpower to 25.

KerryGee, 2021-10-30:
Same issue here on:

Model	WAVLINK WL-WN577A2
Architecture	MediaTek MT7628AN ver:1 eco:2
Firmware Version	OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.301.66596-f28aaa3
Kernel Version	5.4.143
cpu model               : MIPS 24KEc V5.5

The syslog is flooded with "daemon.notice hostapd: ACS: Survey is missing noise floor" and the 5Ghz radio looks like this:

Type: MediaTek MT7610E 802.11nac
Channel: 0 (0.000 GHz)
Bitrate: -

The associated client to a 5Ghz AC base wifi station looks like this:

[Signal: 0 dBm / Quality: 0%] SSID: WIFI-BASE-5GHZ
Mode: Client
BSSID: XX:XX:XX:XX:XX:XX (actually it shows a real bssid but obfuscated for the post with X)
Encryption: -
Associations: -

Additionally - kinda off topic to this bug report (maybe something different) - the LED Configuration Button keeps on spinning after pressing "Add LED Action" forever with no dialog shown. I am not sure whether this realtes to the issue somehow, but wanted to mention it here.

EvanCarroll, 2023-03-03:
I"m also getting this error.

MilesBHuff, 2023-11-15:
Same issue on OpenWrt 22.03.5 r20134-5f15225c1e.
```
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
[... cut ...]
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: ACS: Survey is missing noise floor
Wed Nov 15 04:32:01 2023 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=2447 channel=8
```

tojestzart, 2024-06-18:
after updating packages i have same issue now

ob45, 2025-02-21:
I got the same issue on my ` Xiaomi Mi Router 4A (100M International Edition V2)` .
Openwrt version: `OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.014.55016~7046a1c`

```
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
[... cut ...]
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2457 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2467 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2462 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Fri Feb 21 03:33:18 2025 daemon.notice hostapd: phy0-ap0: ACS-COMPLETED freq=2452 channel=9
```

opty77, 2025-03-21:
Count me in with Strong 1200 since 19.07.7.

opty77, 2026-02-14:
More people, more attention?