openwrt/mt76 #90: iwinfo radio0 scan always returns 'Scanning not possible'

fherbert, 2017-03-21:
I'm running the latest commit for mt76 (https://github.com/openwrt/mt76/commit/712b8e8f69c4a0095fb22211116f9015c0309a06)  and LEDE 17.01-SNAPSHOT, r3277-1392dc8. The hardware is the MT7688 chip (on the linkit7688 board).
My wireless config is:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'platform/10300000.wmac'
	option htmode 'HT40'
	option disabled '0'
```

When I run `wifi up` I notice the following the log: `kern.warn kernel: [31462.057985] MCU message 40 (seq 6) timed out` not sure what this is about.

When I run `iwinfo radio0 scan` all that is returned is 'Scanning not possible'
I've added some debug code to iwinfo so I can see it calling the nl80211_msg function and sending  NL80211_CMD_TRIGGER_SCAN, then I see the following in the log:
```
Mon Mar 20 11:50:48 2017 kern.warn kernel: [31513.147993] MCU message 40 (seq 7) timed out
Mon Mar 20 11:50:49 2017 kern.warn kernel: [31514.147985] MCU message 8 (seq 8) timed out
Mon Mar 20 11:50:49 2017 daemon.info odhcpd[766]: Initial RA router lifetime 0, 0 address(es) available on eth0
Mon Mar 20 11:50:50 2017 kern.warn kernel: [31515.147985] MCU message 40 (seq 9) timed out
Mon Mar 20 11:50:50 2017 kern.info kernel: [31515.156591] IPv6: ADDRCONF(NETDEV_UP): tmp.radio0: link is not ready
Mon Mar 20 11:50:52 2017 kern.warn kernel: [31517.227985] MCU message 8 (seq 10) timed out
Mon Mar 20 11:50:53 2017 kern.warn kernel: [31518.227984] MCU message 40 (seq 11) timed out
Mon Mar 20 11:50:54 2017 kern.warn kernel: [31519.227985] MCU message 8 (seq 12) timed out
Mon Mar 20 11:50:55 2017 kern.warn kernel: [31520.228004] MCU message 40 (seq 13) timed out
Mon Mar 20 11:50:56 2017 kern.warn kernel: [31521.227984] MCU message 8 (seq 14) timed out
Mon Mar 20 11:50:57 2017 kern.warn kernel: [31522.227985] MCU message 40 (seq 15) timed out
Mon Mar 20 11:50:58 2017 kern.warn kernel: [31523.227983] MCU message 8 (seq 1) timed out
Mon Mar 20 11:50:59 2017 kern.warn kernel: [31524.227985] MCU message 40 (seq 2) timed out
Mon Mar 20 11:51:00 2017 kern.warn kernel: [31525.227984] MCU message 8 (seq 3) timed out
Mon Mar 20 11:51:01 2017 kern.warn kernel: [31526.227974] MCU message 40 (seq 4) timed out
Mon Mar 20 11:51:02 2017 kern.warn kernel: [31527.227983] MCU message 8 (seq 5) timed out
Mon Mar 20 11:51:03 2017 kern.warn kernel: [31528.227987] MCU message 40 (seq 6) timed out
Mon Mar 20 11:51:04 2017 kern.warn kernel: [31529.227983] MCU message 8 (seq 7) timed out
Mon Mar 20 11:51:05 2017 kern.warn kernel: [31530.227985] MCU message 40 (seq 8) timed out
Mon Mar 20 11:51:06 2017 kern.warn kernel: [31531.227983] MCU message 8 (seq 9) timed out
Mon Mar 20 11:51:07 2017 kern.warn kernel: [31532.227985] MCU message 40 (seq 10) timed out
Mon Mar 20 11:51:08 2017 kern.warn kernel: [31533.227984] MCU message 8 (seq 11) timed out
Mon Mar 20 11:51:09 2017 kern.warn kernel: [31534.227984] MCU message 40 (seq 12) timed out
Mon Mar 20 11:51:10 2017 kern.warn kernel: [31535.227985] MCU message 8 (seq 13) timed out
Mon Mar 20 11:51:11 2017 kern.warn kernel: [31536.227985] MCU message 40 (seq 14) timed out
Mon Mar 20 11:51:12 2017 kern.warn kernel: [31537.227992] MCU message 8 (seq 15) timed out
Mon Mar 20 11:51:13 2017 kern.warn kernel: [31538.227984] MCU message 40 (seq 1) timed out
Mon Mar 20 11:51:14 2017 kern.warn kernel: [31539.227983] MCU message 8 (seq 2) timed out
Mon Mar 20 11:51:15 2017 kern.warn kernel: [31540.227985] MCU message 40 (seq 3) timed out
Mon Mar 20 11:51:16 2017 kern.warn kernel: [31541.227984] MCU message 8 (seq 4) timed out
Mon Mar 20 11:51:17 2017 kern.warn kernel: [31542.227985] MCU message 40 (seq 5) timed out
Mon Mar 20 11:51:18 2017 kern.warn kernel: [31543.227984] MCU message 8 (seq 6) timed out
Mon Mar 20 11:51:19 2017 kern.warn kernel: [31544.227984] MCU message 40 (seq 7) timed out
Mon Mar 20 11:51:20 2017 kern.warn kernel: [31545.227984] MCU message 8 (seq 8) timed out
Mon Mar 20 11:51:21 2017 kern.warn kernel: [31546.227985] MCU message 40 (seq 9) timed out
```

After these messages finish in the log, iwinfo reports 'Scanning not possible'.

I'm happy to help debug this if I can be of any assistance.

fherbert, 2017-03-25:
I've updated to LEDE Reboot SNAPSHOT r3871-8a27b72 but still having lots of issues. Should I be using this driver for MT7688?

With the latest snapshot I get the following dmesg output and dump (intermittently) when trying to scan.
```
# iwinfo radio0 scan
[56973.047606] MCU message 8 (seq 4) timed out
[56974.047603] MCU message 40 (seq 5) timed out
[56975.047605] MCU message 8 (seq 6) timed out
[56976.047607] MCU message 40 (seq 7) timed out
[56977.047597] MCU message 8 (seq 8) timed out
[56978.047603] MCU message 40 (seq 9) timed out
[56979.047608] MCU message 8 (seq 10) timed out
[56980.047608] MCU message 40 (seq 11) timed out
[56981.047607] MCU message 8 (seq 12) timed out
[56982.047607] MCU message 40 (seq 13) timed out
[56983.047606] MCU message 8 (seq 14) timed out
[56984.047605] MCU message 40 (seq 15) timed out
[56985.047601] MCU message 8 (seq 1) timed out
[56986.047605] MCU message 40 (seq 2) timed out
[56987.047600] MCU message 8 (seq 3) timed out
[56988.047604] MCU message 40 (seq 4) timed out
[56989.047606] MCU message 8 (seq 5) timed out
[56990.047607] MCU message 40 (seq 6) timed out
[56991.047615] MCU message 8 (seq 7) timed out
[56992.047606] MCU message 40 (seq 8) timed out
[56993.047606] MCU message 8 (seq 9) timed out
[56994.047603] MCU message 40 (seq 10) timed out
[56995.047602] MCU message 8 (seq 11) timed out
[56996.047603] MCU message 40 (seq 12) timed out
[56997.047602] MCU message 8 (seq 13) timed out
[56998.047598] MCU message 40 (seq 14) timed out
[56999.047606] MCU message 8 (seq 15) timed out
[57000.047604] MCU message 40 (seq 1) timed out
Scanning not possible
[57002.056121] ------------[ cut here ]------------
[57002.065331] WARNING: CPU: 0 PID: 2662 at /home/fherbert/git/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7688/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1219 mt7603_mac_work+0xf0/0x284 [mt7603e]()
[57002.100626] Modules linked in: nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables i2c_mt7621 mt7603e i2c_dev mt76 mac80211 cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables gpio_mcp23s08 i2c_core vfat fat autofs4 nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_1 nls_cp437 leds_gpio gpio_button_hotplug exfat nls_base
[57002.226989] CPU: 0 PID: 2662 Comm: kworker/u2:0 Not tainted 4.4.53 #0
[57002.239771] Workqueue: phy0 mt7603_mac_work [mt7603e]
[57002.249776] Stack : 87c03410 87765218 00000088 8004ae14 87da16a4 80396d63 80332e04 00000a66
          8035baf0 87c49d64 80390000 80048d80 00000088 8004ae14 8033830c 80390000
          00000003 87c49d64 80390000 80038d90 00000088 87c49d9c 0000012c 00000000
          00000001 00000000 875751dc 87f86200 87f86300 30796870 00000000 00000000
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[57002.320247] Call Trace:
[57002.325086] [<8001486c>] show_stack+0x50/0x84
[57002.333730] [<80025404>] warn_slowpath_common+0xa0/0xd0
[57002.344088] [<800254bc>] warn_slowpath_null+0x18/0x24
[57002.354107] [<875752cc>] mt7603_mac_work+0xf0/0x284 [mt7603e]
[57002.365515] [<80036ca8>] process_one_work+0x1f8/0x334
[57002.375534] [<80037abc>] worker_thread+0x2b0/0x404
[57002.385033] [<8003be74>] kthread+0xd8/0xec
[57002.393156] [<80004478>] ret_from_kernel_thread+0x14/0x1c
[57002.403847] 
[57002.406786] ---[ end trace 62d29b1c9154b382 ]---
```

Sometimes the scan returns successfully, sometimes is returns no results, sometimes it returns this with 'Scanning not possible'.

When I try to connect to an access point, it seems to intermittently connect, then fail:
```
Sat Mar 25 19:27:19 2017 kern.warn kernel: [57234.847603] MCU message 40 (seq 4) timed out
Sat Mar 25 19:27:20 2017 kern.info kernel: [57236.152866] IPv6: ADDRCONF(NETDEV_UP): apcli0: link is not ready
Sat Mar 25 19:27:20 2017 daemon.notice netifd: radio0 (3268): Successfully initialized wpa_supplicant
Sat Mar 25 19:27:21 2017 daemon.notice netifd: Interface 'wlan' is enabled
Sat Mar 25 19:27:46 2017 kern.info kernel: [57262.138678] apcli0: authenticate with b8:c7:5d:08:c6:e1
Sat Mar 25 19:27:46 2017 kern.info kernel: [57262.155811] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 1/3)
Sat Mar 25 19:27:46 2017 kern.info kernel: [57262.233447] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 2/3)
Sat Mar 25 19:27:46 2017 kern.info kernel: [57262.254893] apcli0: authenticated
[... 7 lines cut ...]
Sat Mar 25 19:27:49 2017 daemon.notice netifd: wlan (3356): udhcpc: received SIGTERM
Sat Mar 25 19:28:00 2017 kern.info kernel: [57275.708637] apcli0: authenticate with b8:c7:5d:08:c6:e1
Sat Mar 25 19:28:00 2017 kern.info kernel: [57275.727561] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 1/3)
Sat Mar 25 19:28:01 2017 kern.info kernel: [57277.067609] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 2/3)
Sat Mar 25 19:28:02 2017 kern.info kernel: [57278.067617] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 3/3)
Sat Mar 25 19:28:03 2017 kern.info kernel: [57279.067618] apcli0: authentication with b8:c7:5d:08:c6:e1 timed out
Sat Mar 25 19:28:11 2017 kern.info kernel: [57286.678594] apcli0: authenticate with b8:c7:5d:08:c6:e1
[... cut ...]
Sat Mar 25 19:29:06 2017 kern.info kernel: [57341.277608] apcli0: authentication with b8:c7:5d:08:c6:e1 timed out
```

So it seems to stay connected for about 2 seconds then disconnects. The access point I'm connecting to is in the same room, about 4 meters away. Other devices next to this unit have no issue connecting and scanning (finding this access point) (imac, iphone and esp8266).
When using the mediatek kernel module (I found the widora one for 4.4.50 which has now been removed), the same hardware scans and connects without any issue.

fherbert, 2017-03-25:
I've just run 6 scans, out of the 6 only one even found my access point. The first three found only the Linkit7688 that's sitting next to this unit running in AP mode, the next 2 found nothing, and the last one found the AP I'm trying to connect to:
```
root@itronhab01:/etc/config# iwinfo radio0 scan
Cell 01 - Address: 9E:65:F9:1E:46:84
          ESSID: "ITRON-4684"
          Mode: Master  Channel: 1
          Signal: -43 dBm  Quality: 67/70
          Encryption: none

root@itronhab01:/etc/config# iwinfo radio0 scan
Cell 01 - Address: 9E:65:F9:1E:46:84
          ESSID: "ITRON-4684"
          Mode: Master  Channel: 1
          Signal: -44 dBm  Quality: 66/70
          Encryption: none

root@itronhab01:/etc/config# iwinfo radio0 scan
Cell 01 - Address: 9E:65:F9:1E:46:84
          ESSID: "ITRON-4684"
          Mode: Master  Channel: 1
          Signal: -43 dBm  Quality: 67/70
          Encryption: none

root@itronhab01:/etc/config# iwinfo radio0 scan
No scan results

root@itronhab01:/etc/config# iwinfo radio0 scan
No scan results

root@itronhab01:/etc/config# iwinfo radio0 scan
[57762.268748] apcli0: authenticate with b8:c7:5d:08:c6:e1
[57762.288943] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 1/3)
[57762.353965] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 2/3)
[57762.486720] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 3/3)
[57762.610939] apcli0: authentication with b8:c7:5d:08:c6:e1 timed out
Cell 01 - Address: 9E:65:F9:1E:46:84
          ESSID: "ITRON-4684"
          Mode: Master  Channel: 1
          Signal: -44 dBm  Quality: 66/70
          Encryption: none

Cell 02 - Address: B8:C7:5D:08:C6:E1
          ESSID: "AP"
          Mode: Master  Channel: 1
          Signal: -71 dBm  Quality: 39/70
          Encryption: WPA2 PSK (CCMP)
```

And then shortly after it decides it can connect to the AP:
```
[57855.688619] apcli0: authenticate with b8:c7:5d:08:c6:e1
[57855.707676] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 1/3)
[57855.777096] apcli0: send auth to b8:c7:5d:08:c6:e1 (try 2/3)
[57855.793276] apcli0: authenticated
[57855.807691] apcli0: associate with b8:c7:5d:08:c6:e1 (try 1/3)
[57856.207601] apcli0: associate with b8:c7:5d:08:c6:e1 (try 2/3)
[57856.225387] apcli0: RX AssocResp from b8:c7:5d:08:c6:e1 (capab=0x1031 status=0 aid=3)
[57856.241149] apcli0: associated
```

phatina, 2017-09-25:
Same thing here (Onion Omega2) - any news about this?

kallisti5, 2017-12-27:
Same thing here on my Onion Omega 2+ running libremesh.

iancd, 2018-02-13:
Hi @fherbert Have you tried the latest trunk snapshot build? The name of this issue seems like it should refer to the **MT7688** rather than the mt7603? Please could you update?

fherbert, 2018-05-20:
Hi @iancd, I'm referring to the kernel module that gets loaded for the mt7688 which the mt7603.

fherbert, 2018-05-20:
I've tested with the latest commit (eaee49eb5e8f4bb761de4e1da217b06689b2eb11) and the mt76 driver no longer reports the MCU message timeout error so am closing this issue.

closed 2018-05-20 by fherbert