openwrt/mt76 #92: MT7612E mPCIe module fails in my Dell Laptop

dangowrt, 2017-03-24:
I compiled mt76 for Kernel 4.10.5 but somehow it doesn't work well...
First of all, when loading the module, my system freezes (ie. mouse cursor not moving!) for about 15 seconds! Once loaded the module seems to be discovered, however, both wpa_supplicant as well as `ip link set wlp2s0f0 up` fails with `RTNETLINK answers: Connection timed out` and triggers that MCU error messages in the kernel log (see below).
```
[  988.994385] mt76x2e 0000:02:00.0: enabling device (0000 -> 0002)
[  989.044207] mt76x2e 0000:02:00.0: PCI->APIC IRQ transform: INT A -> IRQ 17
[  989.094375] mt76x2e 0000:02:00.0: ASIC revision: ffffffff
[  992.632032] mt76x2e 0000:02:00.0: Invalid MAC address, using random address 96:02:e8:bb:e1:1c
[  993.078242] mt76x2e 0000:02:00.0: ROM patch already applied
[  993.080736] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  993.080739] mt76x2e 0000:02:00.0: Build: 1
[  993.080742] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[  993.084015] mt76x2e 0000:02:00.0: Firmware running!
[  994.095718] mt76x2e 0000:02:00.0: MCU message 1 (seq 1) timed out
[  994.105601] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[  994.110266] mt76x2e 0000:02:00.0 wlp2s0f0: renamed from wlan0
```
unloading and loading the module results in

```
[ 1053.881343] mt76x2e 0000:02:00.0: MCU message 20 (seq 2) timed out
[ 1093.320641] mt76x2e 0000:02:00.0: MCU message 20 (seq 3) timed out
[ 1095.827263] mt76x2e 0000:02:00.0: MCU message 20 (seq 4) timed out
[ 1127.586682] mt76x2e 0000:02:00.0: MCU message 20 (seq 5) timed out
[ 1137.426500] mt76x2e 0000:02:00.0: Refused to change power state, currently in D3
[ 1137.426519] mt76x2e 0000:02:00.0: PCI->APIC IRQ transform: INT ? -> IRQ 17
[ 1137.426658] mt76x2e 0000:02:00.0: ASIC revision: ffffffff
[ 1137.428418] mt76x2e: probe of 0000:02:00.0 failed with error -2
```

nbd168, 2017-03-24:
Seems like the PCI MMIO access isn't working and it can't get any sane values from the hardware registers. My guess is that either PCI is broken on that platform, or there's a hardware compatibility issue.

airend, 2017-03-24:
Sorry for hijacking this issue, but I wonder if [my 7620 issue](https://github.com/airend/linux-embedded/commit/74042bb0d22b666876f442c8cc5723a7c497e283#commitcomment-21109138) is related somehow. The PCIe drivers are different I think, but in my case I had to revert to the K4.4 version.

dangowrt, 2017-03-25:
@nbd168 might be right, the device looks weird while probing the bus at boot...
```
[    1.224034] pci 0000:02:00.0: [14c3:7662] type 7f class 0x028000
[    1.241534] pci 0000:02:00.0: unknown header type 7f, ignoring device
```

LynxChaus, 2019-10-31:
@dangowrt, see the message "mt76x2e 0000:02:00.0: Invalid MAC address, using random address 96:02:e8:bb:e1:1c" ??
This mean - driver can't load EEPROM data. And start firmware. After that - device unreliable.