openwrt/mt76 #93: mt76 causing kernel error?

guoxiaoqiao, 2017-03-28:
on LEDE 17.01.0,  often appear the following error.

[16309.014835] ------------[ cut here ]------------
[16309.024138] WARNING: CPU: 0 PID: 2094 at net/core/skbuff.c:4194 skb_try_coalesce+0x214/0x350()
[16309.041314] Modules linked in: pppoe ppp_async iptable_nat rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_LOG slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nfnetlink_queue nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt mt76x2e mt76 mac80211 cfg80211 compat xt_set x_tables ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink eeprom_93cx6 leds_gpio gpio_button_hotplug
[16309.213013] CPU: 0 PID: 2094 Comm: haproxy Not tainted 4.4.52 #0
[16309.224985] Stack : 80308e74 00000000 00000001 80350000 82bf20b4 80348063 802ea7cc 0000082e
[16309.224985]    80393438 80340000 8035bc00 803bfc58 02080020 8004a7a4 802efcd4 80340000
[16309.224985]    00000003 80340000 802ee0d8 8380d914 02080020 80048710 02080020 00000000
[16309.224985]    00000001 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[16309.224985]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[16309.224985]    ...
[16309.295898] Call Trace:
[16309.300791] [<8004a7a4>] vprintk_default+0x24/0x30
[16309.310363] [<80048710>] printk+0x2c/0x38
[16309.318391] [<80024d94>] warn_slowpath_common+0xa0/0xd0
[16309.328819] [<80148fb8>] dump_stack+0x14/0x28
[16309.337528] [<80014280>] show_stack+0x50/0x84
[16309.346209] [<80024d94>] warn_slowpath_common+0xa0/0xd0
[16309.356629] [<801ea4c0>] skb_try_coalesce+0x214/0x350
[16309.366726] [<80024e4c>] warn_slowpath_null+0x18/0x24
[16309.376807] [<801ea4c0>] skb_try_coalesce+0x214/0x350
[16309.386898] [<8023e2c0>] tcp_try_coalesce+0x4c/0xd0
[16309.396667] [<8321068c>] nf_conntrack_tuple_taken+0x384/0x558 [nf_conntrack]
[16309.410727] [<8023e8a8>] tcp_queue_rcv+0x60/0x148
[16309.420119] [<802430b8>] tcp_rcv_established+0x504/0x63c
[16309.430719] [<83269308>] 0x83269308
[16309.437700] [<8024ac84>] tcp_v4_do_rcv+0x90/0x1bc
[16309.447087] [<8024d17c>] tcp_v4_rcv+0x6b4/0x9f4
[16309.456143] [<802251a4>] nf_iterate+0x7c/0xc8
[16309.464849] [<80233f0c>] __inet_lookup_established+0x58/0x198
[16309.476325] [<8022b0e8>] ip_local_deliver_finish+0xd4/0x134
[16309.487449] [<8022b5cc>] ip_local_deliver+0xd0/0xe8
[16309.497180] [<8022b014>] ip_local_deliver_finish+0x0/0x134
[16309.508131] [<8022b898>] ip_rcv+0x2b4/0x320
[16309.516477] [<8022b148>] ip_rcv_finish+0x0/0x268
[16309.525700] [<801f9004>] __netif_receive_skb_core+0x61c/0x6dc
[16309.537179] [<8001123c>] do_IRQ+0x1c/0x2c
[16309.545221] [<802870b4>] br_pass_frame_up+0x84/0xf8
[16309.554951] [<801eb1c4>] __kmalloc_reserve.isra.3+0x44/0xa8
[16309.566099] [<80287590>] br_handle_frame_finish+0x468/0x4ec
[16309.577228] [<801ecf50>] __pskb_pull_tail+0x70/0x310
[16309.587159] [<80287990>] br_handle_frame+0x37c/0x414
[16309.597075] [<80251bf4>] tcp4_gro_receive+0x118/0x1dc
[16309.607174] [<80260ba0>] inet_gro_receive+0x2d4/0x2fc
[16309.617254] [<80287614>] br_handle_frame+0x0/0x414
[16309.626811] [<801f8d80>] __netif_receive_skb_core+0x398/0x6dc
[16309.638273] [<801fa2cc>] dev_gro_receive+0x2b4/0x3dc
[16309.648203] [<801fa61c>] napi_gro_receive+0x64/0xf0
[16309.658079] [<831a06e4>] ieee80211_attach_ack_skb+0xfcc/0x194c [mac80211]
[16309.671727] [<831a3560>] ieee80211_sta_uapsd_trigger+0x1e54/0x2d7c [mac80211]
[16309.685981] [<801e636c>] sock_def_readable+0x34/0x6c
[16309.695908] [<801e4b88>] sock_queue_rcv_skb+0x19c/0x1f0
[16309.706350] [<83050000>] backport_dependency_symbol+0x0/0x10 [compat]
[16309.719229] [<83281602>] ppp_unregister_compressor+0x1102/0x24cc [ppp_generic]
[16309.733746] [<831a5760>] ieee80211_mark_rx_ba_filtered_frames+0x12d8/0x13cc [mac80211]
[16309.749547] [<83282c1c>] ppp_input+0x180/0x394 [ppp_generic]
[16309.760860] [<832770a0>] 0x832770a0
[16309.767909] [<831a5f64>] ieee80211_rx_napi+0x710/0x8b0 [mac80211]
[16309.780090] [<801f9004>] __netif_receive_skb_core+0x61c/0x6dc
[16309.791556] [<801fa2cc>] dev_gro_receive+0x2b4/0x3dc
[16309.801511] [<830fd98c>] mt76_rx_complete+0x80/0x108 [mt76]
[16309.812675] [<830fd0e8>] mt76_dma_attach+0xe18/0xeec [mt76]
[16309.823821] [<801f9e5c>] net_rx_action+0x10c/0x2c8
[16309.833415] [<80027314>] __do_softirq+0x250/0x298
[16309.842815] [<8004ef38>] handle_percpu_irq+0x50/0x80
[16309.852718] [<8004b3fc>] generic_handle_irq+0x24/0x3c
[16309.862810] [<800047f4>] handle_int+0x134/0x140
[16309.871838] 
[16309.874798] ---[ end trace 85723521c6150466 ]---

Whether the content of the following URL is worth reference?
https://github.com/raspberrypi/linux/issues/837

guoxiaoqiao, 2017-04-19:
another error in kernel

[157990.208336] ------------[ cut here ]------------
[157990.217810] WARNING: CPU: 0 PID: 0 at /data2/lede2/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7620/mt76-2017-03-17-712b8e8f/mt76x2_mac.c:254 mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]()
[157990.253157] Modules linked in: pppoe ppp_async iptable_nat rt2800soc rt2800pci rt2800mmio rt2800lib pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_LOG slhc rt2x00soc rt2x00pci rt2x00mmio rt2x00lib nfnetlink_queue nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_log_common nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_itu_t crc_ccitt mt76x2e mt76 mac80211 cfg80211 compat xt_set x_tables ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink eeprom_93cx6 leds_gpio gpio_button_hotplug
[157990.425012] CPU: 0 PID: 0 Comm: swapper Tainted: G        W       4.4.52 #0
[157990.439061] Stack : 80308e74 00000000 00000001 80350000 803483dc 80348063 802ea7cc 00000000
[157990.439061]           80393438 00004800 00000001 8196ff00 83055878 8004a7a4 802efcd4 80340000
[157990.439061]           00000003 00004800 802ee0d8 8380ddf4 83055878 80048710 c028c4f0 00000000
[157990.439061]           80347250 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[157990.439061]           00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[157990.439061]           ...
[157990.510148] Call Trace:
[157990.515225] [<80014280>] show_stack+0x50/0x84
[157990.524117] [<80024d94>] warn_slowpath_common+0xa0/0xd0
[157990.534717] [<80024e4c>] warn_slowpath_null+0x18/0x24
[157990.544990] [<830b40c4>] mt76x2_mac_process_rx+0xb0/0x2d8 [mt76x2e]
[157990.557682] [<830b047c>] mt76x2_queue_rx_skb+0x84/0xcc [mt76x2e]
[157990.569854] [<830fcfd8>] mt76_dma_attach+0xd08/0xeec [mt76]
[157990.581138] 
[157990.584284] ---[ end trace 85723521c6150467 ]---

nbd168, 2018-10-25:
Fixed in current versions

closed 2018-10-25 by nbd168