openwrt/mt76 #98: kernel errors - mt7603_mac.c:329 mt7603_mac_fill_rx+0x290/0x3ac [mt7603e]()

camelcamro, 2017-04-21:
still the mt76 errors
(tested with: ZBT-3526 - mt7621) - current TRUNK
Powered by LuCI Master (git-17.107.62951-91c0b91) / LEDE Reboot SNAPSHOT r3978-e209988

[  121.210000] ------------[ cut here ]------------
[  121.210000] WARNING: CPU: 0 PID: 0 at /var/lib/buildbot/slaves/tictex-02/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:329 mt7603_mac_fill_rx+0x290/0x3ac [mt7603e]()
[  121.230000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt76x2e mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables nfsv3 nfs ip_gre gre ifb ip_tunnel xfs libcrc32c vfat fat lockd sunrpc grace ext4 jbd2 mbcache nls_utf8 nls_iso8859_1 nls_cp437 exportfs sha1_generic ecb crc32c_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  121.400000] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.4.61 #0
[  121.400000] Stack : 00000000 00000000 804e6862 00000041 00000000 00000000 80490000 80500000
[  121.400000]    8048a164 80489d83 80404954 00000000 00000000 804e367c 00000000 8ecf0000
[  121.400000]    00000000 80064fa8 80490000 80500000 8048e288 8048e28c 80409268 8fc0dd94
[  121.400000]    00000003 80062cf4 00000000 8ecf0000 8148ed7c 00000000 00000000 00c0dd94
[  121.400000]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  121.400000]    ...
[  121.400000] Call Trace:
[  121.400000] [<8001692c>] show_stack+0x50/0x84
[  121.400000] [<801b736c>] dump_stack+0x84/0xbc
[  121.400000] [<8002cf40>] warn_slowpath_common+0xa0/0xd0
[  121.400000] [<8002cff8>] warn_slowpath_null+0x18/0x24
[  121.400000] [<8ecf3e8c>] mt7603_mac_fill_rx+0x290/0x3ac [mt7603e]
[  121.400000] [<8ecf2c28>] mt7603_queue_rx_skb+0xb4/0xfc [mt7603e]
[  121.400000] [<8ece8fd0>] mt76_dma_attach+0xd00/0xee4 [mt76]
[  121.400000]
[  121.480000] ---[ end trace 4c1ba2fc6726afd5 ]---
[  227.930000] ------------[ cut here ]------------
[  227.930000] WARNING: CPU: 0 PID: 26 at /var/lib/buildbot/slaves/tictex-02/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2017-03-17-712b8e8f/mt7603_mac.c:1228 mt7603_mac_work+0x1bc/0x284 [mt7603e]()
[  227.950000] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress mt76x2e mt7603e ledtrig_usbport mt76 mac80211 cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables nfsv3 nfs ip_gre gre ifb ip_tunnel xfs libcrc32c vfat fat lockd sunrpc grace ext4 jbd2 mbcache nls_utf8 nls_iso8859_1 nls_cp437 exportfs sha1_generic ecb crc32c_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[  228.120000] CPU: 0 PID: 26 Comm: kworker/u8:2 Tainted: G        W       4.4.61 #0
[  228.120000] Workqueue: phy0 mt7603_mac_work [mt7603e]
[  228.120000] Stack : 8fd2b100 80480000 80404954 000004cc 8ecf5428 00000000 8fc04614 8fd2b118
[  228.120000]    00000088 80064fa8 80404954 00000000 0000001a 804e367c 80420404 8fd67d54
[  228.120000]    80480000 80062cf4 80490000 80500000 8048e288 8048e28c 80409268 8fd67d54
[  228.120000]    80480000 800424d0 8fc04614 8fd67d8c 00000877 00000000 00000000 00d67d74
[  228.120000]    8ecf526c 8fe1a900 8fe1a800 30796870 00000000 00000000 00000000 00000000
[  228.120000]    ...
[  228.120000] Call Trace:
[  228.120000] [<8001692c>] show_stack+0x50/0x84
[  228.120000] [<801b736c>] dump_stack+0x84/0xbc
[  228.120000] [<8002cf40>] warn_slowpath_common+0xa0/0xd0
[  228.120000] [<8002cff8>] warn_slowpath_null+0x18/0x24
[  228.120000] [<8ecf5428>] mt7603_mac_work+0x1bc/0x284 [mt7603e]
[  228.120000] [<80040100>] process_one_work+0x214/0x358
[  228.120000] [<80040ffc>] worker_thread+0x2d8/0x440
[  228.120000] [<800457ac>] kthread+0xd8/0xec
[  228.120000] [<80005478>] ret_from_kernel_thread+0x14/0x1c
[  228.120000]
[  228.210000] ---[ end trace 4c1ba2fc6726afd6 ]---

camelcamro, 2017-05-14:
any news ?
@devels: can we help with testing to identify this issue more in detail ?

camelcamro, 2017-05-21:
hmm, happening ~20 times per day if the 2,4GHZ network is used :(

nbd168, 2017-12-03:
Fixed in current versions

closed 2017-12-03 by nbd168