openwrt/mt76 #99: No TX on antenna 2

berniwa, 2017-04-27:
Hi,

currently it seems, that the driver doesn't utilize the second antenna properly.
We currently have a scenario, where we have one fixed antenna connected to ANT2 and one screw-in 
antenna connected to ANT1 is connected to the MT7628AN
As it turns out,  WIFI only works properly if ANT1 (that is optional) is connected and not when only ANT2 is there. 
We did some measurements and it turned out, that we see no data transmissions on ANT2, so ANT2 only seems to receive data, but not transmit.
Is this a known limitation? Can anyone image what the problem might be?

Greets
BWA

e-mailky, 2017-12-27:
Yes, berniwa, I'm encounter this issue too, anyone can help?

iancd, 2018-02-12:
@berniwa Are you able to test with another identical piece of hardware with original firmware and compare results? If you find something unexpected are you able to document a set of test steps for others to follow and reproduce?
Please could you also edit the title of this github issue to include MT7628AN to distinguish it from other issues with different mt76 chipsets?

berniwa, 2018-02-15:
@iancd To be honest, we migrated to the closed-source Mediatek drivers provided by Nossiac (https://github.com/Nossiac/mtk-openwrt-feeds), as we encountered many many issues with the mt76 driver (Kernel faults and lockup, non-working ap, ...).

nbd168, 2018-11-28:
Please try the latest version

xavier83, 2018-11-29:
@berniwa does it work with openwrt 18.06? I'm desperate to try anything that would fix the issues at this point :stuck_out_tongue_winking_eye:

ashbenz, 2018-11-30:
@nbd168 testing on zbt wg3526, 2.4ghz seems a lot better. Will see long term stability. Thanks.

closed 2019-01-25 by nbd168